tbnobody/OpenDTU #852: [JSON API] Add charset to response header

m-froehlich, 2023-04-27:
### Is your feature request related to a problem? Please describe.

When requesting the current status as JSON response no charset header is included.

/api/livedata/status

### Describe the solution you'd like

The HTTP response should be added the used charset in the Content-Type header field like this:

Content-Type=application/json; charset=utf-8

tbnobody, 2023-04-28:
According to iana the application/json content type does not require or support a charset parameter

See http://www.iana.org/assignments/media-types/application/json (especially the last sentence)

m-froehlich, 2023-04-28:
Well, this is not really correct. Yes, using "no chatset", works with the provided content. But I cannot really use "no charset". I have to decide for one. Using "none" in Java, actually falls back to the default one, which is UTF-16 I suppose, but may be something else depending on the operating system and settings. It is never a good idea to rely on that. I am using UTF-8 as a standard.

The point is, that I have to guess, what charset had been used to encode the content. If I cannot find any evidence, I have to test, if it works with "none", which it does or what charset might work. But why should I guess and test? Maybe some of the content only shows up depending on time or what ever. So my tests might have missed some portions of the content and will fail with the complete content using my charset decision.

You have used a certain charset to encode the content. So, please provide the info in the header field, so people can use it to avoid guessing and testing. It really won't hurt you to put that value into the header, will it?

tbnobody, 2023-04-28:
> You have used a certain charset to encode the content. So, please provide the info in the header field, so people can use it to avoid guessing and testing.

There is no guessing and testing... JSON implies(!) that it's utf-8. See RFC 8259 chapter 8.1 (https://www.rfc-editor.org/rfc/rfc8259)

"JSON text exchanged between systems that are not part of a closed ecosystem MUST be encoded using UTF-8"

m-froehlich, 2023-04-28:
I didn't know that. Then I'm ok with that.

Thanks for making this clear.

closed 2023-04-28 by m-froehlich

tbnobody, 2023-04-28:
Just let me know if you encounter any char set specific issues.

m-froehlich, 2023-04-28:
Thanks, I will. Nothing so far. Well, I am using UTF-8 anyway.