tbnobody/OpenDTU #862: Project doesn't compile with "platform = espressif32@>=6.1.0

casimirextreme, 2023-04-29:
### What happened?

The project doesn't compile today due to 

`platform = espressif32@>=6.1.0`

It looks like the latest tag is now 6.2.0 (pushed 4h ago) and add a breaking change.

Forcing to use `platform = espressif32@=6.1.0` let the latest Opendtu master to compile properly

### To Reproduce Bug

Since espressif32 tag v6.2.0

### Expected Behavior

Have the code to compile properly

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

90a0cb60c622c66f84d91bc102a5a7b424b26059

### Relevant log/trace output

```Shell
...
Archiving .pio/build/generic/lib930/libAsyncTCP.a
Compiling .pio/build/generic/lib506/ESP Async WebServer/AsyncWebSocket.cpp.o
Compiling .pio/build/generic/lib506/ESP Async WebServer/SPIFFSEditor.cpp.o
Compiling .pio/build/generic/lib506/ESP Async WebServer/WebAuthentication.cpp.o
Compiling .pio/build/generic/lib506/ESP Async WebServer/WebHandlers.cpp.o
Compiling .pio/build/generic/lib506/ESP Async WebServer/WebRequest.cpp.o
Compiling .pio/build/generic/lib506/ESP Async WebServer/WebResponses.cpp.o
Compiling .pio/build/generic/lib506/ESP Async WebServer/WebServer.cpp.o
Compiling .pio/build/generic/lib919/WiFiClientSecure/WiFiClientSecure.cpp.o
Archiving .pio/build/generic/lib789/libWiFi.a
Compiling .pio/build/generic/lib919/WiFiClientSecure/esp_crt_bundle.c.o
Compiling .pio/build/generic/lib919/WiFiClientSecure/ssl_client.cpp.o
In file included from /home/opendtu/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:176,
                 from .pio/libdeps/generic/ESP Async WebServer/src/ESPAsyncWebServer.h:24,
                 from .pio/libdeps/generic/ESP Async WebServer/src/WebRequest.cpp:21:
.pio/libdeps/generic/ESP Async WebServer/src/WebRequest.cpp: In member function 'const __FlashStringHelper* AsyncWebServerRequest::methodToString() const':
/home/opendtu/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/generic/ESP Async WebServer/src/WebRequest.cpp:965:34: note: in expansion of macro 'F'
   if(_method == HTTP_ANY) return F("ANY");
                                  ^
/home/opendtu/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/generic/ESP Async WebServer/src/WebRequest.cpp:966:38: note: in expansion of macro 'F'
   else if(_method & HTTP_GET) return F("GET");
                                      ^
/home/opendtu/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/generic/ESP Async WebServer/src/WebRequest.cpp:967:39: note: in expansion of macro 'F'
   else if(_method & HTTP_POST) return F("POST");
                                       ^
/home/opendtu/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/generic/ESP Async WebServer/src/WebRequest.cpp:968:41: note: in expansion of macro 'F'
   else if(_method & HTTP_DELETE) return F("DELETE");
                                         ^
[... 306 lines cut ...]
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/generic/ESP Async WebServer/src/WebResponses.cpp:83:22: note: in expansion of macro 'F'
     case 504: return F("Gateway Time-out");
                      ^
/home/opendtu/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/generic/ESP Async WebServer/src/WebResponses.cpp:84:22: note: in expansion of macro 'F'
     case 505: return F("HTTP Version not supported");
                      ^
/home/opendtu/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/generic/ESP Async WebServer/src/WebResponses.cpp:85:22: note: in expansion of macro 'F'
     default:  return F("");
                      ^
*** [.pio/build/generic/lib506/ESP Async WebServer/WebResponses.cpp.o] Error 1
=========================================== [FAILED] Took 30.99 seconds ===========================================

Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:00:30.993
====================================== 1 failed, 0 succeeded in 00:00:30.993 ======================================
```

tbnobody, 2023-05-01:
either wait for a fix in the upstream repo: https://github.com/espressif/arduino-esp32/issues/8108
or just replace this line
https://github.com/tbnobody/OpenDTU/blob/90a0cb60c622c66f84d91bc102a5a7b424b26059/platformio.ini#LL18C20-L18C20

with
`platform = espressif32@6.1.0`

If it's not fixed until tomorrow I will pin it to 6.1.0 myself.

tbnobody, 2023-05-03:
Added a workaround in 09f8a633d7ffe6e6bfcddd29009795ffc5551fc1

Should be fixed upstream soon.

closed 2023-05-03 by tbnobody