tbnobody/OpenDTU #863: HM 600

Mike223311, 2023-04-30:
### Is your feature request related to a problem? Please describe.

Hallo,
👍👍👍
erst mal meinen Dank an das Team der Ahoy- als auch openDTU.
👍👍👍

Ich betreibe ein Balkonkraftwerk (zwei Platten a‘ 385 Watt) mit einem Hoymiles HM 600 mit Seriennummer 114xxxxxxxxx. (Baujahr 2022)
Habe mir die Ahoy- als auch die openDTU gem. Anleitung zusammen gestellt.
Die Ahoy mit ESP8266 und NRF24L01+
Die open mit ESP32 und NRF24L01+
Beide mit der aktuellen Software geflasht.
Beide funktionieren bei mir einwandfrei, keine Probleme.

Problem:
Beide DTU‘s habe ich nun bei einem Freund ausprobiert, der auch einen Hoymiles HM600, allerdings ein älteres Model mit Seriennummer 104xxxxxxxxx betreibt.
Die AhoyDTU erkennt diesen WR und ich kann die Leistungsdaten auslesen, allerdings ist kein Powermanagement möglich.
Die openDTU erkennt den WR nicht, kein Auslesen, kein Powermanagement möglich.

Wäre super wenn die Profis beider Team‘s eine Lösung finden würden.

Grüsse 

### Describe the solution you'd like

-

### Describe alternatives you've considered

-

### Additional context

-

DPO99, 2023-04-30:
Hast Du mit der Opendtu schon Mal einen Wechselrichter auslesen können? Sprich funktioniert die ganze Kiste überhaupt?

Zu ahoy wirst du hier nicht fündig da gibt's ne eigenen GitHub.

Mike223311, 2023-04-30:
Hallo,ja, wie geschrieben funktionieren beide in der angegebenen Konfiguration bei mir einwandfrei. Habe bei beiden DTU‘s alles Daten zum Auslesen und bei beiden auch ein Powerlimit setzen.Ich hab diesen Thread bei Open und Ahoy rein gesetzt.GrüsseAm 30.04.23 um 18:13 schrieb DPO99
                
            
            
                
                    Von: "DPO99" ***@***.***>Datum: 30. April 2023An: "tbnobody/OpenDTU" ***@***.***>Cc: "Mike223311" ***@***.***>,"Author" ***@***.***>Betreff: Re: [tbnobody/OpenDTU] HM 600 (Issue #863)
                
                
Hast Du mit der Opendtu schon Mal einen Wechselrichter auslesen können? Sprich funktioniert die ganze Kiste überhaupt?
Zu ahoy wirst du hier nicht fündig da gibt's ne eigenen GitHub.

tbnobody, 2023-04-30:
> der auch einen Hoymiles HM600, allerdings ein älteres Model mit Seriennummer 104xxxxxxxxx betreibt.

Das ist dann kein HM600 sondern ein MI-600... dieser verwendet ein anderes Protokoll welches aktuell von OpenDTU nicht unterstützt wird. Siehe #509

Mike223311, 2023-04-30:
Danke, ich bin neu hier und konnte noch nicht alles lesen :-(GrüsseAm 30.04.23 um 18:24 schrieb tbnobody
                
            
            
                
                    Von: "tbnobody" ***@***.***>Datum: 30. April 2023An: "tbnobody/OpenDTU" ***@***.***>Cc: "Author" ***@***.***>,"Mike223311" ***@***.***>Betreff: Re: [tbnobody/OpenDTU] HM 600 (Issue #863)
                
                

der auch einen Hoymiles HM600, allerdings ein älteres Model mit Seriennummer 104xxxxxxxxx betreibt.

Das ist dann kein HM600 sondern ein MI-600... dieser verwendet ein anderes Protokoll welches aktuell von OpenDTU nicht unterstützt wird. Siehe #509

closed 2023-05-02 by tbnobody