tbnobody/OpenDTU #885: Invertername in der Live Anzeige/MQTT bei HMS Invertern

MiniOh, 2023-05-06:
### What happened?

Hallo, 

ich teste seit einigen Tagen die parallele Nutzung von HM und HMS Invertern.
Bei den HMS Wechselrichtern, wir nach einiger Zeit oder beim erneuten Aufruf der Webseite, in der Live Ansicht ein inkorrekter Inverter Namen angezeigt.

So sollte es aussehen:

[image]

Nach einigen Sekunden wird aber folgendes angezeigt:

[image]

In der Live Ansicht würde mich das nicht weiter stören, 
allerdings ändern sich auch die Topics/Pfade in der MQTT Übertragung.

Soll:
inverter-open-test-hms/xxxxxxxxxxx/0/powerdc

Ist:
inverter-open-test-hms//0/powerdc

Ist ein solches Verhalten bereits bekannt?
Braucht Ihr mehr Informationen?

Besten Dank schon mal.

### To Reproduce Bug

-

### Expected Behavior

-

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

71c937c

tbnobody, 2023-05-09:
> Ich kann es auf jeden Fall innerhalb von Sekunden reproduzieren.

Könntest du versuchen einen Blick auf die Serielle Konsole zu werfen? Einfach z.B. Putty herunterladen (https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html)

Den Connection Type auf "Serial" stellen, bei "Serial Line" den entsprechenden Port eintragen, Speed = 115200
Anschließend auf "Open" klicken.

Da sollte dann jede Menge Debug Output erscheinen (Hauptsächlich wohl RX und TX Meldungen.)

Interessant wäre, ob zum entsprechenden Zeitpunkt ab dem es schief geht etwaige Exceptions auftreten.

MiniOh, 2023-05-09:
OK, vielen Dank.
Mache ich gleich morgen wenn es hell ist und gebe dann hier Rückmelung.

MiniOh, 2023-05-09:
OK, es passiert auch ohne, dass eine Kommunikation mit dem Inverter stattfindet.

Hier ca. 30 Sekunden Log, vom Neustart, bis zu dem Punkt, dass der Inverter "falsch" angezeigt wird.
```

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (638) esp_core_dump_flash: No core dump  ÉÑ¥Ñ¥½¹found!
E (638) esp_core_dump_flash: No core dump partition found!

Starting OpenDTU
Initialize FS... done
Reading configuration... done
Reading PinMapping... found valid mapping done
Initialize Network... done
Setting Hostname... Configuring WiFi STA using new credentials... done
Initialize NTP... done
Initialize SunPosition... done
Initialize MqTT... done
Initialize WebApi... done
Initialize Display... done
Initialize LEDs... done
Check for default DTU serial... done
Initialize Hoymiles interface... NRF: Connection successful
WiFi connected
WiFi got ip: 192.168.0.15
CMT: Connection successful
  Setting CMT target frequency... 
  Setting radio PA level... 
CMT TX power set to 0 dBm
  Setting DTU serial... 
  Setting poll interval... 
  Adding inverter: 111111111111 - HMS-2000-01 done
done
Switch to WiFi mode
Setting Hostname... done
Configuring WiFi STA using existing credentials... done
Configuring WiFi STA DHCP IP... done
WiFi got ip: 192.168.0.15
Network connected
Connecting to MQTT...
Connected to MQTT.
Fetch inverter: 111111111111
Request SystemConfigPara
TX RealTimeRunData 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 0B 00 64 5A A1 FD 00 00 00 00 00 00 00 00 8F FB 58 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 0B 00 64 5A A1 FD 00 00 00 00 00 00 00 00 8F FB 58 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 0B 00 64 5A A1 FD 00 00 00 00 00 00 00 00 8F FB 58 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 0B 00 64 5A A1 FD 00 00 00 00 00 00 00 00 8F FB 58 
RX Period End
All missing
Nothing received, resend whole request
TX RealTimeRunData 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 0B 00 64 5A A1 FD 00 00 00 00 00 00 00 00 8F FB 58 
RX Period End
All missing
Nothing received, resend count exeeded
[... 558 lines cut ...]
All missing
Nothing received, resend whole request
TX AlarmData 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 11 00 64 5A A2 3E 00 00 00 00 00 00 00 00 AF 54 0D 
RX Period End
All missing
Nothing received, resend whole request
TX AlarmData 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 11 00 64 5A A2 3E 00 00 00 00 00 00 00 00 AF 54 0D 
RX Period End
All missing
Nothing received, resend count exeeded
TX SystemConfigPara 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 05 00 64 5A A2 3E 00 00 00 00 00 00 00 00 BB 40 19 
RX Period End
All missing
Nothing received, resend whole request
TX SystemConfigPara 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 05 00 64 5A A2 3E 00 00 00 00 00 00 00 00 BB 40 19 
RX Period End
All missing
Nothing received, resend whole request
TX SystemConfigPara 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 05 00 64 5A A2 3E 00 00 00 00 00 00 00 00 BB 40 19 
RX Period End
All missing
Nothing received, resend whole request
TX SystemConfigPara 865.00 MHz --> 15 84 65 71 88 80 13 09 52 80 05 00 64 5A A2 3E 00 00 00 00 00 00 00 00 BB 40 19 

```

tbnobody, 2023-05-09:
Also generell sieht man hier keine Exception oder ähnliches. Du testest das mit einem HMS-2000 (ich nehme an du hast die Seriennummer anonymisiert aber es ist im regelbetrieb die eines HMS-2000 eingetragen?) und aktuell ist dieser auch aus?

Ich habe hier zum Debuggen auch einen da. Bzw. gerade in einer DTU eingetragen und kann den Fehler so erstmal absolut nicht nachvollziehen. 

Mir fallen ziemlich häufig die `Websocket: [/livedata][12] connect` bzw. `Websocket: [/livedata][12] disconnect` auf... was machst du da?

MiniOh, 2023-05-09:
Ja, das ist ein HMS-2000, mit anonymisierter SN.
Dieser ist gerade nicht aktiv, da keine DC Spannung mehr geliefert wird.

Was ich bisher gemacht habe:
- v23.5.9 installiert.
- WiFi eingerichtet.
- MQTT aktivert.
- Profil für CMT und NRF importiert.
- Inverter hinzugefügt
- neugestartet

Nach dem Neustart wird der Inverter Namen richtig in der Live Ansicht angezeigt inkl. der SN.
Wenn man entwerder im Menü hin und her klickt, oder die Live Ansicht im Brower aktualisiert, ist der Inverter Namen falsch und die SN fehlt.

[image]

Testweise auch mal die Software neu geflasht. Das Verhalten bleibst aber gleich.
Wenn ich zusätzlich einen HM Inverter einbinde, verhält sich dessen Anzeige korrekt.

Bzgl: Websocket: [/livedata][12] connect bzw. Websocket: [/livedata][12] disconnect
Evtl beim Klicken durch die Menüs bzw. aktualisieren der Webseite?

tbnobody, 2023-05-09:
Habe hier jetzt mal den gleichen Namen "HMS-2000-01" vergeben und gute 3min ständig zwischen "Live-Ansicht" und "Über" hin und her gewechselt. --> Keine Probleme

Magst du ggf. mal einen Screenshot von Info --> System 

* Firmwareinformationen
* Hardwareinformationen
* Speicherinformationen

teilen?

MiniOh, 2023-05-09:
Ich glaube ich habe es gefunden.

Es gibt noch eine AhoyDTU welche online ist, die ebenfalls den Inverter über die gleiche SN ansprechen möchte (dieser antwortet zwar momentan nicht, weil offline, aber dennoch versuchen natürlich beide DTUs die Kommunikation aufzubauen)
Wenn die die AhoyDTU vom Strom trenne, ist die Anzeige in OpenDTU wieder korrekt.
Nehme ich die AhoyDTU wieder in Betrieb, tritt wieder der besagte Fehler auf.

Sorry für die wahrscheinlich unnötige Mühe.

Olli

tbnobody, 2023-05-09:
Ahhh ok. Das bedeutet das Pakete welche die AhoyDTU sendet von der OpenDTU empfangen werden. Diese kann damit aber nicht anfangen. Und dann passieren seltsame Dinge. Ich versuche das mal im Code nachzulesen. Eigentlich sind 2 DTU's an einem Inverter sowieso zum scheitern verurteilt (gibt hier genug Issues in denen der Inverter dann seltsame Werte liefert). Aber es darf auch nicht wie hier beschrieben crashen.

Ich bin aber schon mal froh das es vermeintlich "nur" dieses Problem war :)

tbnobody, 2023-05-09:
Darf ich noch fragen welcher Inverter Typ auf beiden DTU's eingetragen war? (Also HM oder HMS und welche Wattzahl?)

MiniOh, 2023-05-10:
Insgesamt habe ich 7 HM Inverter und 1 HMS welche über Ahoy abgefragt werden.

In OpenDTU hatte ich, einen HMS eingetragen, und auch zeitweise für den Parallelbetrieb zu testen auch mal 2 HM.
Die HM weisen bei mir diesen "Fehler" nicht auf, auch wenn diese von mehreren DTUs angefragt werden.
Gesten Abend bei der Fehlersuche hatte ich aber lediglich den einen HMS in OpenDTU eingetragen.

Getestet hatte ich zum einen nur mit Inverter Namen und SN,
Aber auch mal mit Modulnamen (M1, M2, ...) und 395W.
Aber das machte keinen Unterschied, der Fehler trat immer nach einigen Sekunden auf.

Gerne kann ich noch weitere Sachen testen, falls Bedarf besteht.

Beste Grüße

nicomania, 2023-05-10:
Also bei mir sind drei HM-600 im Dauerbetrieb eingetragen. Probleme mit seltsamen Werten hatte ich noch nicht. Bei mir wird ja auch viel die Web API penetriert. Ist hier vielleicht die Ursache oder Zusammenhang zu suchen?

nicomania, 2023-05-15:
Es passiert bei mir nun mehrmals täglich. Ich bin mir nun auch fast sicher, dass es mit Zugriffen auf die Web API zu tun haben muss. Ich habe bei meinem genannten Plugin jedoch leider keine Möglichkeit die pollingrate einzustellen.
@MiniOh Benutzt du die API? Wenn ja wie und wie oft? Oder nur mqtt? Vielleicht haben wir eine Schnittmenge und können es eingrenzen.

MiniOh, 2023-05-15:
Hallo ich nutze lediglich MQTT, allergings war bei bei mit eine 2. DTU die in Betrieb war, die das Verhalten ausgelöst hatte.
WIe bereits oben geschrieben, war bei mir der Fehler weg, nachdem ich in meiner Produktiven DTU, nicht die kleiche SN abgefragt habe.

homeautomation2022, 2023-05-23:
@tbnobody https://discord.com/channels/984173303147155506/1007518370503082004
Hier im Discord habe ich bereits was geschrieben, was dir helfen kann den Fehler zu beheben und zu provozieren (Anfragen anderer DTU), denke du findest den Fehler am ehesten.
Er besteht zumindets schon lange und betrifft auffällig den Namen und die SN und das nur im RAM.
Er kommt wohl durch falsch gepartse packete, wobei eigentlich die Längenprüfung das parsen abbricht so das es nciht zum Overflow kommen sollte, aber irgendwo ist da noch ein kleiner Fehler bzw. zusätzliche Überprüfung fehlt.

tbnobody, 2023-05-23:
> Hier im Discord habe ich bereits was geschrieben, was dir helfen kann den Fehler zu beheben und zu provozieren (Anfragen anderer DTU), denke du findest den Fehler am ehesten.

Es würde ggf. helfen Fehler etc. direkt auf Github zu reporten. Discord lese ich mangels Zeit nur ganz vereinzelt.

indie89, 2023-05-23:
So, jetzt ist der Fehler zum zweiten Mal an einem Tag aufgetreten. Hab kurz bevor im iobroker openDTU Adapter der "Websocket Loop" losging (siehe mein o.g. fälschlicherweise doppelt geöffnetes Issue) einen einsamen "Packet handling error" im log der openDTU entdeckt. Die "Websocket Loop" entsteht im iobroker durch die invalide UTF-8 Sequenz des Inverter-Namens (Komplette Log-Nachricht im Debug Mode im iobroker: "Websocket error: Error: Invalid WebSocket frame: invalid UTF-8 sequence").

tbnobody, 2023-05-23:
> einen einsamen "Packet handling error" im log der openDTU entdeckt.

Da wäre der genaue Fehler schon interessant :)

indie89, 2023-05-23:
Meinst du folgendes?

`14:52:21.888 > TX ActivePowerControl Channel: 40 --> 51 84 50 96 00 80 12 94 20 81 0B 00 03 E8 00 01 D0 40 C5 
14:52:21.936 > Interrupt received
14:52:22.100 > RX Channel: 3 --> 95 84 50 96 00 84 50 96 00 01 00 01 01 28 01 77 04 53 01 2B 01 66 04 30 00 01 E5 
14:52:22.202 > Interrupt received
14:52:22.250 > RX Channel: 3 --> 95 84 50 96 00 84 50 96 00 02 80 BE 00 01 8F EB 01 D1 01 CB 09 1D 13 8A 08 21 72 
14:52:22.296 > Interrupt received
14:52:22.354 > RX Channel: 3 --> 95 84 50 96 00 84 50 96 00 83 00 02 00 59 03 E8 01 37 00 03 49 92 48 
14:52:23.830 > RX Period End
14:52:23.830 > Packet handling error
14:52:23.943 > TX ActivePowerControl Channel: 61 --> 51 84 52 91 83 80 12 94 20 81 0B 00 03 E8 00 01 D0 40 43 
14:52:24.001 > Interrupt received
14:52:24.147 > RX Channel: 61 --> 51 84 50 96 00 84 52 91 83 81 0B 00 03 E8 00 01 D0 40 27 
14:52:25.869 > RX Period End
14:52:25.869 > All missing
14:52:25.869 > Nothing received, resend whole request`

tbnobody, 2023-05-23:
Ja genau! Das "Packet handling error" erhält man wenn der CRC Code im Datenpaket selbst nicht stimmt. Kann das bisher nur mit 2 DTUs provozieren. Aber an der Stelle sollte das Error Handling korrekt ablaufen. Zumindest gab es hier beim Auftreten dieser Fehler keinen kaputten Speicher.

@homeautomation2022 ich habe auch versucht dein Problem hier nachzustellen mit 2x NRF und den gleichen Inverter abgefragt. Bekam aber nur "Packet handling error". Ich habe nichtsdestotrotz noch was eingebaut um den Fragmentzähler besser zu verifizieren und zu prüfen ob die Fragemente noch in den `_rxFragmentBuffer` passen. Ohne die Empfangenen Daten die zu deinem Beispiel geführt haben wir das aber eher schwierig. (Werde die Änderungen später noch pushen)

indie89, 2023-05-23:
> Wobei ich gerade oben nochmal nachgelesen habe. Bei @MiniOh trat der Fehler auch nur bei Invertern mit CMT Modul auf. Bei dir (@homeautomation2022) ja auch. Ggf. ist da am CMT trotzdem noch was faul. Muss mir hier mal ein 2. Testboard mit CMT bauen

Bei mir ist's auf jeden Fall mit einem nrf+ Bord. 
Kann mit "CRC Code" uns dem "rxFragmentBuffer" nicht so wirklich was anfangen, versuche aber gerne so gut es geht bei der Lösung des Themas zu unterstützen (wenn ich der einzige mit nrf+ Modul und dem Problem sein sollte, stehe ich für alle Rückfragen gerne zur Verfügung). Sagt mir einfach, wie ich helfen kann :-)! Hab auf jeden Fall Mal die neue Firmware installiert und schaue Mal, ob das was ausmacht bei mir.

tbnobody, 2023-05-23:
@homeautomation2022 ich habe gerade eben ein 2. CMT Modul in betrieb genommen.. 
Habe hier auf beiden ESP's unterschiedliche Seriennummern eingetragen. Aber beide DTU's empfangen die anfragen der Gegenseite. Keine Antworten des Wechselrichters. (Fehlermeldung in der Konsole: `Inverter Not found!` --> DTU Empfängt Daten für einen Inverter für den sie selbst keine Seriennummer eingetragen hat)

Das erklärt dann auch das Verhalten das @nicomania gesehen hat (Der ja den gleichen Wechselrichter in 2 DTUs eingetragen hatte). Hier schlägt jetzt nämlich der neue Error Handler zu. (Beim Senden ist Byte 9 == 0x0). Ein Crash an dieser Stelle sollte so nicht mehr auftauchen

Ich frage mich, ob man das CMT Modul, ähnlich wie das NRF Modul, in Hardware nach Adressen filtern lassen kann.

homeautomation2022, 2023-05-23:
Jetzt wo du es sagst, ja, das mit dem Inverter Not found hatte ich auch schon hielt es aber für logisch, da ja das Signal empfangen wird, aber halt nicht in dieser DTU konfiguriert ist sondern in der anderen.
Über eine Filterung auf CMT Seite habe ich nichts gesehen, aber die DTU SN im CMD könnte ja in der Software abgeglichen werden und die mit nicht gleicher DTU ID verworfen? Halte ich erstmal für die einfachste umzusetzende Lösung.

homeautomation2022, 2023-05-24:
Mit nur einem NRF besteht dieser Bug übrigens auch schon lange, konnte aber halt nie genau gefunden werden und war zuletzt auch verschwunden bei mir, vermutlich da ich den Empfang sehr optimiert hatte: https://discord.com/channels/984173303147155506/1007518370503082004/1034946575589462026 erinnerst dich sicher zurück.

indie89, 2023-05-27:
Seit dem Update tritt der spezifische Fehler bei mir tatsächlich bisher nicht mehr auf! Richtig cool!

Dafür gibt es jetzt teilweise Verbindungsabbrüche sodass 1-3 WR tlws über mehrere Stunden nicht mehr geupdaten werden und das erst nach einem Neustart der dtu wieder funktioniert (im dtu Protokoll kommt dann einfach immer "nothing received, resend whole request"). Aber das ist vermutlich ein anderes Thema, was nur bisher immer von den kryptischen wr Namen überholt wurde und daher nie aufgefallen ist...

nicomania, 2023-05-28:
Das kann ich bestätigen. Mit v23.5.23 tritt auch bei mir das Problem bisher nicht mehr auf. Vielen Dank, das ging echt fix!

closed 2023-05-31 by tbnobody