tbnobody/OpenDTU #887: OpenDTU Yield Total / Yield Day reporting problem

obstler42, 2023-05-06:
### What happened?

The values for Yield Total / Yield day fluctuate up/down and thereby mess up energy reporting in HomeAssistant. My guess is this is because some inverters are reachable whereas some are already offline during sunset/sunrise phase. See attached screenshots.

### To Reproduce Bug

record values for yield total/day for at least 2 days

### Expected Behavior

yield total should only ever increase and never go down. yield day should only ever reset to zero on a new day and not go down during the day.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.4.28

### Anything else?

[image]

[image]

tbnobody, 2023-05-06:
* What do you see in the web interface in such phases when the value is zero?
* Do you see any reboots of the ESP32?
* Did you enable or disable the "Expire function" in the OpenDTU Mqtt settings?

obstler42, 2023-05-06:
No reboots of the ESP.

Retain / Expire options are turned on.

Same thing happened again just now, the web interface also shows the reduced Total Yield Day currently (see attached screenshot).

Sensor config in HA (as the totals are apparently not automatically configured in HA):

`  - name: "PV Yield Day"`
`    state_topic: "hel4/opendtu/ac/yieldday"`
`    unit_of_measurement: "Wh"`
`    device_class: energy`
`    state_class: total_increasing`

[image]

tbnobody, 2023-05-06:
Also das yield day nach unten fällt ist nachvollziehbar. da kann man aber auch nichts dran ändern. Scheint dann ähnlich wie #401 zu sein.

obstler42, 2023-05-07:
ok, aber das gleiche passiert leider auch bei Yield Total.. und somit auch das nicht fuer Energy Dashboard brauchbar, da HA dann in der Frueh 400 kWh Produktion aufbucht ;)

[image]

mischasworld, 2023-05-07:
Hallo Obstler,

ich habe das in Iobroker mit einem Skript gelöst:

Ich frage den Wert is_valid ab und schreibe nur in die Datenpunkte wenn dieser 1 ist und damit alle Wechselrichter erreichbar.

vg

obstler42, 2023-05-07:
Ok, ich werde isValid mal mitloggen und schauen ob das mit den richtigen/falschen Werten korreliert -- sofern dass immer 0 ist wenn "falsche" Werte kommen ist der Ansatz mit dem Script durchaus eine Idee ;)

obstler42, 2023-05-07:
Leider klappt das auch nicht... In dem Moment sind alle Werte in MQTT wieder auf 0 gegagen und isvalid wieder auf 1 (es war zuvor richtigerweise ein paar Stunden auf 0 als die Inverter nicht mehr erreichbar waren).

[image]

[image]

tbnobody, 2023-05-08:
> In dem Moment sind alle Werte in MQTT wieder auf 0 gegagen und isvalid wieder auf 1

Das ist ja korrekt und wird auch in meinem verlinkten Beitrag weiter oben beschrieben. Wenn z.B. der Inverter über den Tag 2kWh erzeugt hat (YieldDaily) und um 19:00 Uhr mangels Sonne und Spannung aus geht wird `is_valid` auf 0 gesetzt (der Inverter ist ja nicht mehr erreichbar). Die Statistikdaten (YieldDaily) bleiben auf ihrem letzten empfangenen Wert (also z.B. 2kWh).
Wenn allerdings um 19:05 Uhr nochmals etwas mehr Sonne erscheint und der Inverter sich wieder einschaltet geht `is_valid` wieder auf 1 (der Inverter ist wieder erreichbar und es werden Daten empfangen). In diesem Moment ist YieldDaily allerdings wieder 0 da der Inverter diesen Wert bei jedem Spannungsverlust zurücksetzt. Nur der YieldTotal Wert bleibt bei jedem PowerCycle erhalten.

obstler42, 2023-05-08:
Nein, OpenDTU uptime is 3 Tage und 20 Stunden, kein Neustart seit dem letzten Anstecken. PV Yield Total ist die Summe aus 4 HMT2250.

tbnobody, 2023-05-08:
Hast du ggf. noch eine Übersicht was die Einzelwerte der Inverter zum entsprechenden Zeitpunkt zeigen? Also ich möchte nachvollziehen ob die 0er aus den Invertern kommen oder ob es bei der Summierung noch ein Thema gibt.

tbnobody, 2023-05-08:
Uhm, kann es sein, das du das Polling bei Nacht ausgeschaltet hast? Aktuell werden nur Inverter aufaddiert bei denen auch das Polling aktiv ist. Wenn das bei keinem Inverter der Fall ist --> 0
Es sollte hier aber eher `is_valid` einfach auf 0 gesetzt werden.

obstler42, 2023-05-08:
Ja genau, Polling in der Nacht ist deaktiviert. Gibt es einen sinnvollen Grund wieso man das ueber Nacht aktiv lassen sollte, da die Inverter sowieso nicht erreichbar sind zu der Zeit?

In diesem Fall waere isValid -> 0 vielleicht passender?

Ich hab mit einstweilen als Workaround mit einem Script beholfen dass eine weitere Variable verwendet und nur einen neuen Wert schreibt sofern isValid == 1 und neuer Wert > alter Wert.

tbnobody, 2023-05-08:
Pollen deaktivieren ist schon legitim... 

> In diesem Fall waere isValid -> 0 vielleicht passender?

Aber ist es ist ja alles valide. Der Inverter ist absichtlich deaktiviert und wird daher nicht in die Summe aufgenommen. 
Es sind alle "nicht deaktivierten" Inverter erreichbar. Damit ist das Ergebnis basierend auf diesen Parametern gültig.

Ich bin gerade noch komplett unschlüssig wie man es "besser" machen könnte. Aber einen deaktivierten Inverter mit in die Summe aufzunehmen erscheint mir auch falsch. Weil man könnte einen Inverter ja auch generell deaktivieren. Da mag man sicher auch nicht das er in diversen Summen o.ä. auftaucht.

hcs-svn, 2023-05-30:
Ich habe da ein sehr ähnliches Problem mit den totals, sie sind sporadisch falsch.
Abgefragt werden ein HM300 und ein HM600.
Ich hole die Daten einmal pro Minute von der API (…/api/livedata/status)
Falsche Werte habe ich bisher nur in total gesehen.
Das Problem ist mit den dickeren Pfeilen markiert.

Hier erst mal als Kurven:
[image]

Hinweis: Total_YieldDayAC in der Datenbank ist total.YieldDay.v aus dem JSON
Wenn ich mir die Werte in der Datenbank anschaue, dann sieht man, dass um 02:09:00 der Wert einmal falsch war:

[image]

Wenn man nun die einzelnen Werte und den Datensatz davor und danach betrachtet:

[image]

erkennt man, dass in Total_YieldDayAC einmal um 02:09:00 2,21 kWh fehlen, das ist genau der YieldDay vom HM300, der da fehlt. Und um 02:10:00 ist es wieder richtig.
 
Das „total.Power.v Problem“ der grünen Kurve oben um 13:42 ist ähnlich gelagert. Der Wert war in genau einem Datensatz 0, die restlichen Werte dazu haben gestimmt. Sowohl für den HM300 als auch dem HM600 hat das JSON die korrekte Leistung geliefert, nur die Summe in Total war 0.

[image]

Gerade eben erst gesehen: um 08:45 ist das Problem (auch in der Datenbank sichtbar) auch aufgetreten.
Sorry, dass meine Namen in der Datenbank nicht dem JSON entsprechen, aber ich denke, man versteht, wo die Werte herkommen.

Nach aktuellem Forschungsstand würde ich sagen, dass die Daten korrekt vom WR gekommen sind aber die Berechnung der totals manchmal fehlschlägt.

„Daten auch nachts abrufen“ und „Befehle auch nachts senden“ war für beide WR aus, das habe ich jetzt mal eingeschaltet, um zu sehen, ob das etwas ändert.

Ich kann gerne etwas ausprobieren, falls gewünscht, es ist optisch sehr einfach zu erkennen, ob das Problem da oder weg ist :-)

Und noch zu Info:
[image]

hcs-svn, 2023-05-30:
Man kann ja mal Glück haben :-)

Im Browser auf .../api/livedata/status einige mal F5 gedrückt und plötzlich das JSON unten bekommen
In "total" alles 0.
Danach noch gefühlt 50 mal F5 gedrückt und jedes mal alle Werte in "total" korrekt.

Die Werte in diesem JSON sind alle korrekt, außer die drei in total.

```
{
  "inverters":[
    {
      "serial":"????????????",
      "name":"HM600",
      "data_age":0,
      "poll_enabled":true,
      "reachable":true,
      "producing":true,
      "limit_relative":100,
      "limit_absolute":600,
      "AC":{
        "0":{
          "Power":{
            "v":379.6000061,
            "u":"W",
            "d":1
          },
          "Voltage":{
            "v":238.6999969,
            "u":"V",
            "d":1
          },
          "Current":{
            "v":1.590000033,
            "u":"A",
            "d":2
          },
          "Power DC":{
            "v":397.5999756,
            "u":"W",
            "d":1
          },
          "YieldDay":{
            "v":3566,
            "u":"Wh",
            "d":0
          },
          "YieldTotal":{
            "v":845.552002,
            "u":"kWh",
            "d":3
          },
          "Frequency":{
            "v":49.97000122,
            "u":"Hz",
            "d":2
          },
          "PowerFactor":{
            "v":1,
            "u":"",
            "d":3
          },
          "ReactivePower":{
            "v":0.300000012,
            "u":"var",
            "d":1
          },
          "Efficiency":{
            "v":95.47283936,
            "u":"%",
            "d":3
          }
        }
      },
      "DC":{
        "0":{
          "name":{
            "u":"Süd"
          },
          "Power":{
            "v":137.8000031,
            "u":"W",
            "d":1
          },
          "Voltage":{
            "v":32.5,
            "u":"V",
            "d":1
          },
          "Current":{
            "v":4.25,
            "u":"A",
            "d":2
          },
          "YieldDay":{
            "v":1948,
            "u":"Wh",
            "d":0
          },
          "YieldTotal":{
            "v":454.1809998,
            "u":"kWh",
            "d":3
          },
          "Irradiation":{
            "v":39.37142944,
            "u":"%",
            "d":3
          }
        },
        "1":{
          "name":{
            "u":"West"
          },
          "Power":{
            "v":259.7999878,
            "u":"W",
            "d":1
          },
          "Voltage":{
            "v":30.5,
            "u":"V",
            "d":1
          },
          "Current":{
            "v":8.520000458,
            "u":"A",
            "d":2
          },
          "YieldDay":{
            "v":1618,
            "u":"Wh",
            "d":0
          },
          "YieldTotal":{
            "v":391.3710022,
            "u":"kWh",
            "d":3
          },
          "Irradiation":{
            "v":74.22856903,
            "u":"%",
            "d":3
          }
        }
      },
      "INV":{
        "0":{
          "Temperature":{
            "v":46.5,
            "u":"°C",
            "d":1
          }
        }
      },
      "events":1
    },
    {
      "serial":"????????????",
      "name":"HM300",
      "data_age":4,
      "poll_enabled":true,
      "reachable":true,
      "producing":true,
      "limit_relative":100,
      "limit_absolute":300,
      "AC":{
        "0":{
          "Power":{
            "v":31.5,
            "u":"W",
            "d":1
          },
          "Voltage":{
            "v":237.6000061,
            "u":"V",
            "d":1
          },
          "Current":{
            "v":0.129999995,
            "u":"A",
            "d":2
          },
          "Power DC":{
            "v":33,
            "u":"W",
            "d":1
          },
          "YieldDay":{
            "v":1929,
            "u":"Wh",
            "d":0
          },
          "YieldTotal":{
            "v":101.387001,
            "u":"kWh",
            "d":3
          },
          "Frequency":{
            "v":49.95999908,
            "u":"Hz",
            "d":2
          },
          "PowerFactor":{
            "v":1,
            "u":"",
            "d":3
          },
          "ReactivePower":{
            "v":0,
            "u":"var",
            "d":1
          },
          "Efficiency":{
            "v":95.45454407,
            "u":"%",
            "d":3
          }
        }
      },
      "DC":{
        "0":{
          "name":{
            "u":"Ost"
          },
          "Power":{
            "v":33,
            "u":"W",
            "d":1
          },
          "Voltage":{
            "v":32.20000076,
            "u":"V",
            "d":1
          },
          "Current":{
            "v":1.019999981,
            "u":"A",
            "d":2
          },
          "YieldDay":{
            "v":1929,
            "u":"Wh",
            "d":0
          },
          "YieldTotal":{
            "v":101.387001,
            "u":"kWh",
            "d":3
          },
          "Irradiation":{
            "v":9.428570747,
            "u":"%",
            "d":3
          }
        }
      },
      "INV":{
        "0":{
          "Temperature":{
            "v":35.29999924,
            "u":"°C",
            "d":1
          }
        }
      },
      "events":4
    }
  ],
  "total":{
    "Power":{
      "v":0,
      "u":"W",
      "d":0
    },
    "YieldDay":{
      "v":0,
      "u":"Wh",
      "d":0
    },
    "YieldTotal":{
      "v":0,
      "u":"kWh",
      "d":0
    }
  },
  "hints":{
    "time_sync":false,
    "radio_problem":false,
    "default_password":false
  }
}
```

tbnobody, 2023-05-30:
hab schon einen verdacht woran es liegt. ist wirklich ein ganz enges zeitfenster in dem das auftreten kann.

hcs-svn, 2023-05-30:
Vermutest Du, dass es in DatastoreClass::loop() liegt, wenn zwischen dem 0 setzen oben und der Stelle weiter unten, wo der Wert wieder gesetzt wird, die Abfrage vom Web gerade zu diesem ganz unglücklichen Zeitpunkt kommt?
Da würde auch passen, dass ein WR nicht eingerechnet wird, wenn man mit dem API-call noch unglücklicher in die Schleife reintrifft.

tbnobody, 2023-05-30:
Exakt ja

tbnobody, 2023-05-31:
Sollte mit 24f063dd7b2fe34855b33bdbf1241be5a4cc3381 behoben sein.

closed 2023-06-02 by tbnobody