tbnobody/OpenDTU #890: SH1106 works but chopped text and partially unreadable....(latest version on ESP32)

njordan77, 2023-05-07:
### What happened?

I did connect a 1,3inch I2C display with OLED. works as it should so the start screen is visible. 
on Version [v23.5.3]

BUT the display is chopped, so the text is cut off in between and unreadable moving arround like the screensaver, but not as a complete image but chopped and compltely unreadable. did disable screensaver but this changes not really a lot.

Is this known? and maybe a fix for myself to overcome this.

Many Thanks
Norbert

### To Reproduce Bug

always, directly after booting and after the openDTU intro screen.

### Expected Behavior

x

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.5.3

### Relevant log/trace output

```Shell
x
```

### Anything else?

x

tbnobody, 2023-05-07:
Please send a image of the display and maybe a link to the shop where you bought it.

njordan77, 2023-05-07:
This is the display. its SH1106 configured as I2C device. It works in general, otherwise it would not show the start screen +  you can read the data but its chopped. will try to take a picture to better understand.

https://www.conrad.at/de/p/joy-it-sbc-oled01-3-display-modul-3-3-cm-1-3-zoll-128-x-64-pixel-passend-fuer-entwicklungskits-arduino-raspberry-pi-2619578.html

Hardware Config is the default json from the github page, with NodeMCU+SH1106....so its simply pin 21+22. no custom changes.

tbnobody, 2023-05-08:
> will try to take a picture to better understand.

Yes please

tbnobody, 2023-06-14:
Going to close this issue due to inactivity. Feel free to open it again.

(The movement of the display is btw. intended to  prevent a burn-in)

closed 2023-06-14 by tbnobody

yannickameur, 2023-09-11:
I have same probleme : 
[image]