tbnobody/OpenDTU #907: Logfile schreibt nur 16 Einträge

gmevi1954, 2023-05-09:
### What happened?

Im Ereignislog werden nur 16 Eintrage aufgeführt. Es findet kein first-in firts-out rolling statt was ratsam wäre.

### To Reproduce Bug

Dtu neu starten, fehler tritt nach ca. 50 Minuten auf

### Expected Behavior

Ein Neustart der dtu löscht die Log-Einträge.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.5.3

### Relevant log/trace output

```Shell
Ereignisanzeige

Begin	Ende	Fehlernummer	Meldung
06:03:21	06:03:21	1	Inverter start
11:08:28	11:08:28	2	DTU command failed
11:09:18	11:09:18	2	DTU command failed
11:09:28	11:09:28	2	DTU command failed
11:12:58	11:12:58	2	DTU command failed
11:12:58	11:12:58	2	DTU command failed
11:16:05	11:16:05	2	DTU command failed
11:16:05	11:16:05	2	DTU command failed
11:28:48	11:28:48	2	DTU command failed
11:28:53	11:28:53	2	DTU command failed
11:48:31	11:48:31	2	DTU command failed
11:48:32	11:48:32	2	DTU command failed
11:48:57	11:48:57	2	DTU command failed
11:50:48	11:50:48	2	DTU command failed
11:55:09	11:55:09	2	DTU command failed

kein weiteren Einträge
Status der Live Seite ist:
HM602
Seriennummer: 114182952590
Aktuelles Limit: 600 W | 100 %
Letzte Aktualisierung: vor 10.872 Sekunden / 9.5.2023, 12:15:22
Farbe: rot
```

tbnobody, 2023-05-09:
Das Eventlog kommt 1-zu-1 aus dem Inverter. Es wir nur angezeigt was dieser im Puffer hat. Es gab mal einen Verdacht das man die letzte Empfangene Event ID in einem Command wieder an den Inverter zurücksenden kann aber das wurde noch nicht weiter untersucht bzw. implementiert. 
Der Inverter selbst priorisiert die Events intern aber schon. Also ein AC Stromausfall würde dauerhaft stehen bleiben und z.B. die Events mit ID 2 verdrängen.

gmevi1954, 2023-05-09:
Danke für die Info.
  
Grüße
  
Gerd
  

  

  
  
Gerd Mevißen
  
Im Lohfeld 18
  
52457 Aldenhoven
  

  
Mobil +49 176 9600 8749
  
Telefon +49 2564 908925
  
Fax +49 2464 908926
  
Internet www.mevissen-sdrf.de

stefan-ohg, 2023-05-14:
Gibt es irgendwo eine Auflistung für was die Fehler stehen? Ich habe ab und zu ebenfalls den Fehlercode 2 im Log, aber bisher nichts gefunden was sich dahinter verbirgt.

Gruß

Stefan

tbnobody, 2023-05-16:
Leider nicht, sonst würde da auch nicht "Unknown" stehen :)

stefan123t, 2023-11-22:
@stefan-ohg die Unknow Events wurden in #121 bereits implementiert.
@gmevi1954 die aktuelle Event ID sollte wie in #104 beschrieben auf DTU Seite in den RealTimeRunData_Debug und anderen Nachrichten angepaßt werden. Mit dem Delta kann man prüfen ob es neue Event Einträge auf WR Seite gibt.
Die alten Events sollte man ggf. mit Type_CleanState_LockAndAlarm zurücksetzen können.

@tbnobody kann m.E. als duplicate von #104 geschloßen werden.

tbnobody, 2023-11-22:
Duplicate of #104

closed 2023-11-22 by tbnobody