tbnobody/OpenDTU #910: Inverters going offline

vo5tr0, 2023-05-11:
### What happened?

I upgraded my firmware from Version 71680 (Sorry, I only have the version number from my backed up config file) to the latest Version. The Version I had before was maybe 1-2 month old and never had this kind of problem. I didn't do an update for a long time, because the release note said something about only flashing is possible to do the update, OTA won't work (Can't finde the release note right now)

Since then my inverters are going offline at least once per day. Sometimes a simple reboot of the ESP helps, sometimes I actually have to reboot all inverters to get a connection.

My setup: 
Olimex ESP32 POE ISO with SSD1306 Display powered via PoE
4 Inverters (2x 1500, 1x 800, 1x 600)

### To Reproduce Bug

No idea, how to reproduce this bug, except from waiting

### Expected Behavior

Continuous connection to my inverters until the sun is gone 

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

71c937c

mroenne2022, 2023-05-11:
I think you should read this note ?
Please follow this link: 

https://github.com/tbnobody/OpenDTU

The partition-table was modified : 

[image]

carp3-noctem, 2023-05-17:
I do have the same issue, mainly 2 days after each reboot of the DTU via the Software reboot switch. After i hit restart, it is working normal again.

cactus-online, 2023-06-12:
Hello,

maybe I have the same or a similar issue. I started with openDTU in April this year with version 23.4.17 and every thing worked fine. On June 1th I upgraded to 23.5.31. Since then I have a lot of trouble. 

I noticed it because after a certain (short) time the connection to the inverter fails. The console log looks like this:

`
08:41:47.466 > RX Period End
08:41:47.466 > All missing
08:41:47.466 > Nothing received, resend whole request
08:41:47.466 > TX RealTimeRunData Channel: 3 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 29 00 00 00 00 00 00 00 00 BC F2 0D 
08:41:47.500 > RX Period End
08:41:47.500 > All missing
08:41:47.500 > Nothing received, resend whole request
08:41:47.500 > TX RealTimeRunData Channel: 3 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 29 00 00 00 00 00 00 00 00 BC F2 0D 
08:41:47.766 > RX Period End
08:41:47.766 > All missing
08:41:47.766 > Nothing received, resend whole request
08:41:47.766 > TX RealTimeRunData Channel: 61 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 29 00 00 00 00 00 00 00 00 BC F2 0D 
08:41:48.275 > RX Period End
08:41:48.275 > All missing
08:41:48.275 > Nothing received, resend whole request
08:41:48.275 > TX RealTimeRunData Channel: 3 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 29 00 00 00 00 00 00 00 00 BC F2 0D 
08:41:48.841 > RX Period End
08:41:48.841 > All missing
08:41:48.841 > Nothing received, resend count exeeded
08:41:50.981 > Fetch inverter: 114184854536
08:41:51.138 > TX RealTimeRunData Channel: 40 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 2E 00 00 00 00 00 00 00 00 8C D4 1C 
08:41:51.645 > RX Period End
08:41:51.645 > All missing
08:41:51.645 > Nothing received, resend whole request
08:41:51.645 > TX RealTimeRunData Channel: 75 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 2E 00 00 00 00 00 00 00 00 8C D4 1C 
08:41:52.149 > RX Period End
08:41:52.149 > All missing
08:41:52.149 > Nothing received, resend whole request
08:41:52.149 > TX RealTimeRunData Channel: 3 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 2E 00 00 00 00 00 00 00 00 8C D4 1C 
08:41:52.772 > RX Period End
08:41:52.772 > All missing
08:41:52.772 > Nothing received, resend whole request
08:41:52.772 > TX RealTimeRunData Channel: 23 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 2E 00 00 00 00 00 00 00 00 8C D4 1C 
08:41:53.387 > RX Period End
08:41:53.387 > All missing
08:41:53.387 > Nothing received, resend whole request
08:41:53.387 > TX RealTimeRunData Channel: 40 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 2E 00 00 00 00 00 00 00 00 8C D4 1C 
08:41:53.866 > RX Period End
08:41:53.866 > All missing
08:41:53.866 > Nothing received, resend count exeeded
08:41:54.497 > Websocket: [/livedata][1] connect
08:41:55.992 > Fetch inverter: 114184854536
08:41:55.999 > TX RealTimeRunData Channel: 61 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 33 00 00 00 00 00 00 00 00 DC 41 C4 
08:41:56.663 > RX Period End
08:41:56.663 > All missing
08:41:56.663 > Nothing received, resend whole request
08:41:56.663 > TX RealTimeRunData Channel: 75 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 33 00 00 00 00 00 00 00 00 DC 41 C4 
08:41:57.281 > RX Period End
08:41:57.281 > All missing
08:41:57.281 > Nothing received, resend whole request
08:41:57.281 > TX RealTimeRunData Channel: 75 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 33 00 00 00 00 00 00 00 00 DC 41 C4 
08:41:57.725 > RX Period End
08:41:57.725 > All missing
08:41:57.725 > Nothing received, resend whole request
08:41:57.725 > TX RealTimeRunData Channel: 23 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 33 00 00 00 00 00 00 00 00 DC 41 C4 
08:41:58.295 > RX Period End
08:41:58.295 > All missing
08:41:58.295 > Nothing received, resend whole request
08:41:58.295 > TX RealTimeRunData Channel: 40 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 33 00 00 00 00 00 00 00 00 DC 41 C4 
08:41:58.757 > RX Period End
08:41:58.757 > All missing
08:41:58.757 > Nothing received, resend count exeeded
08:42:01.093 > Fetch inverter: 114184854536
[... 2200 lines cut ...]
09:13:23.634 > Interrupt received
09:13:23.642 > RX Channel: 40 --> 95 84 85 45 36 84 85 45 36 02 57 5F 00 01 5E 23 00 37 00 35 09 11 13 8A 01 BF DE 
09:13:23.650 > RX Period End
09:13:23.650 > Success
09:13:27.786 > Fetch inverter: 114184854536
09:13:27.836 > TX RealTimeRunData Channel: 3 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 C5 97 00 00 00 00 00 00 00 00 D5 F8 AB 
09:13:27.850 > Interrupt received
09:13:27.862 > RX Channel: 40 --> 95 84 85 45 36 84 85 45 36 02 57 5F 00 01 5E 23 00 37 00 35 09 10 13 8A 01 BF DF 
09:13:27.864 > Interrupt received
09:13:27.870 > RX Channel: 40 --> 95 84 85 45 36 84 85 45 36 83 00 00 00 13 03 E8 00 D4 00 04 DD 69 8A 
09:13:28.298 > RX Period End
09:13:28.298 > Middle missing
09:13:28.298 > Request retransmit: 1
09:13:28.298 > TX RequestFrame Channel: 23 --> 15 84 85 45 36 80 17 25 68 81 3C 
09:13:28.312 > Interrupt received
09:13:28.323 > RX Channel: 40 --> 95 84 85 45 36 84 85 45 36 01 00 01 01 37 00 48 00 DF 01 2A 00 53 00 F7 00 01 BA 
09:13:28.353 > RX Period End
09:13:28.353 > Success
`

After that, the communication works well for a while.

I tried 23.6.9 also, but it became more instable. Than I tried 23.6.9 factory binary, but this doens't change anything. Switched back to 23.4.17, now it runs stable again, but the connection loss issue is already existing.

Thanks for any hint.

tklostermannNSD, 2023-06-13:
Bump, got the same issues...

This is an image of Home Assistant where you can see that the Inverter is most of the time "offline". Everything worked fine and well some versions ago (didnt look at the versions). And I even made a completely new setup (new installation with a second ESP), so the "partition thing" shouldn't be a thing here.
[image]

tobi-82, 2023-06-13:
I have the same Problem. The Partion is OK. I have uploaded the Software with VS Code. The Modul is mostly offline.
[image]

tklostermannNSD, 2023-06-14:
I downgraded to 2023.04.29 and I got back my online inverter. But the frontend feels bad again :D

cactus-online, 2023-08-07:
At me, it was a faulty radio module. After changing, this behavior occurs only extremely rare.

stefan123t, 2024-07-10:
Please note that we have renamed the Event Log Alarm Code `2` from `DTU command failed` to `Time calibration`.
This has been learned from the [warn_code.json](https://github.com/tbnobody/OpenDTU/issues/121#issuecomment-1752124208) in the S-Miles Installer App's APK file and concluded in #121.

@vo5tr0 Julian IMHO this may be something like the command sent to the inverter which contains the Timestamp has been delayed too much and does not match the internal timestamp of the inverter. 
This may happen when you have a bad or only somewhat working connection between the DTU and the inverter and the DTU has to repeat the command(s) very often for a longer period of time until the complete response is received from the inverter.

@cactus-online could you **Edit** and clean up your entry a bit ?

You can use the **[/] Slash commands** **Code block** and **Details** like so: 

<details><summary>Details</summary>
<p>

```
/details
/code
```

to get something like this:

```
<details><summary>Details</summary>
<p>

\```
08:41:47.466 > RX Period End
08:41:47.466 > All missing
08:41:47.466 > Nothing received, resend whole request
08:41:47.466 > TX RealTimeRunData Channel: 3 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 29 00 00 00 00 00 00 00 00 BC F2 0D
08:41:47.500 > RX Period End
08:41:47.500 > All missing
08:41:47.500 > Nothing received, resend whole request
08:41:47.500 > TX RealTimeRunData Channel: 3 --> 15 84 85 45 36 80 17 25 68 80 0B 00 64 86 BE 29 00 00 00 00 00 00 00 00 BC F2 0D
08:41:47.766 > RX Period End
08:41:47.766 > All missing
08:41:47.766 > N
\```

</p>
</details> 
```

**Note:** I have a hard time escaping the triple back ticks ` ``` ` inside the example code block.
Just remove the backslash from the two ` \``` ` if you want to copy the details with code block example verbatim.

</p>
</details>

tbnobody, 2024-10-03:
I am closing this issue due to no further activity

closed 2024-10-03 by tbnobody