tbnobody/OpenDTU #914: HMS-1000-2T

choefler1987, 2023-05-12:
### What happened?

Hi,
I have "Unknown model!".

Hoymiles HMS-1000-2T

### To Reproduce Bug

Please add new Hardware Info

### Expected Behavior

Please add new Hardware Info

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.5.9

tbnobody, 2023-05-12:
Please provide a screenshot of the dialog otherwise I am missing the required information.

homeautomation2022, 2023-05-24:
@tbnobody 
hin und wieder kommt es dazu: https://discord.com/channels/984173303147155506/1082394022359871499/1083653456839053324
Da wird irgendwie ein anderes Packet geparst wird, was eigentlich immer zu genau diesen unrealistischen Werten führt. richtig wäre die Ursache zu suchen, könnte ähnlich wie bei dem Namensbug sein, aber als quick fix wäre es auch gut einfach nur den FW Timestamp auf größer 2010 als Plausiblität einzubauen und sonst das Packet verwerfen.
Ansonsten werden hier immer wieder falsche Daten gemeldet...

Interessant wäre natürlich was für Daten da falsch geparst werden, weil auffällig immer Boot-Ver, FW-Ver und HW-Ver gleich sind und Timestamp zumindest mit 19 anfängt und Teilenummer keien Ahnung, so gut kann ich mir es dann auch nicht merken, aber übers zurückrechnen kommt man vielleicht drauf, was da schief geht, um das von vornherein zu vermeiden. Kann auch sein, das da 2 DTUs im Eisnatz waren und es ein Tx der anderen DTU ist.

oggli01, 2023-06-23:
Hallo
ich habe das problem behoben bei WR aus und dann erst den der das problem macht nun laufen beide mal sehen ob es wieder probleme macht wenn der 3. WR dazu kommt aber danke für die info

closed 2023-06-23 by tbnobody