tbnobody/OpenDTU #926: HMS-2000 Blindleistung falsch

D3R-ST3FAN, 2023-05-16:
### What happened?

Der HMS-2000 zeigt im letzten aktuelle build
eine Blindleistung von

Blindleistung  6.553,5  var an.

Sieht für mich nach einer Art Überlauf aus.
[image]

### To Reproduce Bug

Wert wird sofort anghezeigt.

### Expected Behavior

Richtiger Wert wird angezeigt.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.5.9

tbnobody, 2023-05-16:
Könntest du ggf. einen Auszug aus der Konsole (Info --> Konsole) posten? Interessant wären hier die RX Zeilen nach einem RealTimeRunData Befehl.

homeautomation2022, 2023-05-16:
Bei HMS-2000 mit Firmware 1.0.14 ist das ein mehrfach bestätigter bekannter Bug:
Leistungsfaktor(PF) sowie die damit verbundene Blindleistung(Q) sind da falsch und können nicht auf den realen Wert zurückgerechnet werden. Man müsste bei dieser Kombination eigentlich die beiden Werte verbergen, da nie richtig.

Der Hinweis sollte als bekannter Bug vielleicht irgendwo mit in die Readme aufgenommen werden.

Die haben da intern einen Berechnungsfehler drin:
Q scheint falsch zu skalieren / falsch berechnet zu werden und läuft mit zunehmender Leistung beim Sonnenaufgang von 0 bis 0xFFFF hoch und verbleibt da, das kann man bei langsammen Sonnenaufgang und auch wieder Untergang beobachten (nur im ganz niedrigen Watt Bereich).
Da nun der PF aus P und Q berechnet wird und Q irrsinnig hoch ist, resultiert daraus ein PF von nahezu 0.

D3R-ST3FAN, 2023-05-16:
Kann ich bestätigen Firmware-Version ist 1.0.14.

Ist denn eigentlich bei den HMS WRs die FW via DTU-Pro Updatebar?

homeautomation2022, 2023-05-16:
Ja, das macht/kann die original DTU. Es gibt auch schon min. eine neuere FW für den, die auch wieder richtige PF und Q Werte bringt. Die 1.0.12 davor hatte den Fehler übrigens nicht, nur die 1.0.14.

tbnobody, 2023-05-19:
Aufgrund der Tatsache das man an der Stelle eigentlich nicht mehr machen kann, habe ich das Verhalten hier dokumentiert https://github.com/tbnobody/OpenDTU/wiki/FAQ und würde das Issue schließen.

closed 2023-05-19 by tbnobody