tbnobody/OpenDTU #928: OpenDTU nicht mehr über Netzwerk erreichbar

FrodoVDR, 2023-05-17:
### What happened?

Ich habe das Problem das nach einigen Tagen keine Nachrichten per MQTT gesendet werden und die Oberfläche dann auch nicht mehr erreichbar ist. Ein angeschlossenes Display liefert weiterhin Daten.
Ein restart durch kurzes trennen der Stromversorgung behebt das Problem sofort.

Ich verwende OpenDTU mit dem CMT Modul und einem I2C Display (3 HMS Inverter). Bei Verwendung von CMT und NRF tritt der Fehler häufiger auf. 
Deshalb habe ich bereits eine zweite DTU am laufen die nur NRF ohne Display nutzt, dort gab es bisher keine Aussetzer. Scheinbar hat das Problem was mit der CMT Implementierung zu tun oder mit der größeren Anzahl von Invertern.

Leider kann ich anhand der MQTT Daten nicht sehen ob die DTU noch Daten sendet "is_valid = 1" ändert sich nicht mehr wenn die DTU nicht erreichbar ist.

### To Reproduce Bug

Ich weiß nicht wie man den Fehler nachvollziehen könnte, scheinbar habe nur ich das Problem.

### Expected Behavior

Ich würde mir wünschen das die DTU zusätzlich per MQTT einen Timestamp liefert, am besten in Unixtime. Dann kann man eine Abfrage z.B. in Node-Red erstellen ob die Daten noch aktuell sind und gegebenenfalls die DTU restarten.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.5.9

tbnobody, 2023-05-17:
> Ich würde mir wünschen das die DTU zusätzlich per MQTT einen Timestamp liefert, am besten in Unixtime. Dann kann man eine Abfrage z.B. in Node-Red erstellen ob die Daten noch aktuell sind und gegebenenfalls die DTU restarten.

Es gibt mit `dtu/uptime` eine Topic welche ständig nach oben zählt. Hier kann man vorherigen Wert mit aktuellem vergleichen und bei Nicht-Änderung reagieren. Desweiteren gibt es pro Inverter die Topic `[serial]/status/last_update` welches ein Unix Timestamp der letzten Erreichbarkeit des spezifischen Inverters ist. (Siehe https://github.com/tbnobody/OpenDTU/blob/master/docs/MQTT_Topics.md)

Es gibt aktuell auch noch #836. Weiß nicht ob das deinem beobachteten Verhalten entspricht.

schmittv, 2023-05-17:
Ich habe genau das gleiche Verhalten wie oben beschrieben. Ich verwende auch die Version v23.5.9. Nach einigen Tagen wird die Web-Oberfläche langsam und reagiert dann gar nicht mehr. Bei mir ist MQTT deaktiviert. Ich verwende das API "/api/livedata/status" um die vier angeschlossen Wechselrichter auszulesen, 3 x HM-800 und 1x HM-400. Da ich HM Wechselrichter verwende habe ich nur ein NRF Module verbaut. Display ist keines angeschlossen. Meistens funktioniert der API call noch einige Zeit nachdem die Weboberfläche den Geist aufgegeben hat, allerdings mit deutlich langsamer Antwortzeit.

tbnobody, 2023-05-17:
> Ich habe genau das gleiche Verhalten wie oben beschrieben. Ich verwende auch die Version v23.5.9. Nach einigen Tagen wird die Web-Oberfläche langsam und reagiert dann gar nicht mehr

Das würde aber dann mehr nach #836 aussehen...

schmittv, 2023-05-17:
Ich vermute das beides die gleiche Ursache hat.

FrodoVDR, 2023-05-17:
Die Meldungen aus #836 hatte ich auch in der Vergangenheit bei einem der drei Inverter gesehen, aber nicht bei meinem gemeldeten Bug, da hatte ich gar keine Möglichkeit die Oberfläche aufzurufen.

Aber Danke für die Hinweise bezüglich, dtu/uptime  oder [serial]/status/last_update.

wiedimot, 2023-05-24:
Problem besteht auch bei mir. 
Display liefert weiter Daten, jedoch ist die openDTU nicht mehr vom Netzwerk aus erreichbar. Problem tritt ungefähr alle 3-4 Tage auf. Habe auch Firmware v23.5.9

tbnobody, 2023-05-24:
Definiere nicht erreichbar. Reagiert der Host auf Pings? Geht das Webinterface nicht? Geht MQTT nicht?

wiedimot, 2023-05-24:
> Definiere nicht erreichbar. Reagiert der Host auf Pings? Geht das Webinterface nicht? Geht MQTT nicht?
> Welches Funkmodul?

- nRF Funkmodul
- Host Erreichbarkeit kann ich dir noch nicht sagen, werde ich testen, sobald das Problem wieder auftritt. Habe bereits neu gestartet. 
- Webinterface lies sich nicht öffnen
- mqtt hat Daten an HomeAssistant geliefert
- Die API hat auch Daten an Solaranzeige geliefert

Sorry für die unpräzisen Aussagen im letzten comment

tbnobody, 2023-05-24:
In der Version v23.5.23 hab ich nochmal potentiell kaputten Speicher behoben. 
Ich verwende aber selbst ein NRF Modul und habe problemlos Uptimes >10 Tage.

tbnobody, 2023-05-25:
Dann wäre die Ausgabe auf der seriellen Konsole interessant.

balu-, 2023-05-25:
> Dann wäre die Ausgabe auf der seriellen Konsole interessant.

dazu müsste ich einen Rechner via USB anschließen, das trennt die Stromversorgung und "behebt" "leider" das Problem.
Gibts einen anderen Weg an die Konsole zu kommen, den ich versuchen könnte?

unknown0816, 2023-06-13:
Ich habe wahrscheinlich das gleiche Problem. Die Oberfläche lädt nicht mehr und per ping komme ich auch nicht mehr an das Gerät. Scheint also aus dem WLAN geflogen zu sein.. Ich habe mitbekommen, dass sich das ganze (meist) von alleine erholt (und keine Neustart des Gerätes nötig ist). Irgendwann ist die WLAN Verbindung wieder aktiv und man kann wieder zugreifen.
Allerdings passiert das ganze bei mir täglich bis alle zwei Tage. 

Ich habe kein Display angeschlossen und nur ein Inverter registriert.

cj82rnk, 2023-06-26:
Interessant, also scheine ich nicht alleine zu sein.
Auch ich kann alles geschrieben bestätigen. Bei mir ist das Webinterface nicht erreichbar, MQTT schreibt auch keine neuen Daten in meinen ioBroker. Stecker ziehen hilft nicht immer. Problem löst sich dann aber irgendwann von alleine.
Aussetzer können mehrmals täglich vorkommen, aber auch nur vereinzelt ganz kurze oder keine. Alles schwer zu deuten.

Wenn ich weitere Daten liefern kann, meldet Euch

ranax404, 2023-06-29:
Ich habe bei 2 von 3 Installationen (verschiedene Häuser) das gleiche Problem:
- Web-GUI von openDTU nach einigen Tagen nicht mehr erreichbar,
- Ping nicht möglich
- Daten per MQTT werden auch nicht mehr geliefert

- angeschlossene Displays zeigen allerdings weiterhin korrekte Daten
- nach Neustart (Spannung weg) werden die gesammelten Infos per MQTT wieder übertragen

>> openDTU läuft also lokal scheinbar weiter.

Eingesetzte Hardware:
- ESP32 DevKitC v4
- NRF24L01+
- SSD1306 (0.96")

Fehler 1 tritt mit einem HM-700 auf. WLAN-Netzwerk (Fritzbox 7530) ist dort sehr stabil und guter Empfang (-50dBm)
Fehler 2 tritt mit einem HM-1500 auf. WLAN-Netzwerk (Unifi AP 6) ist dort sehr gut (-40dBm)

In beiden Netzen sind noch weitere ESP32/ESP8266 unterwegs, die stabil laufen. Daran kann es also nicht liegen.

Getestet bzw. Fehler sind bisher mit allen 23.6.x Versionen aufgetreten.

Die 3. Installation (identische HW, auch ein HM-1500 und mit Unifi AP 6), aber deutlich schlechterem WLAN (-70dBm bis -75dBm) läuft seit Wochen super stabil und hat zwischendurch auch mal Updates bekommen.

Andre0be, 2023-07-04:
Der ESP32 Chip hat wohl sehr oft Probleme mit dem WLAN. Ich habe dieses Problem auch. Es scheint ein Problem mit den WLAN Kanälen zu geben. Das ist bei Mesh natürlich schlecht, wenn der WLAN Kanal sich ändert. Ich habe bei mir die Erfahung gemacht, dass der ESP32 nur gut mit WLAN Kanal 1 funktioniert. In Summe steht das Projekt auf wackeligen Beinen, wenn ich mir so die Probleme ansehe.

rfelgent, 2023-07-31:
Hallo,

ich denke, es könnte an einem Speicher-Problem liegen.

Ich hatte keine "eingefrorene/nicht erreichbare" WebUI bis ich eines Tages die mqtt-Anbdinung aktiviert hatte. Dann kam es zu "scheinbaren" Ausfällen der WebUI.
Ihr könntet mal das Intervall für die mqtt-Anbindung erhöhen, z.B. auf 60 Sekunden. Default sind 5 Sekunden, was mMn. eh zu klein gesetzt ist. Minuten-genaue Werte sollte die meisten use-cases abdecken...

zerpixelung, 2023-08-13:
Gleiches Problem bei mir. Hier ein paar Infos:

- Ein HM-600 ist eingetragen
- Ping reagiert in der Regel. Mit durchschnittlich 100ms ziemlich lahm im eigenen WLAN mit nur 4m Abstand zum Router, aber ich gehe davon aus, dass das nicht ungewöhnlich in diesem Fall ist, oder?
- Web-GUI reagiert oft nur sporadisch. Manchmal erst nach mehreren Minuten. Dann plötzlich mal wieder sehr zügig.
- Direkt nach der ersten Einrichtung reagierte das Webinterface sofort und schnell. Nun verschlechtert sich das nach und nach.
- Gleiche Probleme mit einer Verbindung zu einem MQTT Broker und ohne MQTT
- Firmware 23.8.8 ist installiert. Mit 23.8.2 hatte ich die gleichen Probleme
- WLAN 2,4 GHz läuft auf Kanal 6 (Auto ist deaktiviert in der Fritzbox)
- Chip-Modell: ESP32-D0WD-V3
- Chip-Revision: 3
- nRF24L01+ 
- DTU Abfrageintervall: 30s
- MQTT Intervall: 30s

zerpixelung, 2023-08-27:
Nachdem OpenDTU anfangs wieder Probleme hatte (siehe Nachtrag vorheriger Post), läuft er nun über eine Woche zuverlässig. Das Web-GUI hat immer schnell reagiert, sämtliche Daten scheinen an den MQTT Server übermittelt worden zu sein.

Da die aktuelle Position der OpenDTU im Raum nur provisorisch ist, werde ich eine weitere Stelle im Haus testen und berichte dann wieder.

ics1702, 2023-09-01:
Ich habe die gleichen Probleme wie oben beschrieben. Ich habe nach einpaar Tagen ist kein Zugriff über Web mehr möglich. Wenn ich allerdings den WLAN Kanal fest auf Kanal 1 Stelle funktioniert es dauerhaft.

Linos88, 2023-10-06:
Auch ich möchte mich der genannten Problematik mit anschließen.
Gerne bin ich bereit der Fehlersuche zu unterstützen, wenn man mir sagt wie :-)

SDK-Version	v4.4.4
Konfigurationsversion	0.1.25
Firmwareversion / git Hash	[v23.9.18]
PIO Umgebung	generic_esp32
Chip-Modell | ESP32-D0WDQ6 mit CMT2300A

zerpixelung, 2023-10-06:
Nach über 4 Wochen stabilen Betriebs an der provisorischen Position, läuft OpenDTU seid 2 Tagen auch an einer Position, die nun der dauerhafte Platz werden soll.

1. Die Position an der OpenDTU instabil lief hatte einen TV, eine Nintendo Switch und eine PlayStation 4 in unmittelbarer Umgebung.
2. Die provisorische Position war direkt auf meinem Schreibtisch. Mit Monitor, Desktop PC, Funk Maus, Fritzbox, Philips Hue Bridge in unmittelbarer Nähe.
3. Die jetzt feste Position ist neben einer Synology NAS, WLAN Drucker und Fritz Repeater.

Ich bin kein Fachmann was das betrifft, aber bei Position 2. und 3. ist kein Bluetooth in der Nähe.

Linos88, 2023-10-12:
Nachtrag zu meinem Problem mit der Erreichbarkeit (https://github.com/tbnobody/OpenDTU/issues/928#issuecomment-175020838)

Augenscheinlich, wenn der ESP32 sich über einen Accesspoint mit dem WLAN (Fritz Mesh) verbindet kommt es nach einer gewissen Zeit zu den genannten Problemen. Zeit bis zum Auftritt ist willkürlich, von wenigen Stunden bis hin zu mehreren Tagen war schon alles dabei.

Während der nicht Erreichbarkeit ist der ESP32 laut FritzBox online und hat seinen Zugangspunkt geändert. 

Abhilfe mit momentaner erkennbarer Verbesserung ist die Positionierung vom ESP32 unmittelbar bei der FritzBox. Dadurch wählt sich der ESP32 direkt in das WLAN von der FB ein und ist derzeit stabil erreichbar.

zerpixelung, 2023-10-23:
Ich habe nun auch eine ähnliche Beobachtung gemacht, die @Linos88 in seinem/ihrem letzten Kommentar genannt hat. Der ESP32 ist bei mir nun in einem Fritz Mesh und macht wieder Zicken.
Bei mir läd das OpenDTU Webinterface noch. Zeigt allerdings nichts mehr in der Live-Ansicht an. Für mich persönlich allerdings nicht mehr so problematisch, weil die Datenübertragen zu meinem MQTT Broker dennoch einwandfrei funktioniert. Sprich in der InFluxDB und Grafana kommt das was ich wissen will zuverlässig an.

unknown0816, 2023-11-24:
Ich gebe auch nochmal ein Update. Bei mir ist der ESP32 mittlerweile durchgeängig erreichbar. Bei mir lag es wohl an dem Fritz.Repeater (mit Mesh), der ab und zu das Signal zum Router verloren hat. Ich habe den Repeater jetzt auf eine Kabelverbindung umgestellt, und seitdem habe ich keine Probleme mit der Konnektivität des ESP32.

Ich vermute allerdings, dass das ganze eher ein Zusammenspiel zwischen Verbindungsabbrüchen zwischen Repeater zu Router und dem ESP32, der wohl mit kurzen Aussetzern nicht klarkommt, ist. Denn kein anderen Gerät was vorher mit dem Repeater verbunden war, hatte solche Probleme.

stefan123t, 2024-07-10:
@FrodoVDR kannst Du noch etwas zu Deinem WLAN Netzwerk schreiben. Einige der Anderen haben ja konkret Fritz!Box mit Mesh Funktion erwähnt. Es gibt dazu m.W. schon länger Berichte zu Problemen mit dem Wechsel vom Router zu einem Access Point der evtl. kurzzeitig besser zu empfangen ist. Offenbar kann das der TCP/IP Stack des ESP32 nicht so gut ab und verhaspelt sich da ab und zu.

Um solche Probleme genauer nachvollziehen zu können braucht es zwingend USB Serial Console Logs von der OpenDTU am Übergang von "geht" zu "geht nicht" und möglichst viele Hintergrund Infos zum WLAN.

Auch die MQTT Bibliothek hatte m.W. in der Vergangenheit Schwierigkeiten mit dem Konstruktor wenn die WLAN Verbindung mal weg war, wurde der nämlich nicht zwingend neu initialisiert. Aber das ist m.W. bereits seit längerem behoben.

Powermaniaxx, 2024-07-13:
Habe leider auch das Problem, dass sich der ESP32-Chip mit OpenDTU immer wieder aufhängt. Wlan ist fest auf 1 Kanal, kein Mesh. Nervig

[image]

`Fetch inverter: 114190930561
16:41:12.007 > Request SystemConfigPara
16:41:12.058 > TX RealTimeRunData Channel: 75 --> 15 90 93 05 61 80 15 09 32 80 0B 00 66 92 92 08 00 00 00 00 00 00 00 00 8F 1F A9 
16:41:12.604 > RX Period End
16:41:12.604 > All missing
16:41:12.604 > Nothing received, resend whole request
16:41:12.604 > TX RealTimeRunData Channel: 3 --> 15 90 93 05 61 80 15 09 32 80 0B 00 66 92 92 08 00 00 00 00 00 00 00 00 8F 1F A9 
16:41:13.155 > RX Period End
16:41:13.155 > All missing
16:41:13.155 > Nothing received, resend whole request
16:41:13.155 > TX RealTimeRunData Channel: 23 --> 15 90 93 05 61 80 15 09 32 80 0B 00 66 92 92 08 00 00 00 00 00 00 00 00 8F 1F A9 
16:41:13.709 > RX Period End
16:41:13.709 > All missing
16:41:13.709 > Nothing received, resend whole request
16:41:13.709 > TX RealTimeRunData Channel: 40 --> 15 90 93 05 61 80 15 09 32 80 0B 00 66 92 92 08 00 00 00 00 00 00 00 00 8F 1F A9 
16:41:14.271 > RX Period End
16:41:14.271 > All missing
16:41:14.271 > Nothing received, resend whole request
16:41:14.271 > TX RealTimeRunData Channel: 61 --> 15 90 93 05 61 80 15 09 32 80 0B 00 66 92 92 08 00 00 00 00 00 00 00 00 8F 1F A9 
16:41:14.767 > RX Period End
16:41:14.767 > All missing
16:41:14.767 > Nothing received, resend count exeeded
16:41:14.820 > TX AlarmData Channel: 75 --> 15 90 93 05 61 80 15 09 32 80 11 00 66 92 92 08 00 00 00 00 00 00 00 00 55 04 72 
16:41:15.612 > RX Period End
16:41:15.612 > All missing
16:41:15.612 > Nothing received, resend whole request
16:41:15.612 > TX AlarmData Channel: 3 --> 15 90 93 05 61 80 15 09 32 80 11 00 66 92 92 08 00 00 00 00 00 00 00 00 55 04 72 
16:41:16.416 > RX Period End
16:41:16.416 > All missing
16:41:16.416 > Nothing received, resend whole request
16:41:16.416 > TX AlarmData Channel: 23 --> 15 90 93 05 61 80 15 09 32 80 11 00 66 92 92 08 00 00 00 00 00 00 00 00 55 04 72 
16:41:17.219 > RX Period End
16:41:17.219 > All missing
16:41:17.219 > Nothing received, resend whole request
16:41:17.219 > TX AlarmData Channel: 40 --> 15 90 93 05 61 80 15 09 32 80 11 00 66 92 92 08 00 00 00 00 00 00 00 00 55 04 72 
16:41:18.015 > RX Period End
16:41:18.015 > All missing
16:41:18.015 > Nothing received, resend whole request
16:41:18.015 > TX AlarmData Channel: 61 --> 15 90 93 05 61 80 15 09 32 80 11 00 66 92 92 08 00 00 00 00 00 00 00 00 55 04 72 
16:41:18.772 > RX Period End
16:41:18.772 > All missing
16:41:18.772 > Nothing received, resend count exeeded
16:41:18.903 > TX SystemConfigPara Channel: 75 --> 15 90 93 05 61 80 15 09 32 80 05 00 66 92 92 08 00 00 00 00 00 00 00 00 41 10 66 
16:41:19.174 > RX Period End
16:41:19.174 > All missing
16:41:19.174 > Nothing received, resend whole request
16:41:19.174 > TX SystemConfigPara Channel: 3 --> 15 90 93 05 61 80 15 09 32 80 05 00 66 92 92 08 00 00 00 00 00 00 00 00 41 10 66 
16:41:19.498 > RX Period End
16:41:19.498 > All missing
16:41:19.498 > Nothing received, resend whole request
16:41:19.498 > TX SystemConfigPara Channel: 23 --> 15 90 93 05 61 80 15 09 32 80 05 00 66 92 92 08 00 00 00 00 00 00 00 00 41 10 66 
16:41:19.644 > RX Period End
16:41:19.644 > All missing
16:41:19.644 > Nothing received, resend whole request
16:41:19.644 > TX SystemConfigPara Channel: 40 --> 15 90 93 05 61 80 15 09 32 80 05 00 66 92 92 08 00 00 00 00 00 00 00 00 41 10 66 
16:41:19.898 > RX Period End
16:41:19.898 > All missing
16:41:19.898 > Nothing received, resend whole request
16:41:19.898 > TX SystemConfigPara Channel: 61 --> 15 90 93 05 61 80 15 09 32 80 05 00 66 92 92 08 00 00 00 00 00 00 00 00 41 10 66 
16:41:20.101 > RX Period End
16:41:20.101 > All missing
16:41:20.101 > Nothing received, resend count exeeded
`

MatzeX71, 2024-08-14:
Ich habe genau das gleiche Problem wie von @FrodoVDR beschrieben. Gibt es mittlerweile einen Workaround?

stefan123t, 2024-08-14:
@MatzeX71 @Powermaniaxx leider kann man mit den bisher verfügbaren Logfiles nicht wirklich das Problem des WiFi Stacks auf dem ESP32 analysieren bzw. dessen Ursache lokalisieren. 

Dazu sind Eure Angaben bzgl. “Ich habe das selbe Problem” leider zu ungenau. In #2184 und #2185 hat @jstammi einige Hinweise gegeben was wir benötigen um Euer Problem besser eingrenzen zu können. Es scheint wohl mit Fritz!Mesh und dem Verbinden mit dem “besten” Access Point durch OpenDTU zusammen zu hängen, aber Näheres ist noch nicht bekannt.
Evtl liesse es sich durch regelmäßige Prüfung und ggf. Neuverbindung mit dem wirklich besten AP beheben oder auch durch Einsatz von WiFiMulti anstelle von WiFi.

OttiNC1, 2024-08-19:
Bei mir ist die DTU vom Router plötlich nicht mehr erreichbar. Taucht auch in der Netzwerk- Liste des Routers nicht mehr auf. Erst nach Router Neustart. DTU- Hardware ist geblieben, Router ist geblieben.

OttiNC1, 2024-08-24:
Die DTU verbindet sich immer noch nicht mit dem Router. Ping ist ebenfalls nicht erfolgreich. Die manuell heruntergeladene Konfig der OPEN DTU scheint nicht aktuell zu sein.

zerpixelung, 2024-08-25:
@OttiNC1 die Ursache des Problems ist bisher leider noch unbekannt. Die bisher stärkste Vermutung ist, dass es am Fritz Mesh liegt. Eindeutig validiert werden konnte dies bisher aber nicht. Es scheint aber als sicher, dass der ESP32 Chip auf etwas externes sensibel reagiert. Am besten, wenn noch nicht geschehen, diesen Thread einmal vollständig lesen und nach Möglichkeit am eigenen Netzwerk und Ort der Hardware auf der OpenDTU läuft, zu experimentieren. Aktuell wird dir hier keine konkrete Lösung genannt werden können. Vielleicht hilft dir folgendes...

---

Nach einigen weiteren Monaten nochmal ein Update von mir:
Ich bezweifle inzwischen, dass es pauschal am Fritz Mesh liegt. Seit Monaten hat OpenDTU keine Verbindungsprobleme mehr bei mir.
Mein Setup (vielleicht hilft dies Einigen die Symptome zu minimieren):
- OpenDTU bzw. der ESP32 hängt zur Stromversorgung an einem USB Port eines Synology DS220+ NAS
- OpenDTU ist etwa 20cm von einem FRITZ!Repeater 2400 mit FRITZ!OS:7.58
- Der Repeater ist via WLAN und Fritz Mesh mit einer FRITZ!Box 7530 mit FRITZ!OS:7.59 verbunden
- NAS ist via TP-Kabel mit dem Repeater verbunden.
- In der Nähe von OpenDTU befindet sich nur das NAS, der Repeater und ein Brother Printer, der via WLAN verbunden ist. Es befinden sich also sehr wenig potentielle Störquellen in der Nähe.
- Die meisten Probleme hatte ich in der Vergangenheit wenn ESP32 Chip in der Nähe von Geräten mit aktivierten Bluetooth war und vielen Geräten, die via WLAN verbunden waren.

Wer die genannten Probleme dieses Threads hat, sollte, zumindest testweise, den ESP32 mit OpenDTU an einen entlegenen Ort von anderer Hardware positionieren. Bei mir hat dies die Symptome drastisch minimiert.

Falls dies auch bei anderen hilft, sollten diese in Erwägung ziehen hier ein kurzes Feedback zu hinterlassen.

stefan123t, 2024-09-07:
We need Logfiles for the issue to be traceable by either support / developers.

Follow the link to the documentation to setup for USB / serial logging:
https://www.opendtu.solar/firmware/howto/serial_console/

OttiNC1, 2024-09-08:
Als ich die openDTU seriell über Putty mit dem Laptop verbunden hatte, war die WIFI- Verbindung zur FritzBox da. Laptop abgestöpselt, Verbindung zur FrizBox weg. 2x wiederholt mit gleichem Ergebnis.

OttiNC1, 2024-09-08:
Fehler! Der 5V Pin auf der Leiterplatte war nicht verlötet.

OttiNC1, 2024-09-08:
Diese Leiterplatte war allerdings heute das Erste Mal im Einsatz.

StefanHermannBerlin, 2024-10-07:
I had the same issue (working display, showing numbers, no connection to the network) and did the same processes to solve it. 
Now **I changed the power supply** and it works again. I hope

StefanHermannBerlin, 2024-10-07:
Same problem. Working display, showing numbers, no network connection.
[capture.txt](https://github.com/user-attachments/files/17276632/capture.txt)

StefanHermannBerlin, 2024-10-07:
I **changed the power supply** and now it is back online – let's see for how long. 
(Of course I restarted it with the old power supply before with no success.)

stefan123t, 2024-10-07:
@FrodoVDR hast Du in der Zwischenzeit Dein Problem mit MQTT und Verbindungsabbrüchen der OpenDTU lösen können oder Logfiles ?

@StefanHermannBerlin Du hast keine Antwort vom WR bekommen, da solltest Du mal die Hardware prüfen und ggf einen eigenen Issue aufmachen. Danke!

closed 2025-04-18 by github-actions[bot]