tbnobody/OpenDTU #930: HMS 2000 Wird angesprochen aber es baut sich keine Verbindung auf

deltahq, 2023-05-17:
### What happened?

Es werden keine Daten vom Wechselrichter an die DTU übertragen

### To Reproduce Bug

Habe die DTU neu geflasht, hat aber nichts gebracht.

### Expected Behavior

Es sollten Daten angezeigt werden

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.5.9

### Relevant log/trace output

```Shell
Websocket: [/livedata][1] connect
20:45:51.660 > Websocket: [/livedata][1] disconnect
20:46:04.658 > Admin mode disabled
```

deltahq, 2023-05-17:
Websocket: [/livedata][2] connect
20:59:52.797 > Websocket: [/livedata][2] disconnect
21:00:02.847 > Websocket: [/livedata][3] connect
21:00:21.052 > Websocket: [/livedata][3] disconnect
21:00:32.417 > Websocket: [/livedata][4] connect
21:01:41.346 > Websocket: [/livedata][4] disconnect
21:01:46.956 > Websocket: [/livedata][5] connect
21:02:03.053 > Websocket: [/livedata][5] disconnect
21:02:07.998 > Websocket: [/livedata][6] connect
21:02:19.594 > Websocket: [/livedata][6] disconnect
21:02:21.919 > Websocket: [/livedata][7] connect
21:02:55.000 > Websocket: [/livedata][7] disconnect
Das ist alles was in der Konsole angezeigt wird

tbnobody, 2023-05-17:
Bitte einen Screenshot von Info --> System bzw. Einstellungen --> Inverter. Weil laut deinem Konsolen Auszug wird nichtmal was gesendet.

deltahq, 2023-05-17:
<html><body>
<div class="card"><div class="card-header text-bg-primary">Firmwareinformationen</div><div class="card-body"><div class="table-responsive">

Hostname | OpenDTU-F5ACE0
-- | --
v4.4.4
0.1.25
v23.5.9
generic
Aktuell!
Software reset CPU
Software reset CPU
18
0 days 00:28:02

</div></div></div>
</body>
</html>Firmwareinformationen
Hostname	OpenDTU-F5ACE0
SDK-Version	v4.4.4
Konfigurationsversion	0.1.25
Firmwareversion / git Hash	[v23.5.9](https://github.com/tbnobody/OpenDTU/releases/tag/v23.5.9)
PIO Umgebung	generic
Firmware-Aktualisierung	Aktuell!
Reset Grund CPU 0	Software reset CPU
Reset Grund CPU 1	Software reset CPU
Anzahl der Konfigurationsspeicherungen	18
Betriebszeit	0 days 00:28:02
Hardwareinformationen
Chip-Modell	ESP32-D0WDQ5
Chip-Revision	1
Chip-Kerne	2
CPU-Frequenz	240 MHz
Speicherinformationen
Typ	Verwendung	Frei	Benutzt	Größe
Heap	
42 %
	166 kB	118 kB	284 kB
LittleFs	
8 %
	176 kB	16 kB	192 kB
Sketch	
77 %
	443 kB	1.477 kB	1.920 kB
Funkmodulinformationen
nRF24 Status	konfiguriert
nRF24 Chip-Status	verbunden
nRF24 Chip-Type	nRF24L01+
CMT2300A Status	nicht konfiguriert
CMT2300A Chip-Status

tbnobody, 2023-05-17:
Du darfst auch gerne direkt screenshots hier posten. Einfach wie gewohnt erstellen (z.B. mit Windows+Shift+S) und einfach in das Kommentar Fenster hier mit Strg+V einfügen.

deltahq, 2023-05-17:
Bin ich doof? Oder kann man hier keine Screenshots posten?
Der Typ des Wechselrichters wird richtig erkannt als HMS -2000

tbnobody, 2023-05-17:
Doch man kann screenshots posten... einfach den Screenshot der in der Zwischhenablage ist in das eingabefeld mit Strg+V einfügen... Dort wird dan ein Upload text usw. angezeigt... wenn du auf Preview gehst siehst du dann das Bild.... Wie sieht denn die Live View aus?

deltahq, 2023-05-17:
Es kommt immer die Fehlermeldung:" Something went wrong"

tbnobody, 2023-05-17:
Bitte alle Fragen beantworten:
* Siehst du Fehlermeldungen in der Live View? Sieht z.B. so aus (können auch andere Fehlermeldungen sein)
[image]

* Ist beim Inverter eingestellt, das er auch Nachts abgefragt werden soll? (In den Inverter Settings)
* Was ist der Status von NTP unter Info --> Ntp? Steht dort die aktuelle Uhrzeit?

deltahq, 2023-05-17:
Die Live Ansicht, sieht genau so aus wie bei meinem HM 1500, nur ohne Daten. Es wird nicht grün, sondern bleibt gelb. Und danke für deine Geduld und Nachsicht. Ich habe keine Ahnung warum ich die Scrennshots nicht uploaden kann

deltahq, 2023-05-17:
Ja, Zeitserver funktioniert!

deltahq, 2023-05-17:
Es gibt keine Fehlermeldungen

tbnobody, 2023-05-17:
Ich sehe gerade, du hast nur ein NRF Modul verbunden. Das kann mit dem HMS-2000 nicht sprechen. Dafür wird ein CMT2300A Funkmodul benötigt.

deltahq, 2023-05-17:
Ok Danke! Das habe ich leider nirgendwo gelesen. Nochmals vielen Dank für deine Mühe!

closed 2023-05-17 by deltahq

tbnobody, 2023-05-17:
Direkt auf der Startseite in der Readme steht bei jedem Unterstützten Inverter dabei welches RF Modul benötigt wird: https://github.com/tbnobody/OpenDTU#currently-supported-inverters

deltahq, 2023-05-17:
Ich habe beim Softwareupdate nur gelesen, dass die DTU nun auch die HMS Serie unterstützt. Sorry nochmal, war mein Fehler. Vielen Dank für deine schnelle und kompetente Hilfe. Lg