tbnobody/OpenDTU #933: Display sleep

webwizzard1405, 2023-05-19:
### What happened?

I operate an inverter via an accumulator at night. I switch off the power of the inverter at 6.00 a.m.. Open DTU can therefore no longer reach the inverter at 6.00 am. However, the display continues to show me the last status and does not go into sleep mode. (Display off). Can you please install the display going off when the inverter is not accessible. I currently have to reboot the Open DTU every morning so that the display stays off during the day.

### To Reproduce Bug

Switch of the power of the inverter

### Expected Behavior

Display go not in sleep mode

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.5.9

### Relevant log/trace output

```Shell
No log
```

### Anything else?

Nothing

tbnobody, 2023-05-19:
Did you enable the power save feature in the settings? Or did you enter multiple inverters? 
The source says, that the display should go into power save mode if either the total produced power equals zero or no inverter is reachable.

https://github.com/tbnobody/OpenDTU/blob/29e9da912642d65cbee1a7cc86b2cddbe1fb9cdb/src/Display_Graphic.cpp#L142-L161

webwizzard1405, 2023-05-19:
Yes, I enable the power save Feature in the settings and there is only one inverter. 
And yes the totalpower is never zero because i turn the AC power  of inverter off.

tbnobody, 2023-05-19:
I think it would be better to turn off the display 5minutes (configureable) after the last inverter is no more reachable. Because currently the display would also turn off if the AC side is disconnected but the inverter is still reachable.

commit 40820c07 (2023-05-22): Use the new data store in the graphic display class.

With this patch, the display turns off if no inverter is reachable anymore. Previously it turned off if the production was zero.
This fixes #933