tbnobody/OpenDTU #938: [Request] Bei mehreren Wechselrichtern die Reihenfolge im JSON bestimmen und fixieren

kixm, 2023-05-20:
### Is your feature request related to a problem? Please describe.

Hallo,

ich habe das problem in einem anderen Projekt schon beschrieben, man kann und sollte es aber vll. auch hier lösen. Hier mein Text von https://github.com/Selbstbau-PV/Selbstbau-PV-Hoymiles-nulleinspeisung-mit-OpenDTU-und-Shelly3EM/issues/14

"
ich habe 2 Wechselrichter im OpenDTU hinterlegt. Gesteuert werden soll nur der große WR. Soweit so gut.

Das Nulleinspeisungs-Skript hat aber dann einmal funktioniert und dann nach ein paar Tagen wieder nicht mehr. Ich habe jetzt einiges ausprobiert und in den LOGS gesucht und bin nun auf den Fehler gekommen, warum das Skript immer den Minimum Wert verwendet hat ihn aber nicht mehr ändern konnte. Der Output sagte "Fehler beim Abrufen der Daten von openDTU". Ich hab mir dann den Output von http://192.168.xxx.yyy/api/livedata/status/inverters angesehen und gemerkt, dass das Skript immer die Daten des ersten opendtu Inverters im JSON abfragt. Dadurch bekam das File immer den Wert des kleineren WR und konnte dort auch den höheren Watt-Wert nicht mehr eintragen. Evtl. kann man das Skript so anpassen, dass es auf Basis der WR-ID das richtige JSON Element ausliest und dort den Wert einträgt. Aktuell scheint es mir als ob die Reihenfolge im JSON evtl. durch die Reihenfolge des Einschaltzeitpunktes definiert und das abundzu wechselt.

Alternativ habe ich schnell probiert den kleinen WR zu entfernen und wieder hinzuzufügen. Evtl. hilft auch diese Vorgehensweise bei der Reihung. Muss das die nächsten Tag noch beobachten.

LG

### Describe the solution you'd like

Bei mehreren Wechselrichtern sollten diese im JSON in einer konfigurierbaren und dann fixen Reihenfolge ausgegeben werden, dann kann ein Skript, wie jenes der Nulleinspeisung auch die Daten vom richtigen Inverter abfragen. 

So nimmt es immer die Daten des WR1 im JSON API Call. Und hier wechselt anscheinend die Reihenfolge, je nachdem, welcher WR als erster von openDTU entdeckt wird? oder liegt es an der Seriennummersortierung?

schnello, 2023-05-22:
Hallo.

Sollte das Script nicht eher so umgebaut werden daß es den jeweiligen Inverter verwendet?

kixm, 2023-05-22:
Hallo, ja mittlerweile habe ich das Skript angepasst und es scheint zu funktionieren. Habe es auf der anderen GitHub Seite schon vermerkt.
Trotzdem denke ich, dass diese erratische Art der WR Sortierung verändert werden sollte. Aktuell sehe ich auf der Opendtu Startseite auch immer den WR auf Platz 1 der mich am Wenigsten interessiert. 

LG

schnello, 2023-05-22:
Json verwendet man halt um eben direkt auf Objekte zugreifen zu können. Ein Objekt ist die Inverter SN. Es macht einfach keinen Sinn JSON zu sortieren. Oder besser gesagt das ist eben Aufgabe vom Client für sich sie Daten richtig zu interpretieren. Startseite ist aber ein Punkt für dich.

Lg

homeautomation2022, 2023-05-23:
Den Wunsch der freien Sortierung (am besten dann überall gleich, also API und Webif, das ist auch am leichtesten umzusetzen) hatte ich im Discord auch schon eingebracht: https://discord.com/channels/984173303147155506/1029761153615790161/1094628132264349716
Mein Quick Fix dafür bei mir ist erstmal, für Rückwärtssortierung im Webif/API (war bei mir zufällig genau das was ich wollte an Reihenfolge):
https://github.com/tbnobody/OpenDTU/blob/master/src/WebApi_ws_live.cpp#L95
`for (int8_t i = Hoymiles.getNumInverters() - 1; i >= 0; i--) {`

Aktuell wird überall nach Seriennummer aufsteigend sortiert, wahrscheinlich da schon so gespeichert wird.

tbnobody, 2023-05-31:
Es gibt mit 1c8bd8091ba504c4d7d2ce7ae8b00d6b6d1a897c eine zusätzliche property "order" im json output nachdem entsprechend mit z.B. `jq` sortiert werden kann. Die Live View verwendet diese Property um die Reihenfolge entsprechend zu bestimmen. Die Reihenfolge selbst wird unter "Einstellungen" --> "Wechselrichter" festgelegt.

closed 2023-06-02 by tbnobody