tbnobody/OpenDTU #949: Inverter name changes to cryptic signs, openDTU crashes

indie89, 2023-05-23:
### What happened?

I use openDTU with 3 Inverters on a ESP32 with a NRF+ Module with external antenna and sufficient power supply (>1A), yet no capacitor. I poll the data every 20s (used to be 5s but I already changed this in order to stop the instabilities). WIFI quality is 100%, I use a mesh-repeater with 20MHz located 50cm away from the DTU.
In the background, I use a iobroker that runs the openDTU Adapter using websocket connections (mqtt is turned off in openDTU) with setting temporary limits every 20 seconds. 

All 1-5 days, the DTU crashes. Part of the crash is a change of one (or more) of the inverter's names, so that it includes cryptic signs (see screenshot). Aftwerwards the openDTU Adapter in iobroker then closes and reopens a websocket connection every ten seconds (unfortunately I can't tell which is first: the websocket loop or the change of inverter's name in openDTU). After some time this leads to a crash of the openDTU adapter in iobroker and normally to a state of the DTU itself, that it is not reachable over Wifi anymore and needs to be hardware-reset (power off/on). Until the DTU finally crashes, polling from the inverters doesn't work anymore (or at least in very long sequences of 5-10 minutes).
[image]

### To Reproduce Bug

I cannot reproduce the bug and it doesn't connect to external factors (at least not in an to me obvious manner). Although I had the same problem with a completely different DTU that did not use an external antenna, so I built a new one, but the bug is not eliminated. I also tried it with just 2 inverters instead of 3, but same bug. I tried to set the poll intervall from 5s to 20s. I tried to change "setting limits" from 5-20 seconds. 

In the logfile I see relatively often "Nothing received, resend whole request" but regularly every inverter is updated within 60seconds.

What I ask myself: which event can lead to a change of name of the inverters? Overloaded? Under-Powered?

### Expected Behavior

I would expect, that the DTU works without crashes. It would be very important to me, for I regularly want to set limits to the inverters and later on want to include another inverter as a battery inverter in order to achieve a "0-Einspeisung". And with the current behaviour, I won't be able to trust the system.

Thanks for your help!!

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.5.22

tbnobody, 2023-05-23:
Thats a duplicate of #885

closed 2023-05-23 by tbnobody

indie89, 2023-05-23:
No, it's unfortunately not...
Same problem (cryptic signs), that ultimately leads to a crash, but no other connected DTU (neither ahoydtu nor opendtu) at all.

tbnobody, 2023-05-23:
Just look at the original error in #885. the following error (two dtu) had nothing todo with the original post.

indie89, 2023-05-23:
Hi, sorry, I'm obviously "standing on the Schlauch ;-)". I see the exact same error but the problem that caused the error (a second dtu connected) is not aplicable to my error. Was there another solution to the error than disconnecting the second DTU? If yes, I didn't see it. How can I solve the problem that causes the error in my case? Pls give me a quick hint... thanks :-)

homeautomation2022, 2023-05-23:
Better connection to the inverters, you have already tried the Poll interval higher, that would also be my only idea if no other DTU is active in parallel.

tbnobody, 2023-05-23:
> Was there another solution to the error than disconnecting the second DTU? 

The issue is still open... No solution till now...