tbnobody/OpenDTU #955: [Request]

Lineflyer, 2023-05-24:
### Is your feature request related to a problem? Please describe.

As the SH1106 display is relatively small, its rather hard to read from a bit of distance. 

### Describe the solution you'd like

I would love to see the option, that the display content can be customized from the UI to e.g. only one or several of a) Total Power, b) Total Yield, c) Total Yield Day and scale accordingly.

### Describe alternatives you've considered

Place the display closer to my desk ;)

### Additional context

I crawled existing issues as I assumes, such request would already exist. I did however not find anything related there AFAICS.

agido-hundt, 2023-05-25:
+1 for this request - i also have a quite high "default distance" to the display.

It would be nice to just display a single value in the maximum size the display "supports".

BigT71, 2023-05-27:
+1 like in #751described, the temperature as a additional display value/option would be great

JenswBE, 2023-06-09:
Since it's related to display content, I'm adding my request here as well. But feel free to ask for a separate issue if you feel it's not related enough.

### Problem

I live in an apartment building with multiple owners. Therefore, we have different solar inverters (1 or more by owner). Since the DTU is located in my apartment, I'm mainly interested to see my own inverter on the display instead of the global totals.

### Solution

Would be nice if instead of just showing the global total, it's possible to select one or more inverters to show on the display.

goanes, 2024-05-18:
+1
I would to see also other information that there are on home assistant. 
I think it is possible to publish something with mqtt maybe?
I saw a projet like (Awtrix)  but on another type of display

stefan123t, 2024-07-10:
@Lineflyer nice feature but we would need to have a template language to specify the display contents in more details on a per user basis. I.e. both some styling for the Font size and location as well as variables from the total / inverter stats.

We could use a display template language like OpenEPaperLink which is JSON based:
https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Json-template

stefan123t, 2024-10-17:
The template parsing code is here https://github.com/OpenEPaperLink/OpenEPaperLink/blob/4c75234fa10371419aebd97f049bb53f7546559a/ESP32_AP-Flasher/src/contentmanager.cpp#L476-L517