tbnobody/OpenDTU #960: "Unknown model"

d-quixote, 2023-05-26:
### What happened?

"Inverter Info" says: "Unknown model! Please report the "Hardware Part Number" and model"

Hardware Part Number: 21692748
Model: HMS-1600

### To Reproduce Bug

connect a HMS-1600

### Expected Behavior

recognize hardware part number "21692748"

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.5.23

tbnobody, 2023-05-26:
Could you please post a screenshot of the dialog? This part number looks a little bit odd. Just want to  double check whether the other parameters where read corretly or if it's some kind of wrong data.

tbnobody, 2023-06-08:
Closing this issue due to inactivity. Feel free to open it again.