tbnobody/OpenDTU #965: No more mqtt - output after several weeks of non-stop traffic

dl7ata, 2023-05-27:
### What happened?

After several weeks there were no more output via mqtt. Dashboard looks good with reliable figures, but mqtt stopped.
After soft restart via dashboard, everything started as expected.
Running Tasmota 12.4.0.1 

### To Reproduce Bug

Don't know, will watch every day.

### Expected Behavior

mqtt should send EVERY time data

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

bf05dc8

tbnobody, 2023-05-27:
What showed Info --> MQTT at this moment? Did it show that is was still connected or did it show disconnected?
What was the status of the LWT topic?

dl7ata, 2023-05-28:
_MQTT_ at this moment:  I did not look at the MQTT on dashboard
_What was the status of the LWT topic?_ They were the last messages with valid values from 17h

tbnobody, 2023-05-31:
> They were the last messages with valid values from 17h

You write "They"... But there is only one LWT topic. By default its `solar/status`
[image]

MichaelEnders, 2023-06-01:
I think  I have seen the same issue: After a few days no more MQTT messages are sent. "MQTT Information"->"Runime Summary"->"Connection Status" shows "not connected". After restart of OpenDTU the connection is established again and MQTT messages are received.

tbnobody, 2023-06-03:
Your version is from Mar 17 ... On Mar 27 I upgraded the MQTT Library from 1.4.1 to 1.4.2. 
This fixed a reconnect error: https://github.com/bertmelis/espMqttClient/releases/tag/v1.4.2

So therefor I would suggest that you are updating your firmware.

dl7ata, 2023-06-03:
I just updated to
_SDK-Version | v4.4.4
0.1.25
v23.6.1_

I'll let you know, if I got that issue again - thanks a lot!

MichaelEnders, 2023-06-05:
Also after updateing to latest sources including Mqtt library 1.4.2 the reconnect problem still occurs.

hannesb, 2023-06-06:
I have the same problem. Updated yesterday to the latest working, today no more Mqtt updates after around 8:00. Web interface works fine.

tbnobody, 2023-06-06:
> I have the same problem.

And what shows the connection status?
What is the status of the LWT topic?
What shows the serial console?

Just saying "it does not work" will not help because at my several test installations I cannot reproduce this even after a longer uptime.

hannesb, 2023-06-06:
I'm have no access to the system right not as i'm not at home. I'll provide more information next weekend.

XuttuX, 2023-06-09:
I have the same problem MQTT does not send any data anymore. I have to rebbot the DTU to send data again.

Please find attached the relevant information.

DTU Firmware version: v23.6.1
LWT-Topic: solar/ dtu/status

[image]
[image]
[image]
[image]

Konsoleoutpt:
```Fetch inverter: 114482998398
14:07:23.456 > TX RealTimeRunData 865.00 MHz --> 15 82 99 83 98 80 14 69 68 80 0B 00 64 83 15 FB 00 00 00 00 00 00 00 00 B2 10 A0 
14:07:23.659 > Interrupt received
14:07:23.865 > RX 865.00 MHz --> 95 82 99 83 98 80 14 69 68 01 00 01 01 3B 01 3B 03 FC 03 FB 0C 90 0C 88 00 01 1E | -88 dBm
14:07:24.070 > Interrupt received
14:07:24.275 > RX 865.00 MHz --> 95 82 99 83 98 80 14 69 68 02 08 D5 00 01 08 74 06 3C 06 4D 08 D3 13 87 17 C3 48 | -88 dBm
14:07:24.479 > Interrupt received
14:07:24.787 > RX 865.00 MHz --> 95 82 99 83 98 80 14 69 68 83 00 05 01 0D 03 E7 01 D9 00 01 FA D3 9E | -88 dBm
14:07:24.992 > RX Period End
14:07:24.992 > Success
14:07:28.477 > Fetch inverter: 114482998398
14:07:28.678 > TX RealTimeRunData 865.00 MHz --> 15 82 99 83 98 80 14 69 68 80 0B 00 64 83 16 00 00 00 00 00 00 00 00 00 89 97 E4 
14:07:28.883 > Interrupt received
14:07:29.087 > Frame kaputt
14:07:29.292 > Interrupt received
14:07:29.497 > RX 865.00 MHz --> 95 82 99 83 98 80 14 69 68 02 08 D5 00 01 08 74 06 3C 06 4D 08 D2 13 87 17 E2 68 | -95 dBm
14:07:29.701 > Interrupt received
14:07:29.813 > Frame kaputt
14:07:29.820 > RX Period End
14:07:29.820 > Last missing
14:07:29.820 > Request retransmit: 3
14:07:29.820 > TX RequestFrame 865.00 MHz --> 15 82 99 83 98 80 14 69 68 83 03 
14:07:29.826 > RX Period End
14:07:29.826 > Last missing
14:07:29.826 > Request retransmit: 3
14:07:29.826 > TX RequestFrame 865.00 MHz --> 15 82 99 83 98 80 14 69 68 83 03 
14:07:29.833 > RX Period End
14:07:29.833 > Last missing
14:07:29.833 > Request retransmit: 3
14:07:29.833 > TX RequestFrame 865.00 MHz --> 15 82 99 83 98 80 14 69 68 83 03 
14:07:29.838 > Interrupt received
14:07:29.845 > Frame kaputt
14:07:29.851 > RX Period End
14:07:29.851 > Last missing
14:07:29.851 > Request retransmit: 3
14:07:29.851 > TX RequestFrame 865.00 MHz --> 15 82 99 83 98 80 14 69 68 83 03 
14:07:29.857 > Interrupt received
14:07:29.864 > Frame kaputt
14:07:29.872 > RX Period End
14:07:29.872 > Last missing
14:07:29.872 > Request retransmit: 3
14:07:29.872 > TX RequestFrame 865.00 MHz --> 15 82 99 83 98 80 14 69 68 83 03 
14:07:29.883 > Interrupt received
14:07:29.892 > Frame kaputt
14:07:29.921 > RX Period End
14:07:29.921 > Last missing
14:07:29.921 > Retransmit timeout
14:07:33.184 > Fetch inverter: 114482998398
14:07:33.217 > TX RealTimeRunData 865.00 MHz --> 15 82 99 83 98 80 14 69 68 80 0B 00 64 83 16 05 00 00 00 00 00 00 00 00 D9 A8 8E 
14:07:33.264 > Interrupt received
14:07:33.272 > RX 865.00 MHz --> 95 82 99 83 98 80 14 69 68 01 00 01 01 3B 01 3B 04 02 04 00 0C A0 0C 9A 00 01 39 | -91 dBm
14:07:33.314 > Interrupt received
14:07:33.321 > RX 865.00 MHz --> 95 82 99 83 98 80 14 69 68 02 08 D5 00 01 08 75 06 3C 06 4E 08 D2 13 86 17 E4 6D | -90 dBm
14:07:33.361 > Interrupt received
14:07:33.379 > RX 865.00 MHz --> 95 82 99 83 98 80 14 69 68 83 00 05 01 0E 03 E7 01 D9 00 01 46 10 E2 | -90 dBm
14:07:33.798 > RX Period End
14:07:33.798 > Success
```

tbnobody, 2023-06-09:
Is this the output of the serial console or just the web console? (The web console does not show everything). The MQTT status shows "getrennt". That means that it is correctly detected. But I am missing a "Connecting to MQTT..." in your logs. This should occour every time if the connection is not established.

It also seems that you are only using  a CMT module. That means you  should disable the NRF module. 

Next strange thing is that you get these "Frame kaputt" messages. This should never happen in a production environment. During the development it only occoured with bad connections and other strange hardware issues or if two DTU's sent at the same frequency etc.

XuttuX, 2023-06-09:
This is only the web console output. I have now updated to the newest version v23.6.9 and will see what happens tomorrow morning.

Yes I am only using the CMT module. Could you advise how to disable the NRF module? I have following pin-mapping:
```
[
    {
        "name": "Generic NodeMCU 38 pin with CMT2300A",
        "cmt": {
            "clk": 12,
            "cs": 27,
            "fcs": 26,
            "sdio": 14,
            "gpio2": -1,
            "gpio3": -1
        },
        "eth": {
            "enabled": false,
            "phy_addr": -1,
            "power": -1,
            "mdc": -1,
            "mdio": -1,
            "type": -1,
            "clk_mode": -1
        }
    }
]
```

Maybe I missed the correct setting somewhere.

joba-1, 2023-06-19:
I have the same problem.
Second time, with weeks in between.
Mqtt status says disconnected.
I rebooted the mqtt server ~4 days ago. Automatic reconnect was no problem then.
A save of unchanged mqtt settings reconnects. No reboot required.

HeinKlocktein, 2023-06-26:
Im letzten Bild vom FHEM Server sieht man schön das um 17.39 der OpenDTU (blaue Linie) das letzte MQTT Signal gesendet hat.
Ein anderes Gerät (esp8266, rote Linie) wird weiterhin empfangen.    
Speicher Button (auf der MQTT Konf. Seite) reicht aus, das es wieder geht. 
-->Neustart natürlich auch. 
  
[condole.txt](https://github.com/tbnobody/OpenDTU/files/11872214/condole.txt)
[image]
[image]
[image]
[image]
[image]

tbnobody, 2023-06-26:
Good to know is that it is detecting it's disconnected state. In this moment you should see "Disconnected from MQTT. Disconnect reason: XXXXXX" in the konsole. For every reconnect attempt you should see ""Connecting to MQTT...""

HeinKlocktein, 2023-06-26:
So I have to wait and keep the console always on?

tbnobody, 2023-06-26:
To be honest, I don't know how to debug it otherwise. The best method would be if you could use the serial console (not the web konsole) as it would also show several critical runtime errors. But the program logic does a reconnect every time it receives a disconnect (which occours periodically if no connection can be established). I currently have no clue what happens in your case as the reconnect does not work. 

I checked several possibilities here (killing the broker, stopping the boker, just removing the network cable from the broker) and it reconnected without any issues.

HeinKlocktein, 2023-06-26:
I am not a professional so your help is very welcome. 
So I plugt it into the USB port, open Kitty (like Putty) and start the serial logging. 
Ok, the sun is down for today :-)

tbnobody, 2023-07-03:
Please see https://github.com/bertmelis/espMqttClient/issues/102

dl7ata, 2023-07-12:
I just compiled and updated with some trouble:
No live data. Looks like, that I am not able to configure the radio.

Device manager:
nRF24 Status | not configured

In tab DTU Settings is only "Serial" input
Tab device manager looks strange - no Param input possible:

[image]

All other param looks good.
Any idea what went wrong?

tbnobody, 2023-07-12:
You used the generic_esp32 environment. This requires a Device Profile (See wiki).
You can also change the environment to generic (which is deprecated btw) and you can compile the source as you did previously

dl7ata, 2023-07-13:
I changed to _default_envs = generic_

> Compiling .pio/build/generic/src/WebApi_device.cpp.o
src/MqttHandleHass.cpp: In member function 'void MqttHandleHassClass::createDeviceInfo(ArduinoJson::V6212PB::JsonObject&, std::shared_ptr<InverterAbstract>)':
src/MqttHandleHass.cpp:264:20: error: 'AUTO_GIT_HASH' was not declared in this scope
     object["sw"] = AUTO_GIT_HASH;
                    ^~~~~~~~~~~~~
*** [.pio/build/generic/src/MqttHandleHass.cpp.o] Error 1

Any other place to change configuration for ESP32?

tbnobody, 2023-07-17:
Going to close this issue asa from my standpoint it should be fixed. Feel free to reopen or open a new one.

closed 2023-07-17 by tbnobody