tbnobody/OpenDTU #972: Display direction toggle

RKeme, 2023-05-29:
### What happened?

Version 23.5.23
Display SH1106
Direction toggle of Display 0/ 180 on repowering

### To Reproduce Bug

Set display to rotate 180 degrees, power off.
Repower (at my site I use openDTU quite randomly).
After reconnecting power display rotation looks like 0 degree.
Set configuration setting to zero degree to get correct rotation.
Switch power off -> switch power on -> need to set configuration setting to 180 degrees to get correct rotation again.

### Expected Behavior

Take configuration from settings for rotation.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

V23.5.23

tbnobody, 2023-05-29:
I've doublechecked it several times now. It works as expected. No need to setup any rotation after any re-powering... the settings are applied as expected.

RKeme, 2023-05-30:
Thx. for checking.
I just had a Wifi outage and afterwards rotation was wrong again.
This is Setting now for 180 degrees After my change and Save
display":{"powersafe":true,"screensaver":true,"rotation":2,"contrast":60,"language":1}},

tbnobody, 2023-05-30:
It sounds like your power supply is too bad. Because a wifi outage has nothing todo with a rotation of the display. But if the display looses it's power during the operation the rotation will not be set (it will only set if changed or at startup)

RKeme, 2023-05-30:
I am happy you are right; I took another power supply and can‘t reproduce the behaviour any more.
Tested this now at least ten times and the toggle of the rotation has gone.
Trank you for sharing this project and for your fast support.