tbnobody/OpenDTU #979: [Request] do not clear Data at evenng if the inverter goes offline

AloisKlingler, 2023-05-29:
### Is your feature request related to a problem? Please describe.

I am fetching statistics by http from opendtu.
currently, if the inverter goes offline at night, opendtu zeroes out all values (also YieldDay and YieldTotal) which messes up everything.

`{"inverters":[{"serial":"138284652162","name":"Gartenhaus","data_age":1973,"poll_enabled":true,"reachable":false,"producing":false,"limit_relative":100,"limit_absolute":2250,"AC":{"0":{"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":0,"u":"V","d":1},"Current":{"v":0,"u":"A","d":2},"Power DC":{"v":0,"u":"W","d":1},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3},"Frequency":{"v":0,"u":"Hz","d":2},"PowerFactor":{"v":0,"u":"","d":3},"ReactivePower":{"v":0,"u":"var","d":1},"Efficiency":{"v":0,"u":"%","d":3}}},"DC":{"0":{"name":{"u":""},"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":230.3999939,"u":"V","d":1},"Current":{"v":204.8399963,"u":"A","d":2},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3}},"1":{"name":{"u":""},"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":230.3999939,"u":"V","d":1},"Current":{"v":0,"u":"A","d":2},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3}},"2":{"name":{"u":""},"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":0,"u":"V","d":1},"Current":{"v":0,"u":"A","d":2},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3}},"3":{"name":{"u":""},"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":0,"u":"V","d":1},"Current":{"v":0,"u":"A","d":2},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3}},"4":{"name":{"u":""},"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":0,"u":"V","d":1},"Current":{"v":0,"u":"A","d":2},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3}},"5":{"name":{"u":""},"Power":{"v":0,"u":"W","d":1},"Voltage":{"v":0,"u":"V","d":1},"Current":{"v":0,"u":"A","d":2},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3}}},"INV":{"0":{"Temperature":{"v":0,"u":"°C","d":1}}},"events":5}],"total":{"Power":{"v":0,"u":"W","d":1},"YieldDay":{"v":0,"u":"Wh","d":0},"YieldTotal":{"v":0,"u":"kWh","d":3}},"hints":{"time_sync":false,"radio_problem":false,"default_password":false}}`

### Describe the solution you'd like

Would be great if opendtu could keep the data (ofc NOT reboot-persistent)

tbnobody, 2023-05-29:
OpenDTU does not zero out anything. I still see my previous values now that it's dark outside. OpenDTU shows exactly what the inverter responds to a data request. It could happen, that after it's quite dark outside, the inverter comes back online for a short period of time, and sends new (but resettet) values. Then this values stay in the buffer until new values arrive.

AloisKlingler, 2023-05-30:
But next morning when the inverter is back online statistics are back again - so i assume it does not send empty values in the evening.

Can this be caused by a timeout or "half" statistics received during it is switching off?
I have to admit that I am live data addicted and therefore have set the inverter poll time to 3 seconds.

lemmi, 2023-06-04:
I have the same problem. What seems to happen is that in the evening the inverter turns down power generation and keeps answering for a while. In the last moment, shortly before not enough power is generated from the solar panels, my inverter often restarts once or twice and only then shuts down. The inverter will zero statistics on a restart for a brief moment, but there doesn't seem to be enough power available to fully initialize again, so the last entries are more or less useless.

For now, I circumvent the issue by setting the `Sunset type` to `Standard Dawn (90.8°)` and stop polling the inverter at night time.

AloisKlingler, 2023-06-05:
thanks @lemmi for this workaround-idea - is working for me as well and helps me a lot. :-)

tbnobody, 2024-01-29:
This issue can be avoided by enabling the [Yield Day Correction](https://tbnobody.github.io/OpenDTU-docs/firmware/configuration/inverter_settings/#yield-day-correction)

closed 2024-01-29 by tbnobody