tbnobody/OpenDTU #980: [Request] add dBi between openDTU and inverter

AloisKlingler, 2023-05-29:
### Is your feature request related to a problem? Please describe.

currently only in console log it is viewable with which signal strength the inverter is reachable.

### Describe the solution you'd like

add anywhere (like in ework is for wifi signal) the last dBi to inverter and if possible give hints if the signal is good, weak, ...

tbnobody, 2023-05-29:
Ah you mean for the CMT2300A module and not for the NRF24 module I assume? Because the NRF24 module does not support such information.

AloisKlingler, 2023-05-30:
Exactly. 👍💪

Brain-Kiel, 2023-05-30:
@tbnobody: 
Some time ago I used this [tmrh20 library](https://nrf24.github.io/RF24/index.html) to read out a chinese "power meter".
The library contains a bool to detect good or week signal qualtiy for nRF24L01+ modules:

[image]

BR, Brain-Kiel

AloisKlingler, 2023-06-01:
an intermediate question - I can see (with CMT2300A and HMT-2250) -61dBm working not always stable - not always means, about once a day the connection suddenly stops and I cannot get it back online again until changing the place of the antenna a bit.
now I changed the place completely - one window and about 4m distance after the window - I now get -64 / -65 dBm. I recognized, I need there already at least +1dBi in transmitting power to get a connection.

is there any experience with which dBm the connection should be stable (and I need then to search the issue elsewhere)?

on my wished position I think it does not work at all - in basement near to a window (trough the window and the light-well there would be also about 6m distance to the inverter). with "transmitting power" set to +20 I get a connection with -84dBm  which is most probably more unstable than stable.

any comments and especially experience are appreciated.

thanks :-)

SuperNinja73, 2024-03-04:
as asked in this topic:
 https://github.com/tbnobody/OpenDTU/discussions/1580#discussioncomment-7898167
Because sometimes, I too have connection problems > 5s with the HMS1000 inverter, I would like to have the return of the RSSI of the inverter thanks to the CMT2300A (which is capable of doing it unlike the nRf) to debug stability problems or disconnection of the connection.
[image]
[image]

_In blue the production published by openDTU HMS1000 2x 435wc: we see a "flat" during which the power has not changed: loss of communication!?
In red dots, the curve as it should have been.
In purple another 1x 180wc photovoltaic panel next to the 2 others with another production monitoring system (WVC)._
@tbnobody 
We only have the Wifi RSSI "Info--> Network information" which is moderately useful. Please add the publication of the inverter RSSI from the CMT2300A.

tbnobody, 2024-10-05:
Implemented in edfe06e31eae286fbb88da69e9b65420cfa8628a

closed 2024-10-05 by tbnobody