tbnobody/OpenDTU #990: hm-400 no power to the grid

kthemall, 2023-06-02:
### What happened?

the hoymiles hm-400 does not deliver power to the grid. maybe the logs tell someone what the problem is?? 
systeminfo:

Hostname | OpenDTU1
-- | --
v4.4.4
0.1.25
v23.6.1
generic
Aktuell!
Software reset CPU
Software reset CPU
42
0 days 00:04:39

thx tom

### To Reproduce Bug

does not work either.

### Expected Behavior

if i send a power command it should raise or degrade the power of the inverter correctly.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.1

### Relevant log/trace output

```Shell
Websocket: [/livedata][2] disconnect
11:44:15.090 > Fetch inverter: 112183814498
11:44:15.154 > TX RealTimeRunData Channel: 40 --> 15 83 81 44 98 80 11 20 52 80 0B 00 64 79 B9 EF 00 00 00 00 00 00 00 00 D4 0B 37 
11:44:15.207 > Interrupt received
11:44:15.276 > RX Channel: 3 --> 95 83 81 44 98 83 81 44 98 01 00 01 01 07 00 29 00 6B 00 00 A9 0C 0F 3E 09 23 6F 
11:44:15.655 > RX Period End
11:44:15.655 > Last missing
11:44:15.655 > Request retransmit: 2
11:44:15.655 > TX RequestFrame Channel: 61 --> 15 83 81 44 98 80 11 20 52 82 AA 
11:44:15.972 > Interrupt received
11:44:16.021 > RX Channel: 23 --> 95 83 81 44 98 83 81 44 98 82 13 8A 00 00 00 00 00 00 00 00 00 F9 00 06 5D 1E 32 
11:44:16.071 > RX Period End
11:44:16.071 > Success
11:44:16.126 > TX AlarmData Channel: 75 --> 15 83 81 44 98 80 11 20 52 80 11 00 64 79 B9 EF 00 00 00 00 00 00 00 00 0E 10 EC 
11:44:16.206 > Interrupt received
11:44:16.263 > RX Channel: 3 --> 95 83 81 44 98 83 81 44 98 84 ED 43 40 94 00 06 43 F6 86 22 0D 8D 00 5A 77 5C 8D 
11:44:16.591 > RX Period End
11:44:16.591 > Middle missing
11:44:16.591 > Request retransmit: 1
11:44:16.591 > TX RequestFrame Channel: 3 --> 15 83 81 44 98 80 11 20 52 81 A9 
11:44:16.714 > RX Period End
11:44:16.715 > Middle missing
11:44:16.715 > Request retransmit: 1
11:44:16.715 > TX RequestFrame Channel: 23 --> 15 83 81 44 98 80 11 20 52 81 A9 
11:44:16.821 > RX Period End
11:44:16.821 > Middle missing
11:44:16.821 > Request retransmit: 1
11:44:16.821 > TX RequestFrame Channel: 40 --> 15 83 81 44 98 80 11 20 52 81 A9 
11:44:16.873 > Interrupt received
11:44:16.941 > RX Channel: 75 --> 95 83 81 44 98 83 81 44 98 01 00 01 80 01 00 01 88 AE 88 AE 00 00 00 00 B0 0C A9 
11:44:17.011 > RX Period End
11:44:17.011 > Middle missing
11:44:17.011 > Request retransmit: 2
11:44:17.011 > TX RequestFrame Channel: 61 --> 15 83 81 44 98 80 11 20 52 82 AA 
11:44:17.077 > Interrupt received
11:44:17.200 > RX Channel: 75 --> 95 83 81 44 98 83 81 44 98 02 00 02 37 A4 37 A4 0A A3 F9 B0 B0 2E 00 03 72 AD 37 
11:44:17.256 > RX Period End
11:44:17.256 > Middle missing
11:44:17.256 > Request retransmit: 3
11:44:17.256 > TX RequestFrame Channel: 75 --> 15 83 81 44 98 80 11 20 52 83 AB 
11:44:17.315 > Interrupt received
11:44:17.511 > RX Channel: 23 --> 95 83 81 44 98 83 81 44 98 03 72 AD 0A AC 01 BE 80 24 00 04 43 F5 43 F5 07 F9 0E 
11:44:17.574 > RX Period End
11:44:17.574 > Success
```

### Anything else?

no thanks

DPO99, 2023-06-02:
Please share a screenshot from the live data view.

kthemall, 2023-06-02:
[image]

tbnobody, 2023-06-02:
Only 1,3W on String 1 seems to be much to low to start producing....

DPO99, 2023-06-02:
Same here iff the inverter is turned off. Can you try to turn it on?

[image]

[image]

tbnobody, 2023-06-02:
If you show us a image of the event log we would see if the inverter is turned off

kthemall, 2023-06-02:
[image]
[image]

hi, hope you can see something strange in the pics. thx tom

kthemall, 2023-06-02:
sorry forgot to mention, i use the inverter on a 8s lifepo battery.

tictrick, 2023-06-02:
How big is the battery?

Can it be that the voltage of the battery drops and the BMS switches off as a result? The inverter may then first need a reboot so that it can produce again.

tbnobody, 2023-06-03:
I would convert this to a discussion since, as of now, it is not an OpenDTU issue.