tbnobody/OpenDTU #992: MQTT value of solar/ac/power is jumping from time to time to 0

RyHoRuK, 2023-06-02:
### What happened?

I am using OpenDTU and one HM-800 MicroInverter connected with MQTT with Home Assistant. Everything seems to be fine until I only use the values of "p_ac" the AC value of the HM-800. When I look at the values of "solar/ac/power" I have always jumps to a 0 value which is not correct because the HM-800 MicroInverter has all the time a valid value.
I use only 1 MicroInverter and MQTT is configured to publish every 5sec.

[image]
[image]

### To Reproduce Bug

I can't reproduce it manually but I can see this behavior every day in Home Assistant und in an MQTT-Explorer.

### Expected Behavior

the wrong jumps to the value 0 should never happen.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.5.23

tbnobody, 2023-06-02:
This is absolutly right and should already be fixed in v23.5.31
Please update your version and if the error still occours feel free to open this issue again

RyHoRuK, 2023-06-02:
I updated to 23.6.1 Problem is still there.
Should I downgrade to 23.5.31?

RyHoRuK, 2023-06-02:
Actual Screenshot from 5min ago with Version 23.6.1
[image]

tbnobody, 2023-06-02:
Can you create three diagrams using mqtt explorer showing 0/power, ac/power and ac/is_valid?

RyHoRuK, 2023-06-02:
sorry I was offline...
[image]
[image]
[image]

tbnobody, 2023-06-02:
Are you running ahoy in parallel for the same inverter?

RyHoRuK, 2023-06-02:
Yes. But there is not this problem.

tbnobody, 2023-06-02:
How do you know this? It is exactly the problem! Look at your last screenshot. The power for the individual inverter drops also to zero, not only the total value in the `ac` topic. That is exactly the behavior if you are running two DTUs. 
Don't know what HASS is doing, I just look at the raw MQTT values.

RonnieLast, 2023-06-02:
I'm seeing what I think is the same problem but with a different net effect.
My energy graphs are showing massive increases during times when the microinverters are having connection problems, typically at the end of the solar day. The panels delivering around ~10Wh of power in a 5min period, but the HA recorder is seeing many multiples of that. I

Looking at the data in HA recorder, I see the following incorrect data:

[image]

When I look into the HA logs I see that the sensor is cycling due to what I believe is the lost connection and I think that HA is summing the energy utility meter value due to the cycling (877Wh -> 0Wh -> 883Wh -> 0Wh...).

```
9614:2023-06-02 16:00:10.512 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.hm350_yieldday, value dropped from 877.0 to 0.0, triggered by state with last_updated set to 2023-06-02T15:56:29.701817+00:00
9615:2023-06-02 16:00:10.512 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.hm350_yieldday, value dropped from 883.0 to 0.0, triggered by state with last_updated set to 2023-06-02T15:58:09.604525+00:00
9616:2023-06-02 16:00:10.512 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.hm350_yieldday, value dropped from 886.0 to 0.0, triggered by state with last_updated set to 2023-06-02T15:59:11.493230+00:00
9617:2023-06-02 16:00:10.512 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.hm350_yieldday, value dropped from 887.0 to 0.0, triggered by state with last_updated set to 2023-06-02T15:59:39.020766+00:00
9627:2023-06-02 16:05:10.507 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.hm350_yieldday, value dropped from 65531.0 to 891.0, triggered by state with last_updated set to 2023-06-02T16:00:41.159554+00:00
9628:2023-06-02 16:05:10.507 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.hm350_yieldday, value dropped from 896.0 to 0.0, triggered by state with last_updated set to 2023-06-02T16:02:21.065861+00:00
```
I updated to v23.6.1 this morning as I thought that issue #887 sounded like this problem, but it recurred again this afternoon.

tbnobody, 2023-06-02:
> I'm seeing what I think is the same problem but with a different net effect.

If it's also the same issue then disable your second DTU.

tbnobody, 2023-06-02:
> is it also necessary to stop Homilies DTULight?

The Inverter supports ONE (1) DTU.
https://github.com/tbnobody/OpenDTU/wiki/FAQ#q-opendtu-shows-wrongunrealistic-or-just-missing-values

RyHoRuK, 2023-06-02:
Thanks @tbnobody then the issue can be closed. Thanks for your help

closed 2023-06-02 by tbnobody

RonnieLast, 2023-06-03:
> > I'm seeing what I think is the same problem but with a different net effect.
> 
> If it's also the same issue then disable your second DTU.

I'm confirming this as resolved for me once I unplugged the Hoymiles DTU.