tbnobody/OpenDTU #997: Enable OpenDTU to automatically connect to the strongest AP in a network with multiple APs of similar SSID

juergen-weber, 2023-06-04:
### Is your feature request related to a problem? Please describe.

After extending the Wifi network with a bunch of repeaters (mesh mode, same SSID) the OpenDTU device won't reliably connect to the closest AP with the strongest signal.

### Describe the solution you'd like

This seems to be a known behaviour and requires extracode to handle it. An example can be found here:
https://www.esp32.com/viewtopic.php?t=18979

tbnobody, 2023-06-04:
What exact extra code do you mean? And what is the difference compared to 
https://github.com/tbnobody/OpenDTU/blob/5558dffa7077fa3b9d49c09e58993374b9c87153/src/NetworkSettings.cpp#L124-L125

which was introduced here: #605

But also keep in mind: #618

juergen-weber, 2023-06-07:
I have carefully reviewed #605 and #618. Together with my own experience with ESP32 devices in WiFi, I have to say that a change will probably never solve all problems.
Due to the non-uniformity of wifi installations (different manufacturers, repeaters with same/different SSID, mesh, different roaming methods, etc.) the complexity is so high that it would take a lab to systematically approach a general solution here.
My specific problem (OpenDTU only sporadically finds the AP with the highest field strength) only occurs when my AVM devices are in the mesh and publish the same SSID. For my own small esp32 projects in my network environment, I could only find one solution so far: https://www.esp32.com/viewtopic.php?t=18979#p71569.
I close this topic now and suggest to continue this topic in the "Discussions" section.

closed 2023-06-07 by juergen-weber