tbnobody/OpenDTU #999: [Request] MQTT - Add automatic reconnect/restart function

Guyver111, 2023-06-04:
### Is your feature request related to a problem? Please describe.

I set up a HomeAssistant System for the openDTU of my brother and hitting a snag.
The MQTT connection on the openDTU would be lost after X days and only return after i restart the DTU.

At first i thought it was the ongoing MQTT connection bug, but now i believe the problem is man made...
My brothers server, where the Homeassistant or rather the MQTT server is running on, gets shutdown over night, from 12PM to 7AM.

The next morning the server starts automatically, but it seems like the openDTU MQTT does not automatically reconnect to the MQTT server by itself.

### Describe the solution you'd like

My idea, which would also work as a workaround for the ongiong MQTT connection bug, would be to add funtions for:

1. Unplanned reconnect:
If connection lost, try to reconnect to MQTT server every XX minutes, till reconnect or till XX tries failed.

2. Timed/planned Reconnect:
Reconnect to MQTT server at XX:YY am/pm every day

tbnobody, 2023-06-04:
What opendtu version are you using?

DPO99, 2023-06-04:
I think you should investigate in this mqtt issue....i use opendtu with iobroker and it works for weeks without any connection issues.

tbnobody, 2023-06-04:
> I think you should investigate in this mqtt issue...

I fully agree. I would rather harden the reconnect handling (which works by the way without any problems in my setup, even if I stop, disconnect or kill the mqtt broker) then implementing a reconnect at a specific time.

Guyver111, 2023-06-05:
Hi,
i just updated openDTU on Monday last week, so when the problem persisted, when i checked yesterday, i directly went and opened this thread and totally missed the update... which naturally contains an update for mqtt....
Thats just my luck. Sorry

I updated openDTU, yesterday, from v23.5.23 to v23.6.1.
I checked and it looks like at the moment the connection seems to resume when the server gets turned on again.

I will keep an eye on it this week and keep you updated.

P.S.:
> ....this is not the sense/purpose of a server ;)
I know, but the energy prices went way up this year so it was decided to turn the box of over night (Not my decission)

Guyver111, 2023-06-14:
Hi,
i updated the DTU last friday to 23.6.9
i checked again today, because my brother notified me that HA doesn't show actual data.

It was rebooted on Saturday (Vbat power on reset and for APP CPU, reset by PRO CPU).
The uptime is now 3d 17h.

Going by HomeAssistant, the DTU MQTT stopped reconnecting 2 days ago.
the 2 days before that the MQTT reconnected always around 7:20am, after the Server was started, to HA again.

tbnobody, 2023-06-14:
* What showed the Info --> MQTT Page as connection status?
* What showed the Console?

MrZwickie, 2023-07-03:
Same thing here with HomeAssistant running on a VM in Virtual Box as a test for a later installation on a Raspi: If I stop the VM over night OpenDTU does not reconnect to HomeAssistant's MQTT-broker until I restart OpenDTU. My Tasmota power meter instead instantly reconnects after the VM is up again. 

MQTT-Page shows "disconnected" as connection status and console does not show anything in terms of MQTT - only the communiation with the Inverter.

tbnobody, 2023-07-03:
Could you try if you see anything in the console in the moment when you stop your VM? Maybe it's just a issue that the reconnect does not work if it can not reconnect over a long time.

MrZwickie, 2023-07-03:
> Maybe it's just a issue that the reconnect does not work if it can not reconnect over a long time.

I think so too: This is what console looks like if i pause the VM for a minute: 
```
[...]
09:39:35.881 > Disconnected from MQTT.
09:39:35.881 > Disconnect reason:TCP_DISCONNECTED
[...]
09:39:37.885 > Connecting to MQTT...
09:39:40.944 > Disconnected from MQTT.
09:39:40.944 > Disconnect reason:TCP_DISCONNECTED
[...]
09:39:45.946 > Disconnected from MQTT.
09:39:45.946 > Disconnect reason:TCP_DISCONNECTED
[...]
09:40:53.277 > Connecting to MQTT...
09:40:54.980 > Connected to MQTT.
[...]
```

It does not reconnect only if disconnected for a longer time (for example overnight).

tbnobody, 2023-07-03:
The messages you wrote, is exactly what I would have expected in a working environment. I will setup a test DTU and let it connect to a offline MQTT broker for multiple hours. During this time it should stop reconnecting... 
Then it would be interesting if it's a issue within the MQTT library or in the external reconnect handling in OpenDTU.

tbnobody, 2023-07-03:
Please see https://github.com/bertmelis/espMqttClient/issues/102

tbnobody, 2023-07-12:
please check version [v23.7.12](https://github.com/tbnobody/OpenDTU/releases/tag/v23.7.12) 
It fixes a reconnect error of the mqtt client.

tbnobody, 2023-07-17:
Going to close this issue asa from my standpoint it should be fixed. Feel free to reopen or open a new one.

closed 2023-07-17 by tbnobody