tbnobody/OpenDTU #1005: Hoymiles HMT channels in wrong order

harmstorf, 2023-06-06:
### What happened?

The strings/channels numeration of HMT-2250-6T is in the wrong order.

According to the Hoymiles manual and physical engravings on the inverter case, the numeration starts with one (1) from the top to six (6) at the bottom of the case.

OpenDTU reports the actual string/channel one (1) as string/channel six (6).

### To Reproduce Bug

Connect a panel to channel one (1) and read data via OpenDTU. It gets reported as channel six (6).

### Expected Behavior

The OpenDTU channel/string numeration should be identical to the inverter.

[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

5558dff

tbnobody, 2023-06-07:
Have you also tried to connect a panel to port 2? Is it shown as string 5? So is it really completely mirrored?

harmstorf, 2023-06-07:
Yes, completely mirrored.

Current (5558dff) -> Expected (as defined by Hoymiles)
1 -> 6
2 -> 5
3 -> 4
4 -> 3
5 -> 2
6 -> 1

closed 2023-06-08 by tbnobody

harmstorf, 2023-06-09:
Fix confirmed, release 170978c

gmke, 2023-06-09:
your fix completely ruined my time series DB data of the modules. I first thought of a new bug and flashed back to the old version, before I learned about this change. Could you please make it optional for historical data owners :-) I don't care about the numeration being identical to the label on the case.

gmke, 2023-06-09:
I fear, I don't have the skill to write a script to fix the DB. Would be the history DB in iobroker. Easiest solution seems to be to replug the cables into the old order.

homeautomation2022, 2023-06-11:
Bitte prüft das nochmal gegen und berichtet eure Wechselrichter Firmwareversion, eventuell hat Hoy das nur ab oder bis zu einem bestimmten FW-Stand verdreht.
Es wäre das erste Gerät, was nicht die Daten in der Reihenfolge übermittelt, wie die Ports beschriftet sind. Schon komisch aber möglich. Die Tatsache, das bisher noch keiner gesagt hat es wäre falsch herum macht mich stutzig, dass es für die meisten bis jetzt wohl richtig war. Testen kann man es ja einfach mit abziehen oder verschatten.

gmke, 2023-06-11:
also ich habe das jetzt nachgeschaut. Bei mir sind die Ports des HMT 2250TR.PL.HM nicht wie hier oben angeführt beschriftet sondern genau andersrum. Oben an der Antenne/Halterung ist Port 6 unten ist Port 1. 

Seriennummer | 138280425647
-- | --
Produktionsjahr | 2022
Produktionswoche | 4
Modell | Unbekanntes Modell! Bitte melden Sie die "Hardware Teilenummer" und das Modell (z.B. HM-350) hier als Problem.
Ermittelte max. Leistung | 0 W
Bootloader-Version | 0.0.0
Firmware-Version | 0.2.56
Firmware-Erstellungsdatum | 1911-09-09 11:52:32
Hardware-Teilenummer | 124097
Hardware-Version | 00.00

Seriennummer	
Produktionsjahr	2022
Produktionswoche	4
Modell	Unbekanntes Modell! Bitte melden Sie die "Hardware Teilenummer" und das Modell (z.B. HM-350) [hier](https://github.com/tbnobody/OpenDTU/issues) als Problem.
Ermittelte max. Leistung	0 W
Bootloader-Version	0.0.0
Firmware-Version	0.2.56
Firmware-Erstellungsdatum	1911-09-09 11:52:32
Hardware-Teilenummer	124097
Hardware-Version	00.00

Mit der alten OpenDTU Version vom Mai wurde alles richtig angezeigt! Jetzt ist es falsch und sogar nicht nur die Nummerierung auch bei Einzelwerten wie zb der Einstrahlung sind Vertauschungen. Mag sein dass es verschiedene Untermodelle gibt, für mich passte die Ursprungsversion.

gmke, 2023-06-11:
[image]

beschienen waren zu dem Zeitpunkt die SO Module die bei mir an Port 1-4 angeschlossen sind. Die SW Module lagen noch om Schatten und sind an Port 1-2. Dargestellt aber mit der verdrehten Nummerierung und anderen Vertauschungen.

homeautomation2022, 2023-06-11:
> also ich habe das jetzt nachgeschaut. Bei mir sind die Ports des HMT 2250TR.PL.HM nicht wie hier oben angeführt beschriftet sondern genau andersrum. Oben an der Antenne/Halterung ist Port 6 unten [...]

alter Bug: einfach nochmal OpenDTU rebooten, da gabs ein Fehler beim Auswerten des Paketes

harmstorf, 2023-06-12:
Produktionsjahr | 2022
Produktionswoche | 45
Modell | HMT-2250
Ermittelte max. Leistung | 2.250 W
Bootloader-Version | 0.0.1
Firmware-Version | 1.0.10
Firmware-Erstellungsdatum | 2021-09-17 15:00:00
Hardware-Teilenummer | 271790337
Hardware-Version | 01.00

Alles funktioniert und ist zugeordnet wie erwartet.

gmke, 2023-06-15:
eben die v23.6.14 aufgespielt und alles passt wieder.... Danke!

commit 593a3302 (2023-06-08): Fix: String numbers for HMT inverters are mirrored

Fixes: #1005