tbnobody/OpenDTU #1010: Probleme ab 7. Wechselrichter HM1500

acsmonck, 2023-06-07:
### What happened?

Bis zur Installation von 6 HM1500 funktionierte alles einwandfrei, beim 7. Wechselrichter trat dann folgendes Problem auf:
- Es werden nur noch 3 Eingänge angezeigt
- Fehler 209 String 1: no input
- Fehler 214 MPPT B String 3 & String 4 abnormal wiring
- Fehler 208 MPPT B Input under voltage

Die Verkabelung stimmt 100%ig, Panele funktionieren einwandfrei.
Der Wechselrichter wurde auf einen neuen ausgetauscht, auch beim neuen tritt derselbe Fehler auf. Ich weiß nicht inwieweit OpenDTU auf die Inverter schreibt, habe jedoch aufgrund des Fehlerbildes den Verdacht, dass OpenDTU den Inverter crashed.
Über die originale DTUpro werden die selben Fehler angezeigt und die selben Werte upgeloaded.
Gibt es eine Möglichkeit eines Factory Resets des Wechselrichters durchzuführen?

### To Reproduce Bug

.

### Expected Behavior

.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.5.23

tbnobody, 2023-06-07:
OpenDTU schreibt nichts in den Wechselrichter und kann auch keinen Factory Reset durchführen.

>  Über die originale DTUpro werden die selben Fehler angezeigt und die selben Werte upgeloaded.

Wenn man 2 DTUs am gleichen Wechselrichter betreibt treten unvorhergesehene Events auf. Das ist ein nicht unterstütztes Setup. Siehe: https://github.com/tbnobody/OpenDTU/wiki/FAQ#q-opendtu-shows-wrongunrealistic-or-just-missing-values

Aber du kannst ja mal testen wenn du in OpenDTU nur den einen Inverter Einträgst der nicht funktioniert. Dann ist dies quasi wieder ein Erster.

acsmonck, 2023-07-26:
Have you checked the polarity of the panels to input 3 & 4?

tbnobody, 2023-07-26:
> The other one shows only strings 1,2,5 and 6. Event log mentions input 3 and 4 abnormal wiring.

The eventlog is generated by the inverter not by the DTU. If the inverter says abnormal wiring that means it's really something related to the inverter.

sharkman80, 2023-07-26:
My case is solved: wiring indeed was ”abnormal”: male and female mc4 connectors in the extension cables of one panel had been installed incorrectly, hence the polarity was wrong.

tbnobody, 2024-01-26:
is this issue still relevant or can it be closed?

closed 2024-01-29 by tbnobody