tbnobody/OpenDTU #1017: MQTT funkrioniert nach Update auf v23.6.9 nicht mehr

EvilBlood, 2023-06-09:
### What happened?

Nach dem Update auf die FW. v23.6.9 nciht mehr. 
In der Config ist es aktiviert aber es sendet keine Daten mehr.
MQTT Settings einmal neu Speichern behebt das Problem. 

### To Reproduce Bug

Update auf FW v23.6.9 ausführen

### Expected Behavior

MQTT sollte nach Update einfach weiter daten senden. 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.9

### Anything else?

Githash wird auch nicht mehr im Webui angezeigt.

tbnobody, 2023-06-09:
Mach doch mal ein paar screenshots. Von der Stelle mit dem  Git Hash und von Info --> MQTT. Klappt hier auf 5 ESPs nämlich ohne Probleme.

EvilBlood, 2023-06-09:
[image]
bei dem Blauen kassten stand mal der hash. nun ist nur noch die Firmware version da.

[image]
[image]

Das sind die Settings diese habe ich nur gespeichert und dann lief MQtt wieder.

tbnobody, 2023-06-09:
> bei dem Blauen kassten stand mal der hash. nun ist nur noch die Firmware version da.

Das ist korrekt so. Wenn der aktuelle Git-Hash auf einen Git-Tag zeigt wird dort einfachhalber der entsprechende Git-Tag angezeigt. In dem Fall verhält sich der Tag genauso wie ein Git-Hash. 
Du kannst also auf https://github.com/tbnobody/OpenDTU/tree/170978c6f8ca97e2b6dfe9826b76466c65315c59 gehen oder auf https://github.com/tbnobody/OpenDTU/tree/v23.6.9 das macht keinen Unterschied. Ebenfalls sieht man die Tags ja auch wunderbar im Git-Log.

Leider ist es jetzt natürlich nicht mehr nachvollziehbar warum das MQTT nicht verbunden hat. Dafür hätte man im entsprechenden Zustand genauer hinsehen müssen. (Screenshot von Info --> MQTT und Wert der `solar/daniel/dtu/status` Topic)

EvilBlood, 2023-06-09:
Also die Settings waren genau die gleichen wie auf dem Screenshot ich habe nur auf speichern gedrückt.

Der MQTT wert war Offline und warn mit dem MQTT Explorer für über 2 Minuten keine Aktualisierungen auf dem Pfad solar/daniel zu sehen (zu dem Zeitpunkt hatte ich schon ca. eine Stunde keine Daten mehr erhalten via MDTT.

Im Webinterface war der Wechselrichter ohne Probleme erreichbar und hat seine werte auch aktualisiert.

HeinKlocktein, 2023-06-13:
Ist bei mir genau so.
Speichern oder den ESP neustarten und die MQTT Übertragung läuft wieder. Der MQTT Server läuft und empfängt von anderen ESPs.

tbnobody, 2023-06-14:
Und ich frage noch einmal nach folgendem wenn das passiert :)
Screenshot von Info --> MQTT und Wert der `solar/dtu/status` Topic

3joern, 2023-06-19:
Salute, sorry wenn ich hier reinkrätsche, mir fällt in letzter Zeit auch öfters auf, dass gewisse Daten per MQTT nicht mehr rausgeschickt werden. Die Betonung auf gewisse Daten, hier zB was beim ioBroker ankommt, einmal die Power, die hat oft Aussetzer (siehe 14:22 - 14:28 kam nichts durch), während die Temperatur zur gleichen Zeit problemlos alle 60sek geliefert wird. Frage, ist das das gleiche Problem hier und 2 was kann das sein?

Power
[image]

Temperatur
[image]

HeinKlocktein, 2023-06-26:
> Und ich frage noch einmal nach folgendem wenn das passiert :) Screenshot von Info --> MQTT und Wert der `solar/dtu/status` Topic

Siehe https://github.com/tbnobody/OpenDTU/issues/965

tbnobody, 2023-07-12:
please check version [v23.7.12](https://github.com/tbnobody/OpenDTU/releases/tag/v23.7.12) 
It fixes a reconnect error of the mqtt client.

tbnobody, 2023-07-17:
Going to close this issue asa from my standpoint it should be fixed. Feel free to reopen or open a new one.

closed 2023-07-17 by tbnobody