tbnobody/OpenDTU #1021: Prometheus syntax error with wifi_bssid

Gerbyte, 2023-06-10:
### What happened?

After updating to v23.6.9 I get the following error in my Prometheus:
`strconv.ParseFloat: parsing "2C:3A:FD:FD:B1:92": invalid syntax while parsing: "wifi_bssid 2C:3A:FD:FD:B1:92"`
[image]

### To Reproduce Bug

Install version v23.6.9 on the openDTU and query with Prometheus.
The following Scrape Config must be added in Prometheus:

```
  - job_name: "node-exporter-opendtu"
    metrics_path: '/api/prometheus/metrics'
    static_configs:
      - targets: ['IP_of_openDTU']
```

### Expected Behavior

To solve the problem, the metric "wifi_bssid" could be changed to a metric that is recorded as a string instead of a floating point number. This can be achieved by changing the metric type from "gauge" to "label". Here is an example of how the metric "wifi_bssid" can be captured as a string:
```
# TYPE wifi_bssid label
wifi_bssid{value="2C:3A:FD:FD:B0:91"} 1
```

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.9

tbnobody, 2023-06-10:
> wifi_bssid{value="2C:3A:FD:FD:B0:91"} 1

Is the "1" behind the closing bracket a type or is it required?

tbnobody, 2023-06-10:
I am not very fimiliar with prometheus but would it make sense to output it in the following way?

```
# HELP wifi_station WiFi Station info
# TYPE wifi_station label
wifi_station{bssid="D0:C7:xx:xx:xx:xx"} 1
```

jfrede, 2023-06-10:
this should at least fix the strconv.ParseFloat since prometheus tries to parse the MAC as a value. 
Only numbers are allowed as values and a MAC is not a number.
You can put the string into the label as you mentioned .

closed 2023-06-11 by tbnobody

jfrede, 2023-06-12:
Sry missed the TYPE is has to be gauge not label now we get the error. 
invalid metric type "label"

commit 08f47c2f (2023-06-11): Fix: Prometheus syntax error with wifi_bssid

Fixes #1021