tbnobody/OpenDTU #1022: Temporarily reporting yieldtotal of 2113405kWh for HM-600

jstammi, 2023-06-10:
### What happened?

Suddenly 1st time/starting yesterday I see a yieldtotal reportedly at 2113405kWh. Multiple times already reporting this exact number.
Plus meanwhile twice a directly following update with still much too high but varying values.

Nominal values ~28xkWh at the moment.

I have never seen this before (since 27.2.2023 when installing my "Balkonkraftwerk").

Yesterday it happened before and after having upgraded to the latest version.
And it happened also in the evening, so it does not seem tempreature related.

Yesterday I saw it only in the recording of my connected homeassistant, whose energy charts now got completely messed up.
Today I saw it "live", could check also some other values in the mqtt that are not announced to homeassistant sensors
(Btw: why? Just forgotten or by intention?)

That same time many other values go done to 0.
The is_valid at least for AC remained at 1.

### To Reproduce Bug

No idea, it just happens

### Expected Behavior

The data seems corrupted and should not be propagated

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

Until 9.6. 18:25: aa4868c/v23.5.31 Afterwards: 170978c/v23.6.9

### Relevant log/trace output

```Shell
Invalid data events upo to now:

9.6.2023 
11:43:33 2.113.405,00
18:40:23 2.113.405,00
19:47:41 2.113.405,00
19:47:45   694.419,44

10.6.2023
12:00:36 2.113.405,00
12:00:40   729.219,06
12:20:55 2.113.405,00
13:26:12 2.113.405,00
```

tbnobody, 2023-06-10:
Some further information are required:
* Are you using multiple DTUs (DTU Lite, DTU Pro or ahoy) on the same inverter?
* What inverter type are you  using?
* Please provide a snapshot of the console at the time when this large values occour

jstammi, 2023-06-10:
This is a single dtu (this openDTU) accessing a single HM-600 inverter, that 2 panels are attched to (and there seems no other pv close to us AFAIK/AFAIS).

For the console snapshot I need to wait for the next time this is happening. I just started capturing the console as it is available by the web browser.

Btw: the webapi misses the description of the `/console` websocket endpoint, by which it is possible to continously capture it, e.g. using `wscat`:
```
wscat -c <opendtu>/console >opendtu.log
```

jstammi, 2023-06-10:
It happened again at 18:47:04.
A single wrong value.
2s later the nominal value 285,97kWh was reported again.

The websocket aborted at some time. Unfortunately I do not see any obvious timestamps. Hopefully there one somewhere in the data shown in hex only ... ?
And hopefully it stopped only afterwards - but I am quite sure about that, as file's last change time is shown as 19:19. And this is corresponding to a series of Error-2-DTU-command-failed events, starting exactly that time.

Else I have to redo the capturing until next occurence with prepending each line with a timestamp ... ?
[console.log.gz](https://github.com/tbnobody/OpenDTU/files/11713086/console.log.gz)

jstammi, 2023-06-11:
Ok, it happened 3 times today. 3x single times propagating the `2.113.405 kWh`.
After the 3rd one a nominal value were shown, followed by an single invalid one again, but this time with value `3.549.897,5`

Unfortunately I failed again to capture with prefixing by timestamp (found the correct statement but did not copy it to capturing script :-( ).
Next try tomorrow. If it is still needed. As I checked the captured logfile for spurious messages after the `RX Period End` one:

2 types of messages came to my intention:
3x `Packet handling error`
1x (with shown doubled in 2 succeeding line) `FATAL: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 18) stats packet too large for buffer`
IMHO this correlates well with symptoms, 3x the same value, latest with different one.

Following log excerps for each of the 4. Alays with one inverter fetch block before and afterwards (I could attach again full log if this is needed and if it helps also without timestamps).

#1
```
Fetch inverter: 1141829XXXXX
TX RealTimeRunData Channel: 3 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 86 0B 0F 00 00 00 00 00 00 00 00 DA 57 55 
Interrupt received
RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 02 3B F0 00 02 35 02 09 B1 09 AC 09 08 13 85 01 57 B5 | -30 dBm
Interrupt received
RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 0F 03 E8 01 70 00 0E AD 95 B5 | -80 dBm
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 23 --> 15 82 97 74 01 80 14 22 76 81 34 
Interrupt received
RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 50 00 35 00 B2 01 51 00 36 00 B5 00 02 92 | -80 dBm
RX Period End
Success

Fetch inverter: 1141829XXXXX
TX RealTimeRunData Channel: 40 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 86 0B 11 00 00 00 00 00 00 00 00 7A D6 6A 
Interrupt received
RX Channel: 3 --> 95 82 97 74 01 82 97 74 01 02 3B F0 00 02 35 02 09 B1 09 AC 09 08 13 86 01 57 B6 | -80 dBm
Interrupt received
RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 0F 03 E8 01 71 00 0E AE C1 E3 | -80 dBm
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 61 --> 15 82 97 74 01 80 14 22 76 81 34 
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 75 --> 15 82 97 74 01 80 14 22 76 81 34 
Interrupt received
RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 50 00 35 00 B2 01 52 00 36 00 B5 00 02 91 | -80 dBm
Interrupt received
RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 02 3B F0 00 02 35 02 09 B1 09 AC 09 08 13 86 01 57 B6 | -80 dBm
RX Period End
Packet handling error
Interrupt received
RX Channel: 3 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 0F 03 E8 01 71 00 0E AE 75 57 | -80 dBm

Fetch inverter: 1141829XXXXX
TX RealTimeRunData Channel: 3 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 86 0B 13 00 00 00 00 00 00 00 00 1A CF 11 
Interrupt received
RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 02 3B F0 00 02 35 02 09 B1 09 AC 09 08 13 86 01 57 B6 | -80 dBm
Interrupt received
RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 0F 03 E8 01 71 00 0E AE C1 E3 | -80 dBm
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 23 --> 15 82 97 74 01 80 14 22 76 81 34 
Interrupt received
RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 50 00 35 00 B2 01 51 00 36 00 B5 00 02 92 | -80 dBm
RX Period End
Success
```

#2
[... 161 lines cut ...]
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 75 --> 15 82 97 74 01 80 14 22 76 81 34 
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 3 --> 15 82 97 74 01 80 14 22 76 81 34 
Interrupt received
RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 52 00 2B 00 90 01 4C 00 2C 00 92 00 02 8C | -80 dBm
RX Period End
FATAL: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 18) stats packet too large for buffer
FATAL: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 18) stats packet too large for buffer
Success

Fetch inverter: 1141829XXXXX
TX RealTimeRunData Channel: 23 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 86 10 D0 00 00 00 00 00 00 00 00 9F FB 78 
Interrupt received
RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 51 00 2B 00 90 01 4D 00 2C 00 92 00 02 8E | -80 dBm
Interrupt received
RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 02 3B F6 00 02 35 08 09 B7 09 B2 09 03 13 86 01 14 EA | -80 dBm
Interrupt received
RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 0C 03 E8 01 66 00 0E 06 9B 05 | -80 dBm
RX Period End
Success
```

jstammi, 2023-06-11:
Just checked for the line numbers, those messages appeared. And I guess No 4 is not related to the 4th invalid number being reported. As it appeared much later only in the log. But should have been close to No 3 then:

```
$ grep -n -e FATAL -e 'Packet handling error' melkhoern-console.51304.log
10094:Packet handling error
20905:Packet handling error
22953:Packet handling error
29068:FATAL: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 18) stats packet too large for buffer
29069:FATAL: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 18) stats packet too large for buffer
```

jstammi, 2023-06-13:
It just happened several times again. I just had a quick look to the log. And there are NO error messages arround the times it showed up.

BUT: there was at least the last 2 times (that I have log now available for with millis resolution) some millis after the "Success" message in the end of the nominal "Fetch Inverter" block a log message starting another block for "TX AlarmData Channel", as shown below.

Just as a guess, no time to check the code by now myself: is it possible that by bad timing it's handling clashes with an mqtt data propagation in progress? Do they use same mem (buffer?) that are not guarded against parallel access?

`
[10:57:45.318758] Fetch inverter: 114182977401^M
[10:57:45.318869]
[10:57:45.368728] TX RealTimeRunData Channel: 61 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 2F 89 00 00 00 00 00 00 00 00 EB 69 F6 ^M
[10:57:45.368798]
[10:57:45.377606] Interrupt received^M
[10:57:45.377674]
[10:57:45.386291] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 60 03 3B 0B 5F 01 5D 03 3C 0B 4A 00 02 B8 | -80 dBm^M
[10:57:45.386357]
[10:57:45.418763] Interrupt received^M
[10:57:45.418823]
[10:57:45.426304] RX Channel: 23 --> 95 82 97 74 01 82 97 74 01 02 47 74 00 02 40 8A 02 43 02 48 09 0D 13 8B 15 A4 4A | -80 dBm^M
[10:57:45.426369]
[10:57:45.905681] RX Period End^M
[10:57:45.905808] Last missing^M
[10:57:45.905862] Request retransmit: 3^M
[10:57:45.905910] TX RequestFrame Channel: 75 --> 15 82 97 74 01 80 14 22 76 83 36 ^M
[10:57:45.905956]
[10:57:45.917678] Interrupt received^M
[10:57:45.917747]
[10:57:45.926290] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 EF 03 E8 01 F4 00 03 48 53 FF | -80 dBm^M
[10:57:45.926355]
[10:57:46.028918] RX Period End^M
[10:57:46.029028] Success^M
[10:57:46.029064]
[10:57:46.085640] TX AlarmData Channel: 3 --> 15 82 97 74 01 80 14 22 76 80 11 00 64 88 2F 89 00 00 00 00 00 00 00 00 31 72 2D ^M
[10:57:46.085709]
[10:57:46.104900] Interrupt received^M
[10:57:46.104935]
[10:57:46.114123] RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 01 00 01 80 01 00 01 2A F5 2A F5 00 00 00 00 80 02 97 | -80 dBm^M
[10:57:46.114157]
[10:57:46.148459] Interrupt received^M
[10:57:46.148497]
[10:57:46.156521] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 02 00 02 3C 76 3C 76 00 00 00 05 80 02 00 03 70 D1 B0 | -80 dBm^M
[10:57:46.156586]
[10:57:46.177779] Interrupt received^M
[10:57:46.177844]
[10:57:46.189483] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 83 70 D1 FF FF FF FB 65 39 EF | -30 dBm^M
[10:57:46.189549]
[10:57:46.839582] RX Period End^M
[10:57:46.839668] Success^M
`

jstammi, 2023-06-13:
It just happened several times again. I just had a quick look to the log. And there are NO error messages arround the times it showed up.

BUT: there was at least the last 2 times (that I have log now available for with millis resolution) some millis after the "Success" message in the end of the nominal "Fetch Inverter" block a log message starting another block for "TX AlarmData Channel", as shown below.

Just as a guess, no time to check the code by now myself: is it possible that by bad timing it's handling clashes with an mqtt data propagation in progress? Do they use same mem (buffer?) that are not guarded against parallel access?

(corrupted data arrived with time 10:57:45 at the homeassistant)
`[10:57:45.318758] Fetch inverter: 114182977401
[10:57:45.318869]
[10:57:45.368728] TX RealTimeRunData Channel: 61 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 2F 89 00 00 00 00 00 00 00 00 EB 69 F6 
[10:57:45.368798]
[10:57:45.377606] Interrupt received
[10:57:45.377674]
[10:57:45.386291] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 60 03 3B 0B 5F 01 5D 03 3C 0B 4A 00 02 B8 | -80 dBm
[10:57:45.386357]
[10:57:45.418763] Interrupt received
[10:57:45.418823]
[10:57:45.426304] RX Channel: 23 --> 95 82 97 74 01 82 97 74 01 02 47 74 00 02 40 8A 02 43 02 48 09 0D 13 8B 15 A4 4A | -80 dBm
[10:57:45.426369]
[10:57:45.905681] RX Period End
[10:57:45.905808] Last missing
[10:57:45.905862] Request retransmit: 3
[10:57:45.905910] TX RequestFrame Channel: 75 --> 15 82 97 74 01 80 14 22 76 83 36 
[10:57:45.905956]
[10:57:45.917678] Interrupt received
[10:57:45.917747]
[10:57:45.926290] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 EF 03 E8 01 F4 00 03 48 53 FF | -80 dBm
[10:57:45.926355]
[10:57:46.028918] RX Period End
[10:57:46.029028] Success
[10:57:46.029064]
[10:57:46.085640] TX AlarmData Channel: 3 --> 15 82 97 74 01 80 14 22 76 80 11 00 64 88 2F 89 00 00 00 00 00 00 00 00 31 72 2D 
[10:57:46.085709]
[10:57:46.104900] Interrupt received
[10:57:46.104935]
[10:57:46.114123] RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 01 00 01 80 01 00 01 2A F5 2A F5 00 00 00 00 80 02 97 | -80 dBm
[10:57:46.114157]
[10:57:46.148459] Interrupt received
[10:57:46.148497]
[10:57:46.156521] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 02 00 02 3C 76 3C 76 00 00 00 05 80 02 00 03 70 D1 B0 | -80 dBm
[10:57:46.156586]
[10:57:46.177779] Interrupt received
[10:57:46.177844]
[10:57:46.189483] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 83 70 D1 FF FF FF FB 65 39 EF | -30 dBm
[10:57:46.189549]
[10:57:46.839582] RX Period End
[10:57:46.839668] Success`

jstammi, 2023-06-13:
Happened another 3 times now. Always as described before, with `TX AlarmData Channel` after `Success` for `Fetch inverter`.

tbnobody, 2023-06-14:
Folgendes macht mich stutzig:

```
TX SystemConfigPara Channel: 3 --> 15 82 97 74 01 80 14 22 76 80 05 00 64 86 10 CE 00 00 00 00 00 00 00 00 F1 75 88 
Interrupt received
RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 02 3B F6 00 02 35 08 09 B7 09 B2 09 04 13 86 01 15 EC | -80 dBm
RX Period End
Last missing
Request retransmit: 3
TX RequestFrame Channel: 23 --> 15 82 97 74 01 80 14 22 76 83 36 
Interrupt received
RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 83 00 00 00 0C 03 E8 01 67 00 0E 4C FF 2A | -80 dBm
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 40 --> 15 82 97 74 01 80 14 22 76 81 34 
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 61 --> 15 82 97 74 01 80 14 22 76 81 34 
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 75 --> 15 82 97 74 01 80 14 22 76 81 34 
RX Period End
Middle missing
Request retransmit: 1
TX RequestFrame Channel: 3 --> 15 82 97 74 01 80 14 22 76 81 34 
Interrupt received
RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 52 00 2B 00 90 01 4C 00 2C 00 92 00 02 8C | -80 dBm
RX Period End
FATAL: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 18) stats packet too large for buffer
FATAL: (lib/Hoymiles/src/parser/SystemConfigParaParser.cpp, 18) stats packet too large for buffer
Success
```
Hier kommen auf eine SystemConfigPara 3 Antwortpakete. Das dürfte garnicht sein. Kannst du ggf. mal einen Screenshot vom Inverter-Info Dialog posten? 
[image]

jstammi, 2023-06-14:
Hi, kann ich machen. Kann sich aber auch gleich erledigen :-)

Ich habe mich gerade durch den Code gewühlt. Und verstehe ganz bestimmt noch nicht viel. Aber an einer ähnlichen Stelle bin ich auch gerade hängen geblieben - nur irgendwie umgekehrt.

Die Fehlermeldungen korrelieren nicht mit den korrupten Daten (könnten aber dann doch damit zusammenhängen, komme ich später drauf zurück).

Ich die Pakete oben von gestern "per Hand" dekodiert. Da stehen vernünftige Daten drin, nichts als korrupt für mich erkennbar. Könnte auch noch ne Nebenläufigkeit vlt sein - vor 2h war das noch meine Vermutung.
Aber mir ist in den beiden Paketen direkt davor etwas aufgefallen. Die dürfte nach meinem bisherigen Verständnis so nicht verarbeitet/gezeigt werden:

```
[10:57:41.684989] TX SystemConfigPara Channel: 23 --> 15 82 97 74 01 80 14 22 76 80 05 00 64 88 2F 85 00 00 00 00 00 00 00 00 25 33 60 
[10:57:41.700984] Interrupt received
[10:57:41.711307] RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
[10:57:41.888241] RX Period End
[10:57:41.888309] Success

[10:57:43.176678] Fetch inverter: 114182977401
[10:57:43.226306] TX RealTimeRunData Channel: 40 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 2F 87 00 00 00 00 00 00 00 00 8B 25 D4 
[10:57:43.242392] Interrupt received
[10:57:43.251989] RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
[10:57:43.732508] RX Period End
[10:57:43.732594] Success
```

a) "Fetch inverter" ist mit nur EINEM Paket zufriedengestellt? AFAIS müssten das immer 3 sein ... ? Im Sourcecode sehe ich allerdings nicht den Fall abgefangen, dass nur das erste Paket gekommen ist (mit `fragmentId` 1 aus `0x81` extrahiert btw; `InverterAbstract` Zeile 157)

b) Das RX Paket in "Fetch Inverter" ist exakt jenes aus dem "SystemConfigPara" zuvor

c) Zufällig(?) sind die letzten beiden Bytes vor dem CRC `0x7DF8`, welches das erste Wort für die MQTT veröffentlichte CH1 YieldTotal von `2113405000 = 0x7DF80048` ist. Das 2. Wort `0x0048` müsste im 2. Paket drin sein, dass aber nach Log nicht empfangen wurde - keine Ahnung, wo die 0x48 herkommt. Sehe ich erst ein paar Sekunden später in nem TX, aber da sind die Daten über MQTT schon längst raus.
CH2 Yield Total wird zur gleichen Zeit übrigens als 0 empfangen.
Und damit with dann `CH0 YieldTotal` zu dem Wert in CH1.

d) Bei Deinen zu vielen Paketen, die zu Fehlermeldung führen, ist auch wieder dieses SystemConfigPara Command beteiligt

Also IMHO liegt da irgendwo des Problems Ursache. Aber für heute steige ich an der Stelle erstmal aus.

Wechselrichter Info:
[image]

jstammi, 2023-06-14:
Oh man, im Fenster schließen sehe ich doch nochmal was ... hätte ich noch 2 Zeilen weiter nach oben geschaut ... vollständig steht da nämlich:

```
[10:57:41.024986] Fetch inverter: 1141829XXXXX
[10:57:41.025093] Request SystemConfigPara
[10:57:41.684989] TX SystemConfigPara Channel: 23 --> 15 82 97 74 01 80 14 22 76 80 05 00 64 88 2F 85 00 00 00 00 00 00 00 00 25 33 60 
[10:57:41.700984] Interrupt received
[10:57:41.711307] RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
[10:57:41.888241] RX Period End
[10:57:41.888309] Success

[10:57:43.176678] Fetch inverter: 1141829XXXXX
[10:57:43.226306] TX RealTimeRunData Channel: 40 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 2F 87 00 00 00 00 00 00 00 00 8B 25 D4 
[10:57:43.242392] Interrupt received
[10:57:43.251989] RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
[10:57:43.732508] RX Period End
[10:57:43.732594] Success
```

Und bei dem Beispiel No 4 mit der Fehlermeldung, die Du angeschaut hattest desgleichen. Die ersten beiden Zeilen in meinem Log.

AFAIU werden da 2 Commands quasi parallel in die Queue gegeben ... also doch eher eine Nebenläufigkeit?
Wie sind denn die beiden Commands gegeneinander abgegrenzt? Kann es da zu ner theoretischen Überschneidung kommen?

Ich glaube ich hatte eine Stelle gesehen, wo in eine Queue das Command reingeht. Und danach die Adresse des 1. Commands in der Queue als die dazu gehörige angenommen wird. Wenn jetzt das 2. Command in die Queue gepackt wird, bevor für das 1. die Adresse ermittelt wurde? Werden beide Cores der CPU (AzDelivery ESP32 D1 Mini) frei verwendet oder ist eine fest zB für WLAN reserviert?

jstammi, 2023-06-14:
`HoymilesRadio.h` Zeilen 20-24 ... das meinte ich. Wenn für 2 Commands zuerst 2x Zeile 22 abgearbeitet wird, und danach erst 2x 23, dann würde mit ner falschen Adresse danach weiter gearbeitet ... könnte das die Symptome erklären? Oder bin ich auf dem komplett falschen Dampfer?

tbnobody, 2023-06-14:
>  Wenn für 2 Commands zuerst 2x Zeile 22 abgearbeitet wird, und danach erst 2x 23, dann würde mit ner falschen Adresse danach weiter gearbeitet ... 

Kann ja nicht passieren. Das läuft linear auf einem Kern ab. Immer im Kontext der Main Loop.

jstammi, 2023-06-15:
Ok, das dann nicht.

An die serielle Schnittstelle komm ich so einfach & auf die Schnelle jetzt nicht ran.

Hab mir gerade noch mal einen anderen Vorfall angeschaut, und jetzt auch mal die Command Pakete. Und danach steht für mich ziemlich fest, dass eintreffende Pakete einem falschen Command zugeordnet werden. Was aber nicht wirklich verwunderlich ist, da es AFAIS keine Möglichkeit gibt, die RX Pakete einem konkreten TX zuzuordnen. Per Zähler zB, ab da ist nichts dazu drin, weder im TX, noch im RX - jedenfalls soweit die Bedutung der Bits & Bytes bekannt ist (bei TX sind da noch ein paar `0x00`te, beim RX nur die ersten beiden in der Payload, AFAIS immer `0x0001` - unabhängig vom Command Packet Type, aus das die Antwort erfolgt)

Eine Zuordnung zum Command kann also nur über die Anzahl der erwarteten Pakete sowie deren Länge(n) erfolgen. Und das sehe ich derzeit nicht im Code. Dafinde ich nur die Prüfung, dass es gültige Paket Sequenz war - egal ob aus 1 oder 3 Paketen bestehend.

Wenn also ein einzelnes SystemConfigPara Paket ankommt (warum auch immer) mit Fragment ID 81, kann das einem gerade wartenden RealTimerunData  Command zugeordnet werden, ohne dass die Prüfung das ablehnt. Das einzelne Fragment wird per `memcpy` in `StatisticsParser::appendFragment` auf die ersten 14 Bytes im Parser-Buffer kopiert (der vorher per `StatisticsParser::clearBuffer` ge-`0x00`t wurde). Teil 2+3 kommen nicht mehr, das wird aber nicht bemerkt. Und falls danach & vor der nächsten RealTimerunData  Runde MQTT die Daten dort rausholt, kommt dabei Schrott raus (wobei für mich noch ungeklärt ist, woher regelmäßig die `0x48` herkommt für Byte 17 (2. Byte im 2. Paket).

Was IMHO, mit dem was ich jetzt gesehen habe, rein muss, ist eine Prüfung vor dem Parsen Inverter spezifisch von:
 - Fragment ID(s) (beim HM-600 ist `0x81` falsch)
 - Anzahl der Pakete und deren Längen (RealTimerunData  beim HM-600: 3, 0x01 und 0x02 16 Bytes, 0x83 10 Bytes)

Noch offen ist, warum der Inverter das SystemConfigPara Antwort Paket noch ein 2. Mal zu senden scheint, obwohl RealTimeRunData angefordert wurde.

```
[12:10:57.604303] Fetch inverter: 1141829XXXXX
[12:10:57.604411] Request SystemConfigPara
[12:10:57.604445]
[12:10:57.708925] TX RealTimeRunData Channel: 61 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 40 B1 00 00 00 00 00 00 00 00 A4 69 EE 
[12:10:57.708990]
[12:10:58.260097] RX Period End
[12:10:58.260195] All missing
[12:10:58.260229] Nothing received, resend whole request
[12:10:58.260260] TX RealTimeRunData Channel: 75 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 40 B1 00 00 00 00 00 00 00 00 A4 69 EE 
[12:10:58.260290]
[12:10:58.271680] Interrupt received
[12:10:58.271745]
[12:10:58.280002] RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 69 03 79 0C 84 01 64 03 84 0C 84 00 02 67 | -80 dBm
[12:10:58.280081]
[12:10:58.306198] Interrupt received
[12:10:58.306262]
[12:10:58.322153] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 02 48 E0 00 02 41 F4 03 AF 03 B2 09 1C 13 89 17 E8 E5 | -80 dBm
[12:10:58.322218]
[12:10:58.352592] Interrupt received
[12:10:58.352657]
[12:10:58.363458] RX Channel: 23 --> 95 82 97 74 01 82 97 74 01 83 00 05 01 06 03 E8 02 39 00 04 1E 2C F2 | -80 dBm
[12:10:58.363523]
[12:10:58.760648] RX Period End
[12:10:58.760750] Success
[12:10:58.760784]
[12:10:58.867899] TX SystemConfigPara Channel: 3 --> 15 82 97 74 01 80 14 22 76 80 05 00 64 88 40 B1 00 00 00 00 00 00 00 00 6A 66 21 
 0: 05          dataType
 1: 00 
 2: 64 88 40 B1 time MSB    1686651057 = Tue 13 June 2023 12:10:57
 6: 00 00 00 00 
10: 00 00 00 00 Password
14: 6A 66       CRC
[12:10:58.867964]
[12:10:58.907622] Interrupt received
[12:10:58.907687]
[12:10:58.916633] RX Channel: 40 --> 95 82 97 74 01 82 97 74 01 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
[12:10:58.916698]
[12:10:59.091387] RX Period End
[12:10:59.091491] Success
[12:10:59.091524]
[12:10:59.775529] Fetch inverter: 1141829XXXXX
[12:10:59.775630]
[12:10:59.822338] TX RealTimeRunData Channel: 23 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 40 B4 00 00 00 00 00 00 00 00 F4 56 84 
[12:10:59.822403]
[12:10:59.833745] Interrupt received
[12:10:59.833810]
[12:10:59.843258] RX Channel: 61 --> 95 82 97 74 01 82 97 74 01 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
[12:10:59.843323]
[12:11:00.325888] RX Period End
[12:11:00.325965] Success
[12:11:00.325999]
[12:11:01.885181] Fetch inverter: 1141829XXXXX
[12:11:01.885293]
[12:11:01.936312] TX RealTimeRunData Channel: 40 --> 15 82 97 74 01 80 14 22 76 80 0B 00 64 88 40 B6 00 00 00 00 00 00 00 00 94 4F FF 
[12:11:01.936389]
[12:11:01.944526] Interrupt received
[12:11:01.944604]
[12:11:01.954038] RX Channel: 3 --> 95 82 97 74 01 82 97 74 01 01 00 01 01 69 03 79 0C 84 01 64 03 85 0C 84 00 02 66 | -80 dBm
[12:11:01.954103]
[12:11:01.972978] Interrupt received
[12:11:01.973065]
[12:11:01.981973] RX Channel: 3 --> 95 82 97 74 01 82 97 74 01 02 48 E0 00 02 41 F5 03 AF 03 B3 09 1E 13 89 17 E8 E7 | -80 dBm
[12:11:01.982049]
[12:11:02.038103] Interrupt received
[12:11:02.038204]
[12:11:02.047135] RX Channel: 75 --> 95 82 97 74 01 82 97 74 01 83 00 00 01 06 03 E8 02 39 00 04 CE 41 4A | -80 dBm
[12:11:02.047185]
[12:11:02.437853] RX Period End
[12:11:02.437934] Success
 0: 00 01 
 2: 01 69       CH0 FLD_UDC   36,1   V    
 4: 03 79       CH0 FLD_IDC    8,89  A
 6: 0C 84       CH0 FLD_PDC  320,4   W
 8: 01 64       CH1 FLD_UDC   35,6   V
10: 03 85       CH1 FLD_IDC    9,01  A
12: 0C 84       CH1 FLD_PDC  320,4   W
14: 00 02 48 E0 CH0 FLD_YT   149,728 kWh
18: 00 02 41 F5 CH1 FLD_YT   147,957 kWh
22: 03 AF       CH0 FLD_YD    94,3   Wh
24: 03 B3       CH1 FLD_YD    94,7   Wh
26: 09 1E       AC FLD_UAC
28: 13 89       AC FLD_F
30: 17 E8       AC FLD_PAC   612,0   W
32: 00 00       AC FLD_Q
34: 01 06       AC FLD_IAC
36: 03 E8       AC FLD_PF
38: 02 39       FLD_T
40: 00 04       FLD_EVT_LOG
42: CE 41       CRC
```

tbnobody, 2023-06-15:
> Eine Zuordnung zum Command kann also nur über die Anzahl der erwarteten Pakete sowie deren Länge(n) erfolgen.

Absolut falsch! Es ist exakt bekannt was angefragt wurde, und nur das kann als Antwort kommen. Der Inverter hat einen festen Timeout innerhalb dessen er die Antwort liefert. Wenn innerhalb dieser Zeit nichts kommt, wird auch nicht mehr kommen. 

Die Frage ist, warum passiert das nur bei dir und sonst augenscheinlich nicht. Kann auch am Caching des Websockets liegen. Wie gesagt, direkte Serielle Verbindung.

jstammi, 2023-06-15:
Mutige Aussage. Wo wir doch SEHEN, dass es auch anders sein kann ;-). 

Dass die richtige Antwort kommt ist "nur" eine Annahme. Nämlich dass der Inverter an der Stelle richtig sendet, Inhaltlich und was das Timing angeht. Können wir nicht sicherstellen, auch nicht anhand dessen Sourcecode sehen, und haben wir keinen Einfluß drauf. Im Gegenteil, genau wie mit korrupten Paketen müssen auch semantisch falsche Pakete, wo das möglich ist, erkannt und verworfen werden IMHO. Sonst ist es nicht robust.

Anhand DER DATEN PAKETE ist diese eindeutige Zuordnung AFAIS nicht möglich. Und die beschriebenen semantischen Prüfungen fehlen halt noch. AFAIS kein Drama. Ich würde jetzt in InverterAbstract vor Zeile 169 eine `validateRxFragment` Methode noch einschieben, die ableitende Implementationen für spezifische Checks überschreiben können. Hab jetzt noch keinen Überblick, in wieweit das bei den Modellen vereinheitlicht werden kann. Aber würde ich mich sonst nächste Woche dran setzen.

Und ja, wäre sicher spannend herauszufinden warum dem so ist. Ändert aber nichts an den derzeit fehlenden Prüfungen der eingehenden Daten.
Und dass das bei niemandem anders passiert ... sicher? Fällt bei mir nur auf, weil das das Energie Dashboard "zerstört". Ist das bei vielen so am Start? Oder würde ein einzelner sporadischer Ausreißer in den Daten wirklich vielen/jemandem auffallen?
Und dann auch nur mit einem Inverter? Weil sobald mehrere im Spiel sind, dürfte das Timing ein ziemlich anderes sein ... ?

tbnobody, 2023-06-15:
> Dass die richtige Antwort kommt ist "nur" eine Annahme. Nämlich dass der Inverter an der Stelle richtig sendet, Inhaltlich und was das Timing angeht. Können wir nicht sicherstellen, auch nicht anhand dessen Sourcecode sehen, und haben wir keinen Einfluß drauf.

Lese dir bitte den GESAMTEN Thread auf Microcontroller.net durch bevor du so etwas sagst. Das hat schon so seine Richtigkeit.

jstammi, 2023-06-28:
Successfully running this afternoon your fix patched by what I provided to you as pr https://github.com/tbnobody/OpenDTU/pull/1096: besides several CRC errors, twice the check saved me from more parse results on wrong packets:
{code}
[16:36:42.174641] Packet handling error
[16:45:29.745919] ERROR in RealTimeRunData: Received fragment size: 16 min. expected size: 42
[16:45:29.745956] Packet handling error
[17:07:39.541591] Packet handling error
[17:07:47.424778] Packet handling error
[17:18:05.228533] ERROR in RealTimeRunData: Received fragment size: 16 min. expected size: 42
[17:18:05.228568] Packet handling error
[17:24:42.706939] Packet handling error
[17:25:51.622376] Packet handling error
[17:59:52.796870] Packet handling error
[18:05:42.892168] Packet handling error
[18:27:12.536583] Packet handling error
[18:33:31.867578] Packet handling error
[18:35:15.513693] Packet handling error
{code}

tbnobody, 2023-06-28:
great to hear that it's working...
on the other hand.... seems to be shitty firmware on the inverter if I see this wrong formattd packages
will push your merged PR this evening...

jstammi, 2023-06-28:
Shitty FW ... ok.

But they are not wrong formatted. The inverter sends packets belonging to a different TX (again) while OpenDTU is waiting for a statistics packet.

Search in the comments for `95 82 97 74 01 82 97 74 01 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9` and you will see, that this very same packet is received first for `TX SystemConfigPara` and again afterwards for `TX RealTimeRunData`. Sometimes even on different channel, so this is for sure sent again by the inverter. For whatever reasons.

For #1084 the CRC is valid, too, but I do not see the packet having arrived for another TX in the log. So the cause may be a different one. The countermeasure is the same ;-).