tbnobody/OpenDTU #1023: default system profile disables CMT2300A and enables Nrf24 even though its not even available

blitterz, 2023-06-11:
### What happened?

firmware update to v23.6.9, downgrade does not help

### To Reproduce Bug

update firmware I guess

### Expected Behavior

Should leave CMT2300a active not enable a Chip that is not even installed:

[image]
[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

170978c

tbnobody, 2023-06-11:
It seems that you either deleted your device profile or did not select it correctly. Just reupload your device profile and enable it again? 
What firmware .bin file did you flash? opendtu-generic.bin or opendtu-generic.factory.bin?

blitterz, 2023-06-11:
the profile has been modified to work with CMT2300a, now after firmware upgrade the profile has been overwritten with the default installation profile. i did a factory reset, but its even worse now because now the display and led is not working anymore.

[opendtu-generic.bin](https://github.com/tbnobody/OpenDTU/releases/download/v23.6.9/opendtu-generic.bin)

tbnobody, 2023-06-11:
If you've done a factory profile then you've just deleted all your settings (which is the intention of a factory reset). 
By doing a OTA device update it's not possible to delete the pin_mapping. There is just no code which could ever do this. 
But just re-upload your modified device profile and set it up as previously?

blitterz, 2023-06-11:
I have reflashed the device, now old Settings are back:

blitterz, 2023-06-11:
I have reflashed the device, now old settings are back:

[image]
[image]

I will try to update the firmware soon again

tbnobody, 2023-06-11:
As far as I see it, it will fail again. Because you built the pin config into your current firmware. There is no device profile.
Please see https://github.com/tbnobody/OpenDTU/blob/master/docs/DeviceProfiles.md

tbnobody, 2023-06-11:
Would close this issue as it is (for now) not really a issue.

closed 2023-06-11 by tbnobody