tbnobody/OpenDTU #1025: Reihenfolge der Tracker verdreht

Saarblick, 2023-06-11:
### What happened?

Kanal 1 in der DTU ist Kanal 6 auf meinem HMT2250 und umgekehrt.
Reihenfolge ist  kpl. verdreht.

[image]

### To Reproduce Bug

??

### Expected Behavior

Richtige Reihenfolge

### Install Method

Self-Compiled

### What git-hash/version of OpenDTU?

23.6.9

tbnobody, 2023-06-11:
Siehe Issue #1005
Es war vorher verdreht und wurde nun richtig gestellt um mit der Beschriftung am Inverter zusammenzupassen

homeautomation2022, 2023-06-11:
Bisher hat das außer einer hier noch keiner bestätigt, also sollte das nochmal von anderen Personen mit HMT geprüft werden.
Er nutzt laut Angaben die aktuelle Version, mit einer alten würde es bei ihm passen.

Saarblick, 2023-06-11:
Ich habs! 
Das PDF stimmt nicht mit der gegossenen Beschriftung am Umrichter überein!
Bei mir ist die 6 an der Antenne!

[image]

homeautomation2022, 2023-06-11:
> Ich habs! Das PDF stimmt nicht mit der gegossenen Beschriftung am Umrichter überein! Bei mir ist die 6 an der Antenne!
> 
> [image]

Das ist es. 👍 
In echt ist es richtig ins Gehäuse eingegossen und entspricht der Datenreihenfolge, wie in den alten OpenDTU Versionen: https://www.heise.de/select/ct/2022/23/2200714255484693681/img_20221005_160754.jpg
Alle gerenderten Bilder von Hoymiles sind falsch herum!

tbnobody, 2023-06-11:
@harmstorf can you confirm this? Seems the numbers on the inverter and in the datasheet are different.

harmstorf, 2023-06-11:
https://www.hoymiles.com/wp-content/uploads/2022/08/User-Manual_HMT-18002250-6T_Global_EN_V202207.pdf

Page 7: numbering from top (1)to bottom (6). 

This equals the engravings on the case of my inverter and is en par with the new numbering introduced with your fix. 

Everything looks good to me now: numbering equals offical manual equals my measurings.

tbnobody, 2023-06-11:
So there are invertes with the numbering upside down and downside up....
@harmstorf can you provide a image of your inverter? I googled around a bit and all I found where images of inverters where channel 6 was right to the antenna

harmstorf, 2023-06-11:
Yes, will provide a photo later as it requires some effort to reach/dismount.

Saarblick, 2023-06-11:
I will do the same , for me it’s easier when I’m at home

Saarblick, 2023-06-11:
pictures of my HMT2250, bought 2023

[image]

[image]

harmstorf, 2023-06-11:
Mine matches Saarblick‘s, which is also identical to the official manual I have mentioned before. The recent fix matches this numeration.

tbnobody, 2023-06-11:
> which is also identical to the official manual I have mentioned before.

Unfortunatly the manual shows both. Please compare Page 7 and 8 (PDF Pages, not  page numbers written in the PDF)

[image]

vs

[image]

tbnobody, 2023-06-11:
Is there only a numbering on the black colored metal or also on the aluminium back side?

homeautomation2022, 2023-06-11:
Das schwarze mit den Nummern ist bereits die Rückseite, das Ding hat keine Blechplatte wie die kleineren. Ich würde auch sehr stark behaupten, dass alle gleich gegossen wurden nur die Anleitungen sind verkehrt.

Saarblick, 2023-06-11:
No, in docu is the 1 near to the antenna, on the Hmt is the 6 near to the antenna.. The Docu are wrong

homeautomation2022, 2023-06-11:
Alle virtuellen Bilder sind falsch gegenüber den echten eingegossenen Zahlen im Gehäuse:
- Hoymiles Anleitungen
- Hoymiles Webgrafiken vom Inverter

homeautomation2022, 2023-06-11:
Der Fehler liegt bei Hoymiles:
In den Dokus sind die Varianten mit Blechdeckel abgebildet, die gibt es auch in echt und haben wirklich die Beschriftung wie im Datenblatt eingestanzt.
Die Variante mit dem Gussdeckel(mit Kühlrippen) auf der Rückseite sind andersherum(wie die Daten kommen) beschriftet.

Also es sind beide Varianten im Umlauf, bei uns aber fast ausschließlich die mit Gussdeckel, also etwas bessere Kühlung.

harmstorf, 2023-06-11:
When detaching the panel attached to the connectors near the antenna (labeled „6“ on the case) the current OpenDTU version shows panel 6 as not connected. Everything as expected.

Jean-GO, 2023-06-12:
Hallo, bei mir ist auch alles jetzt falsch herum, mein Anschluss 6 ist auch bei der Antenne laut Gehäuse.

harmstorf, 2023-06-12:
Dass jetzt alles „falsch herum“ ist, entspricht doch der Erwartung. 

Zieht doch bitte mal das Panel am Anschluss nahe der Antenne (6) ab. Welches Panel fällt in OpenDTU dann weg? Bei mir ist es nun Panel 6 - und nicht, wie vor der Änderung, Panel 1.

Jean-GO, 2023-06-12:
Kann ich später mal machen, aber ich habe laut Wechselrichter 1 und 2 Morgens, 3 und 4 Mittags und 5 und 6 Abends (Panels) angeschlossen (die Zahlen auf den Wechselrichter), was vorher gestimmt hatte und jetzt stimmt es nicht mehr.

Saarblick, 2023-06-12:
> Dass jetzt alles „falsch herum“ ist, entspricht doch der Erwartung.
> 
> Zieht doch bitte mal das Panel am Anschluss nahe der Antenne (6) ab. Welches Panel fällt in OpenDTU dann weg? Bei mir ist es nun Panel 6 - und nicht, wie vor der Änderung, Panel 1.

Nein, ich habe die neuste SW drauf, habe kanal 6 abgezogen und kanal 1  hat keine Leistung mehr in der DTU angezeigt. Das war vor 4 Wochen vor drm update noch nicht der Fall

nG64, 2023-06-12:
Ich hab jetzt einfach wieder eine alte Software auf die DTU installiert, nun ist wieder alles richtig

harmstorf, 2023-06-12:
Dann muss es doch noch andere Unterschiede geben. Ich ziehe das Panel an dem Anschluss 6 (bei der Antenne) ab und String 6 ist nun ohne Leistung. Alte OpenDTU zeigt dann String 1 ohne Leistung.

Jean-GO, 2023-06-12:
So habe es mal getestet,
wenn ich am Hoymiles HMT-2250-6T die 1 ziehe (der Aufdruck am Wechselrichter) geht jetzt String 6 aus.
Eigentlich ist es egal, habt Sache Strom kommt.
Aber wenn es geht, kann man nicht einfach in der Programmierung aufnehmen, dass man die String 1 - String 6, die Fenster verschieben kann, dass man seine eigene Reihenfolge machen kann.

tbnobody, 2023-06-12:
So wie ich  das sehe gibt es aktuell wohl 2 Bauweisen die man nicht auseinander halten kann. 
Sprich irgendjemand (bin noch unsicher) wird in Zukunft damit leben müssen das die Reihenfolge auf dem Inverter nicht mit der Beschreibung auf dem Inverter zusammen passt.

tictrick, 2023-06-12:
Vielleicht ist es Seriennummern oder Firmware abhängig?

Saarblick, 2023-06-12:
> > Oder ein benutzerdefinierbares Mapping via config?
> 
> Das würde ich mal erstmal ausschließen. Kosten / Nutzen Verhältnis macht keinen Sinn. Nur damit die Reihenfolge in der WebGUI zufällig dem entspricht wie es auf dem Inverter aufgemalt ist. Für MQTT ist es sowieso egal und hier steht ja auch noch #317 an.

So ganz egal ist es ja nicht..
Die Solarprognose die ich einsetze hatte ja auch mal eine Schnittstelle zu OpenDtu eingerichtet die jetzt nicht mehr passt..
Ich warte derzeit auf euer Ergebniss, evtl. muss ich in der Datenbank und in Grafana der Solarprognose einiges anpassen.
Da ich eh nicht der Expert bin und da schnell mal ein Tag drauf geht warte ich mal..
Derzeit stimmt nichts mehr.
Da die Gesamtleistung aber ok ist kann ich vorläufig damit leben, vorzeigbar ist es allerdings nicht..

Ist anhand der Seriennummer evtl. Das Produktionsdatum zu erkennen?

tbnobody, 2023-06-12:
> Ist anhand der Seriennummer evtl. Das Produktionsdatum zu erkennen?

Bisher hat noch keiner der HMT Besitzer die Inverter Infos gepostet.

jstammi, 2023-06-12:
Vielleicht auch erkennbar aus einer der Angaben unter dem MQTT "device" Pfad?
[image]
(HM-600 Beispiel ... aber die relevanten werden da wohl auch was stehen haben?)

Saarblick, 2023-06-12:
Dann fange ich mal an:
Seriennummer 138284597448
Hat Probleme mit der aktuellen Software die 6 ist bei der Antenne in der DTU ist halt verdreht.
Gekauft Mai 2023

tbnobody, 2023-06-12:
Wie wäre es einfach mit den Kompletten Wechselrichter Informationen?
[image]

vaterlangen, 2023-06-12:
Habe auch das gleiche Problem mit meinem dieses Jahr gekauftem HMT-2250. Nach dem FW Update werden auf dem (aktuell) einzigem beschalteten String keine Daten (außer der Einstrahlung) angezeigt. Nach Downgrade auf 23.6.1 funktioniert wieder alles korrekt.

Beschriftung endet mit ch6 bei der Antenne.

Ich fange jetzt einfach Mal an mit Daten zum WR:

Serial | 13828403XXXX
-- | --
Production Year | 2022
Production Week | 40
Model | HMT-2250
Detected max. Power | 2,250 W
Bootloader Version | 0.0.1
Firmware Version | 1.0.10
Firmware Build Date | 2021-09-17 15:00:00
Hardware Part Number | 27179XXXX
Hardware Version | 01.00

tbnobody, 2023-06-12:
Und zensieren ist in dem Moment eher Kontraproduktiv...

vaterlangen, 2023-06-12:
Sorry, mach der Gewohnheit...hab's aktualisiert

Saarblick, 2023-06-12:
Suppi:
[image]

tbnobody, 2023-06-12:
Noch ein wichtiger Hinweis: Stellt bitte sicher beim Testen, dass in den Inverter Einstellungen keine Max. String Leistung eingestellt wurde. Der Parameter muss bei ALLEN Strings 0 sein. Sonst werden ggf. Strings ausgeblendet und das wird ggf. als Fehler interpretiert.

Hintergrund:
* Wenn alle Max. String Leistungen 0 sind werden alle Strings in der Live View angezeigt
* Wenn die Leistung eines oder mehrerer Strings >0 ist, werden alle Strings mit 0 ausgeblendet

Wenn also z.B. in einer alten Version String 1 physikalisch belegt war und dort auch eine Max. String Leistung eingegeben wurde, dann wird in der aktuellen Version der physikalische String 1 (der jetzt ja 6 ist) ausgeblendet und der String 6 (der jetzt 1 ist) eingeblendet (und ist in diesem Moment ja 0)

Jean-GO, 2023-06-13:
[image]

Firmwareversion: v23.6.11
Reihenfolge von der Antenne an 6-1
Reihenfolge in der DTU 6-1

tictrick, 2023-06-13:
Ich nehme mal an es ist noch verfrüht, nach 3 Invertern schon Schlüsse zu ziehen, aber bis jetzt ...

Hardware-Teilenummer 271790337 hat Antenne 6-1 und DTU 1-6
Hardware-Teilenummer 271790338 hat Antenne 6-1 und DTU 6-1

271790337 scheint die ältere Hardware zu sein, was sich auch dadurch bestätigt, dass es früher in der DTU passte und aufgrund des Issues für 271790338 geändert wurde.

Jean-GO, 2023-06-13:
Wie ist das jetzt geneint, der Sting ind der DTU oder die Live-Ansicht, bei mir ist es so:
[image]

harmstorf, 2023-06-13:
Ist "Anschluss 6" der an der Antenne und bezieht sich auf die tatsächliche Beschriftung am Wechselrichter, oder auf die im Handbuch?

Jean-GO, 2023-06-13:
Wenn ich am Wechselrichter der Anschluss 6 abziehe, ist der Anschluss also Sting 1 bei mir.

harmstorf, 2023-06-13:
Mein HMT-2250 hat die Hardware-Teilenummer 271790337 und seit der Code-Änderung (um nicht "Fix" zu sagen) ist alles korrekt. Details meines Setups habe ich bereits hier geteilt: https://github.com/tbnobody/OpenDTU/issues/1005#issuecomment-1586733586

tbnobody, 2023-06-13:
D.h. dass @harmstorf und @Saarblick die gleichen Inverter besitzen (Gleiche Hardware ID, Produktionswoche, Firmware etc.) aber die Reihenfolge der Strings unterschiedlich ist. Wenn das wirklich so stimmt (tut das ggf. nochmal kritisch prüfen) kann man die beiden Reihenfolgen nicht auseinander halten.

vaterlangen, 2023-06-13:
Oder es gibt evtl eine zweite Fertigung, die das Ding anders zusammenbaut. Müsste man dann aber eigentlich an der S/N erkennen können...
Liergt die von  @harmstorf schon irgendwo vor?

harmstorf, 2023-06-13:
Ich war eben auf dem Dach und habe das Panel, welches ich als "6" gekennzeichnet habe, abgezogen. OpenDTU meldet dass Panel 6 nichts produziert - soweit also korrekt. Dann habe ich am Wechselrichter die Verbindung vom mit "6" (bei der Antenne) gekennzeichneten Anschluss getrennt. Und wie erwartet war in OpenDTU wieder Panel 6 ohne Produktion.

Ich weiß nicht, ob Hoymiles die Geräte vollständig automatisiert produziert oder ob es während einer manuellen Produktion zu Verdrahtungsfehlern kommen kann. Für den Betrieb ist die Zuordnung ja irrelevant, lediglich für die Optimierung hilfreich.

homeautomation2022, 2023-06-13:
Die Platine sollte gleich aussehen bei allen HMT´s und die Kabel sind nur reingeführt und dann direkt angelötet, da kann nichts verdreht sein.
Das kann nur in der Firmware gedreht sein, aber warum? Am Anfang hat man ja offenbar noch die Prototypen falsch herum beschriftet und sich dann darauf geeinigt oben ist die 6 unten die 1.

homeautomation2022, 2023-06-13:
> 

Das klingt interessant und sollte verfolgt werden.

homeautomation2022, 2023-06-13:
> Das klingt interessant und sollte verfolgt werden.

> Ich nehme mal an es ist noch verfrüht, nach 3 Invertern schon Schlüsse zu ziehen, aber bis jetzt ...
>
> Hardware-Teilenummer 271790337 hat Antenne 6-1 und DTU 1-6 Hardware-Teilenummer 271790338 hat [...]

Das klingt eher richtig und sollte mal mit mehreren hier validiert werden.

tbnobody, 2023-06-13:
> Das klingt eher richtig und sollte mal mit mehreren hier validiert werden.

Ist leider nicht so, da sowohl @harmstorf als auch @Saarblick die Teilenummer 271790337  haben

harmstorf, 2023-06-13:
Wäre es sonst mit geringem Aufwand möglich, eine generische Investierung der Kanäle zu implementieren? Hardcoded die Default-Zuordnung und per Switch in den WR-Settings dann die Nummerierung umdrehen? Damit wäre vermutlich auf einem Schlag erstmal allen ein Werkzeug in die Hand gegeben, die auflaufenden Daten korrekt zuzuordnen.

tictrick, 2023-06-13:
Hier einfach was zu implementieren würde doch nur dafür sorgen, dass das Problem größer statt kleiner wird.

Da der HMT 2250 drei MPPT Tracker hat, sollte es doch so sein, dass wenn ich Modul 1+2 abziehe dann eine Meldung kommt das MMT-Tracker 1 keine Spannung hat. Mit dieser Annahme wird es doch dann so sein, dass diese Meldungen auch falsch zugeordnet sein werden wenn einfach mal was verdreht wird.

tbnobody, 2023-06-13:
> Wäre es sonst mit geringem Aufwand möglich, eine generische Investierung der Kanäle zu implementieren?

Nein, absolut nicht!

harmstorf, 2023-06-13:
Wurde mit https://github.com/tbnobody/OpenDTU/commit/593a33020f4b51bbb63c35ab16312c43c2452930 noch jemandem außer mir geholfen, oder hat die Mehrheit nun Probleme? Ich würde mir sonst die Änderung als Patch beiseitelegen und vorerst meine eigenen Builds bei Updates nehmen, bis eine für alle passende Lösung gefunden ist.

Saarblick, 2023-06-13:
> 

Ich habe jetzt das Phänomen das wenn ich in der Konfig die 6 Kanäle (auch die nicht belegten) nicht mit einer max. Wattzahl angebe, verschwinden auch belegte Kanäle. Also scheint da auch was durcheinander zu kommen.

dragricola, 2023-06-14:
Ich habe 2 HMT2250, die ich im März 2023 gekauft habe. Der Eingang an der Antenne hat die Nummer 6. Bis zum Update von v23.6.1 auf v23.6.11 hat die Zuordnung der Eingänge in der DTU mit der Nummerierung am Gahäuse  übereingestimmt. Mit v23.6.11 hat sich auch bei mir die Zuordnung der Eingänge in der DTU umgekehrt. Die WR-Daten sind bei beiden Invertern bis auf die Seriennummern identisch:
[image]
Nachdem ich v23.6.1 wieder installiert habe, stimmt die Zuordnung wieder.

harmstorf, 2023-06-14:
Passt für mich. Werde entsprechend die Beschriftungen an den Kabeln und OpenDTU anpassen. Letztlich alles nur für die Visualisierung relevant, nicht für die Funktion.

Saarblick, 2023-06-17:
Danke, nach der Umstellung auf die alte Version passt auch die Schnittstelle zur Solaranzeige wieder

closed 2023-06-23 by tbnobody