tbnobody/OpenDTU #1032: DTU command failed

Bankaifan, 2023-06-13:
### What happened?

I upgraded to version v23.6.11. Since then the system seems to become instable after several hours of uptime.
The gui then starts to become slow and some minutes later unreachable.
Before it becomes unreachable I see this messages for some or all inverters:
[image]

The system is often reachable again after ~2-5 minutes - I then see "software reset cpu" as reset-reason.

After this state, the system stays instable and keeps crashing till i powercycle it.

### To Reproduce Bug

I have no idea right now.
I have 3 HM-300 inverters, MQTT is active.

### Expected Behavior

Should stay stable

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

ee4b479

### Relevant log/trace output

```Shell
Websocket: [/livedata][17] disconnect
16:18:27.600 > Fetch inverter: 112183218595
16:18:28.259 > TX RealTimeRunData Channel: 40 --> 15 83 21 85 95 80 19 47 92 80 0B 00 64 88 7A AF 00 00 00 00 00 00 00 00 DB C8 4A 
16:18:28.508 > Interrupt received
16:18:28.676 > RX Channel: 75 --> 95 83 21 85 95 83 21 85 95 82 13 89 06 D0 00 01 00 4A 03 E8 01 A3 00 1C 5A C2 DD | -80 dBm
16:18:28.870 > RX Period End
16:18:28.870 > Middle missing
16:18:28.870 > Request retransmit: 1
16:18:28.870 > TX RequestFrame Channel: 61 --> 15 83 21 85 95 80 19 47 92 81 6A 
16:18:29.029 > Interrupt received
16:18:29.229 > RX Channel: 23 --> 95 83 21 85 95 83 21 85 95 01 00 01 01 65 01 FF 07 22 00 01 B8 FC 09 66 09 26 2F | -80 dBm
16:18:29.410 > RX Period End
16:18:29.410 > Success
16:18:30.527 > Fetch inverter: 112183823895
16:18:32.026 > TX RealTimeRunData Channel: 75 --> 15 83 82 38 95 80 19 47 92 80 0B 00 64 88 7A B4 00 00 00 00 00 00 00 00 2B 76 01 
16:18:32.154 > Interrupt received
16:18:33.546 > RX Channel: 23 --> 95 83 82 38 95 83 82 38 95 82 13 89 0B B9 00 03 00 80 03 E8 02 28 00 06 48 9F AC | -80 dBm
16:18:34.207 > RX Period End
16:18:34.207 > Middle missing
16:18:34.207 > Request retransmit: 1
16:18:34.207 > TX RequestFrame Channel: 3 --> 15 83 82 38 95 80 19 47 92 81 74 
16:18:34.366 > Interrupt received
16:18:34.798 > RX Channel: 40 --> 95 83 82 38 95 83 82 38 95 01 00 01 01 59 03 8F 0C 47 00 02 3D D5 07 75 09 27 BC | -30 dBm
16:18:34.949 > RX Period End
16:18:34.949 > Success
16:18:35.088 > Fetch inverter: 112183213583
16:18:35.211 > TX RealTimeRunData Channel: 23 --> 15 83 21 35 83 80 19 47 92 80 0B 00 64 88 7A BA 00 00 00 00 00 00 00 00 4B 3A 9B 
16:18:35.991 > Interrupt received
16:18:36.927 > RX Channel: 61 --> 95 83 21 35 83 83 21 35 83 01 00 01 01 57 03 8D 0C 28 00 00 80 F0 07 7E 09 25 4C | -80 dBm
16:18:37.105 > Interrupt received
16:18:38.574 > RX Channel: 61 --> 95 83 21 35 83 83 21 35 83 82 13 89 0B 9C 00 00 00 7F 03 E8 02 3C 00 09 E6 22 7D | -80 dBm
16:18:39.448 > RX Period End
16:18:39.448 > Success
16:18:41.295 > Fetch inverter: 112183218595
16:18:43.068 > TX RealTimeRunData Channel: 40 --> 15 83 21 85 95 80 19 47 92 80 0B 00 64 88 7A C0 00 00 00 00 00 00 00 00 29 21 3E 
16:18:43.216 > RX Period End
16:18:43.216 > All missing
16:18:43.216 > Nothing received, resend whole request
16:18:43.216 > TX RealTimeRunData Channel: 61 --> 15 83 21 85 95 80 19 47 92 80 0B 00 64 88 7A C0 00 00 00 00 00 00 00 00 29 21 3E 
16:18:45.275 > Interrupt received
16:18:45.658 > RX Channel: 40 --> 95 83 21 85 95 83 21 85 95 01 00 01 01 65 01 FF 07 1E 00 01 B8 FD 09 67 09 25 10 | -30 dBm
16:18:45.830 > Interrupt received
16:18:45.947 > RX Channel: 3 --> 95 83 21 85 95 83 21 85 95 82 13 88 06 CC 00 01 00 4A 03 E8 01 A4 00 1C A7 37 CF | -80 dBm
16:18:46.107 > RX Period End
16:18:46.107 > Success
16:18:46.488 > Fetch inverter: 112183823895
16:18:46.613 > TX RealTimeRunData Channel: 75 --> 15 83 82 38 95 80 19 47 92 80 0B 00 64 88 7A C5 00 00 00 00 00 00 00 00 79 1E 4A 
16:18:46.694 > Interrupt received
16:18:46.768 > RX Channel: 23 --> 95 83 82 38 95 83 82 38 95 01 00 01 01 5A 03 8D 0C 43 00 02 3D D6 07 76 09 28 B6 | -80 dBm
16:18:48.258 > Interrupt received
16:18:48.487 > RX Channel: 40 --> 95 83 82 38 95 83 82 38 95 82 13 89 0B B6 00 03 00 80 03 E8 02 28 00 06 6A CC D2 | -80 dBm
16:18:49.534 > RX Period End
16:18:49.534 > Success
16:18:52.089 > Fetch inverter: 112183213583
16:18:52.182 > TX RealTimeRunData Channel: 3 --> 15 83 21 35 83 80 19 47 92 80 0B 00 64 88 7A CB 00 00 00 00 00 00 00 00 19 52 D0 
16:18:52.579 > RX Period End
16:18:52.579 > All missing
16:18:52.579 > Nothing received, resend whole request
16:18:52.579 > TX RealTimeRunData Channel: 23 --> 15 83 21 35 83 80 19 47 92 80 0B 00 64 88 7A CB 00 00 00 00 00 00 00 00 19 52 D0 
16:18:53.722 > RX Period End
16:18:53.722 > All missing
16:18:53.722 > Nothing received, resend whole request
16:18:53.722 > TX RealTimeRunData Channel: 40 --> 15 83 21 35 83 80 19 47 92 80 0B 00 64 88 7A CB 00 00 00 00 00 00 00 00 19 52 D0 
16:18:56.202 > Interrupt received
16:18:56.881 > RX Channel: 3 --> 95 83 21 35 83 83 21 35 83 82 13 87 0B 9C 00 00 00 7F 03 E8 02 3C 00 09 36 8B 0A | -80 dBm
16:18:57.037 > RX Period End
16:18:57.037 > Middle missing
16:18:57.037 > Request retransmit: 1
16:18:57.037 > TX RequestFrame Channel: 61 --> 15 83 21 35 83 80 19 47 92 81 CC 
16:18:57.303 > Interrupt received
16:18:57.482 > RX Channel: 3 --> 95 83 21 35 83 83 21 35 83 01 00 01 01 56 03 8F 0C 28 00 00 80 F1 07 7F 09 23 49 | -80 dBm
16:18:59.179 > RX Period End
16:18:59.179 > Success
16:18:59.315 > Fetch inverter: 112183218595
```

Bankaifan, 2023-06-13:
- Previous Version was from 6.6.2023 i think it was 23.6.1
- Poll interval is 5 seconds
- MQTT publisch interval is 5 seconds too
- No, its just me calling the web interface.

Serial log is not provided via the gui, is it?
I can try to get it later via. arduino Code

tbnobody, 2023-06-13:
Just see: https://github.com/tbnobody/OpenDTU/wiki/FAQ#q-how-to-observe-the-esp32-boot-process-by-looking-at-the-serial-console

Just compared 23.6.1 with 23.6.11 
https://github.com/tbnobody/OpenDTU/compare/v23.6.1...v23.6.11

There was nothing included which could lead to such errors.... Did you try to perform a complete power cycle (not just software restarts)

tobiasboecker, 2023-06-13:
Well, having a very similar issue. Unfortunately I bricked my openDTU (USB port is some missplaced :-| . So for the moment I am not able to to post some logs.

The problem begins as soon as the inverters connect to grid. I had the inverters installed (without grid connection) and got all data by MQTT. After givin grid connection the problems begann. After some Testing with dbm power and placing (including USB bricking) the situation was the same . Firmware Version I think was v23.5.3x. The issue even present after a complete cycle this morning - only message was, inverter start and after a few hours the inverters begann to flap between available or nor. Im some cases after a few minutes or after 2 hours I got some data. And it staret all over again.... maybe it is not just a problem with placing the open DTU

Inverter 2xHMS-1800

Bankaifan, 2023-06-15:
Not it fails again...

I see a a few minutes without any logging-activity:
08:22:00.161 -> Interrupt received
08:22:00.161 -> RX Channel: 3 --> 95 83 82 38 95 83 82 38 95 82 13 88 02 60 00 00 00 1A 03 E8 00 D9 00 0A 27 4B A0 | -80 dBm
08:22:00.255 -> RX Period End
08:22:00.255 -> Success
08:22:00.255 -> TX SystemConfigPara Channel: 75 --> 15 83 82 38 95 80 19 47 92 80 05 00 64 8A AE 06 00 00 00 00 00 00 00 00 86 1B AB 
08:22:00.287 -> Interrupt received
08:22:00.287 -> RX Channel: 23 --> 95 83 82 38 95 83 82 38 95 81 00 01 03 E8 00 00 03 E8 FF FF FF FF 01 68 7D F8 F9 | -80 dBm
08:22:00.519 -> RX Period End
08:22:00.519 -> Success
08:26:12.485 -> Websocket: [/livedata][1] connect
08:26:13.997 -> Fetch inverter: 112183823895
08:26:13.997 -> Request SystemConfigPara
08:26:14.073 -> TX RealTimeRunData Channel: 23 --> 15 83 82 38 95 80 19 47 92 80 0B 00 64 8A AF 05 00 00 00 00 00 00 00 00 7D 51 16 
08:26:14.597 -> RX Period End
08:26:14.597 -> All missing
08:26:14.597 -> Nothing received, resend whole request
08:26:14.630 -> TX RealTimeRunData Channel: 40 --> 15 83 82 38 95 80 19 47 92 80 0B 00 64 8A AF 05 00 00 00 00 00 00 00 00 7D 51 16 
08:26:15.182 -> RX Period End
08:26:15.182 -> All missing

And at the time of the error:
**08:26:18.893 -> Request retransmit: 1
08:26:18.893 -> TX RequestFrame Channel: 75 --> 15 83 82 38 95 80 19 47 92 81 74 
08:26:19.020 -> RX Period End
08:26:19.020 -> Middle missing
08:26:19.020 -> Retransmit timeout
08:26:19.020 -> TX SystemConfigPara Channel: 3 --> 15 83 82 38 95 80 19 47 92 80 05 00 64 8A AF 05 00 00 00 00 00 00 00 00 B3 5E D9 
08:26:19.200 -> RX Period End
08:26:19.200 -> All missing
08:26:19.200 -> Nothing received, resend whole request
08:26:19.200 -> TX SystemConfigPara Channel: 23 --> 15 83 82 38 95 80 19 47 92 80 05 00 64 8A AF 05 00 00 00 00 00 00 00 00 B3 5E D9 
08:26:19.458 -> RX Period End
08:26:19.458 -> All missing
08:26:19.458 -> Nothing received, resend whole request
08:26:19.458 -> TX SystemConfigPara Channel: 40 --> 15 83 82 38 95 80 19 47 92 80 05 00 64 8A AF 05 00 00 00 00 00 00 00 00 B3 5E D9 
08:26:19.665 -> RX Period End
08:26:19.665 -> All missing
08:26:19.665 -> Nothing received, resend whole request
08:26:19.665 -> TX SystemConfigPara Channel: 61 --> 15 83 82 38 95 80 19 47 92 80 05 00 64 8A AF 05 00 00 00 00 00 00 00 00 B3 5E D9 
08:26:19.948 -> RX Period End
08:26:19.948 -> All missing
08:26:19.948 -> Nothing received, resend whole request
08:26:19.948 -> TX SystemConfigPara Channel: 75 --> 15 83 82 38 95 80 19 47 92 80 05 00 64 8A AF 05 00 00 00 00 00 00 00 00 B3 5E D9 
08:26:20.179 -> RX Period End
08:26:20.179 -> All missing
08:26:20.179 -> Nothing received, resend count exeeded
08:26:20.179 -> Fetch inverter: 112183213583**

grisu3210, 2023-06-16:
I also had these messages but after upgrading to 23.6.14 they do not appear any more

his1220, 2023-06-19:
I had this messages in all versions since April? There is a lot of lag and from time to time nearly every day 1 or 2 there is a network failure completely where I have to restart the ESP

Bankaifan, 2023-06-19:
I didn't had the issue again since I installed v23.6.14
<https://github.com/tbnobody/OpenDTU/releases/tag/v23.6.14>

Daniel Kaiser

*Tel: 0151 42341894*

his1220, 2023-06-19:
I will try a reinstall !

Hermarcel, 2023-06-21:
Issue occurs here as well. Fusion V2 with 4 HMS-2000-4T's.

Bankaifan, 2023-06-26:
Was quiet now for several days, now I had the problem again.

tklostermannNSD, 2023-06-27:
Looks like its the same as https://github.com/tbnobody/OpenDTU/issues/910

fila612, 2023-07-20:
currently with 23.7.10 no more effects since nearly two weeks.

pile001, 2023-08-14:
Hi, I have a similar problem. Every day but at different times comes exactly 15 times the same message. Currently I use the firmware v23.8.8. The problem was also with at least 4 firmware versions before.

[image]

[image]

koalabamu, 2023-09-03:
got a brandnew DTU and HMS-2000 - same here
[image]

stefan123t, 2024-07-10:
Please note that we have renamed the Event Log Alarm Code `2` from `DTU command failed` to `Time calibration`.
This has been learned from the [warn_code.json](https://github.com/tbnobody/OpenDTU/issues/121#issuecomment-1752124208 ) in the S-Miles Installer App's APK file and concluded in #121.

@Bankaifan IMHO this may be something like the command sent to the inverter which contains the Timestamp has been delayed too much and does not match the internal timestamp of the inverter.
This may happen when you have a bad or only somewhat working connection between the DTU and the inverter and the DTU has to repeat the command(s) very often for a longer period of time until the complete response is received from the inverter.

I have the same when the inverter does not answer reliably, e.g. in the evening hours I see up to 15 Events in the Event Log:

##### Event Log

Start | Stop | ID | Message
-- | -- | -- | --
05:32:31 | 05:32:31 | 1 | Inverter start
15:13:41 | 15:13:41 | 2 | Time calibration
18:16:02 | 18:16:02 | 2 | Time calibration
18:16:55 | 18:16:55 | 2 | Time calibration
18:45:18 | 18:45:18 | 2 | Time calibration
18:45:24 | 18:45:24 | 2 | Time calibration
19:04:15 | 19:04:15 | 2 | Time calibration
19:04:52 | 19:04:52 | 2 | Time calibration
19:05:08 | 19:05:08 | 2 | Time calibration
19:07:15 | 19:07:15 | 2 | Time calibration
19:07:31 | 19:07:31 | 2 | Time calibration
19:09:47 | 19:09:47 | 2 | Time calibration
19:10:09 | 19:10:09 | 2 | Time calibration
19:10:27 | 19:10:27 | 2 | Time calibration
19:10:43 | 19:10:43 | 2 | Time calibration

@tbnobody do we want to give the Alarm Number in our [AlarmData / AlarmUpdate](https://github.com/lumapu/ahoy/wiki/Protocol#wie-sieht-das-device-info-kommando-alarm-data-0x11--alarm-update-0x12-aus-) Queries another try ?

```
15 74403329 78563412 80 1200 62D80183 0000 <000F> 00000000 FFC4 2A --- AlarmUpdate 0x12
^^------------------------------------------------------------------ MainCmd 0x15 REQ_ARW_DAT_ALL
   ^^^^^^^^--------------------------------------------------------- WR Serial ID
            ^^^^^^^^------------------------------------------------ DTU Serial ID
                     ^^--------------------------------------------- MultiFrameID 0x80
                        ^^------------------------------------------ SubCmd bzw. DataType: 0x11 = AlarmData, 0x12 AlarmUpdate
                          ^^---------------------------------------- rev Protocol Revision ?
                             ^^^^^^^^------------------------------- UNIX timestamp 62BE1CE0 -> 2022-07-01 00:00:00
                                      ^^^^-------------------------- Gap always 0x0000
                                           ^^^^--------------------- 0x0000, nur bei AlarmData: WarnSerNub (Warning Serial Number)  
                                                ^^^^^^^^------------ Password always 0x0000
                                                         ^^^^------- CRC16 / CRC-Modbus über die UserData, excl. Frame ID!
                                                              ^^---- CRC8
```

What happens if we acknowledge some of the Alarms we have received using the AlarmUpdate Command sending back the WarnSerNub (Warning Serial Number) from the AlarmData message we received earlier ?

In some of the data blobs we have gathered initially there is also AlarmData1 and AlarmData2, which is not yet understood or parsed. 

@tbnobody could we introduce a data donation window like we have it with the Grid Profile in order to gather some more raw AlarmData responses from the real world ? 

It is pretty difficult to catch the AlarmData in the Console and format it into 12 bytes.

Especially with the above WCode `0x02` `Time calibration` we do frequently see the fields AlarmData1 and AlarmData2 being set to 0xffff and 0xfffb as in the below example.

For me this looks like the offset between the timestamps received from the OpenDTU and the inverters internal clock, i.e. +/- 5 seconds.

```
b0 02 00 48 4c f7 4c f7 00 00 00 05 // 0xB0 = PM,PM, 0x4CF7 = 06:28:23 + 12:00:00 = 18:28:23
b0 02 00 49 55 da 55 da ff ff ff fb // 0xB0 = PM,PM, 0x55DA = 07:06:18 + 12:00:00 = 19:06:18
```

closed 2025-04-18 by github-actions[bot]