tbnobody/OpenDTU #1034: [Request] Scheduled inverter reboot

Jamie-Jameson, 2023-06-14:
### Is your feature request related to a problem? Please describe.

Hi there,

I just started with an HMS-600 and OpenDTU a couple of days ago. I had very hard times setting up the system because after a day or so I didn't receive any data from the inverter anymore. The log always shows "all missing" and "nothing received...". All values in the live display stayed at 0.0, and also no information about the inverter was available.

It all looked like an issue with the soldering, but we didn't find anything. Then, just for fun I triggered a "reboot" via the OpenDTU web interface. And magically, all data was received and displayed instantly.

### Describe the solution you'd like

In the inverter setup, add an option to reboot the inverter regularly, maybe every N days or something.
I propose this in the "Edit inverter" section, maybe below the "Receive / Send" configuration section with the following values:
Reboot automatically every (N) days <-- combobox, with default Zero OR have a calender thing with a day selection?
Reboot time: <-- either type a time, OR even better, an option to do this during night time (as given from the day/night calculation)

Solarteur, 2023-06-14:
What do you want to automatically restart - the inverter or OpenDTU?
You write about the inverter, but you probably mean OpenDTU.
With a stable setup (solder connections, cables if necessary) and a stable power supply, OpenDTU will run for weeks without the need for a reboot.
Therefore, in my view, there is no need for an automatic reboot.

tbnobody, 2023-06-14:
I fully agree with @Solarteur it does not make sense to create evil hacks or bad workarounds if it just fixes symptoms but not the reason of the problem.

closed 2023-06-14 by tbnobody

helgeerbe, 2023-06-14:
On openDTU-onBattery we have implemented a scheduled reboot of the inverter. Problem seems to be, that permanently powered by a battery, the inverters show some strange behaviors over time (> 1 day). A planned daily reboot seems to fix this. As a benefit, the daily power counter becomes useful.

Jamie-Jameson, 2023-06-14:
I mean the inverter actually. And I don't know that the issue was exactly, but when I rebooted the inverter from within OpenDTU everything suddenly worked as expected.

The problem is that there is no way I could solve anything with the inverter, since it is proprietary obviously. That's why would say that there might be use cases / setups where an automatic reboot makes life easier for the user.

Plus, @helgeerbe mentioned issues with battery setups, but at least this is another use case there it solves real-life issues.