tbnobody/OpenDTU #1037: Interface / Life-Bit zum Nulleinspeise - Script mit automatischer Leistungsanpassung im Fehlerfall.

SMGatAUXde, 2023-06-15:
### Is your feature request related to a problem? Please describe.

Hallo, 
ich nütze die OpenDTU im Zusammenspiel mit einem Regelungs-Script (HoymilesZeroExport - sehr empfehlenswert) zur Nulleinspeisung wg.  Batteriebetrieb.

Nun gibt es durchaus mal Verbindungsabbrüche oder Ausfälle. 
Meine Sorge ist nun, dass es einen Ausfall gibt, im ungünstigen Fall bei einer hohen Leistung und die Regelung nicht mehr die OpenDTU erreicht.

### Describe the solution you'd like

Für diesen Fall wäre es gut, wenn es eine Möglichkeit gäbe, dass die DTU den WR auf einen vorgewählten Leistungsfaktor zurückstellt. Zum Beispiel in der Höhe der Grundlast. 
Ggf. auch Tageszeitabhängig, wenn man davon ausgehen kann, dass die Power von der PV und nicht vom Akku kommt.

Meine Vorstellung hierzu wäre, dass die OpenDTU und das Regelungsscript ihre gegenseitig Verbindung verifizieren sozusagen eine Form von Life-Bit austauschen.

Hat hier jmd. Vorschläge; Könnte man das Implementieren? Würde mich hierzu auch an die andere Seite (HoymilesZeroExport) bzgl. Umsetzung wenden.

Gruß

SMGatAUXde

Dom1n1c, 2023-07-11:
Try to set a more restrictive power limit. Normally it shouldn't be an issue to ,donate‘ up to 600W (in 2024 800W) to the grid, if it's a small inverter (e.g. HM600/800)

I can't relate why we should overcomplicate things just for a few cents. Some people use water heaters or other appliances in order to increase the power consumption :D 

My opinion is: Try to use as much as possible, but don't be afraid to add some regenerative energy to the grid. Optimizing the appliances to run whenever is a good way. Some were also using bigger inverters with some restrictive limits. Example: HM1200/1600 with 4*380Wp, limited to 600W. You will be limited to those smaller values, but you will be able to use more of the solar energy throughout the day

stefan123t, 2024-07-10:
@SMGatAUXde please double check the **Dynamic Power Limiter (DPL)** function in the daughter project[ OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/).

I have described a [possible PM+DPL setup](https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352) with a Shelly Pro3EM as **Power Meter (PM)** here in issue #272.

@3DJupp the example I have given sets the Export donation to -300W, i.e. half the allowed Export of -600 W (in 2024 -800W), as I agree that we should always strive to give something back to the community.

Actually I would love to setup a **cooperative** that we contribute our exports to and that invoices these exports as a virtual cooperative PV power plant. The proceedings from these donations can then be used to subsist our local maker spaces.