tbnobody/OpenDTU #1038: Thank you / sponsorships?

suhlig, 2023-06-15:
### Is your feature request related to a problem? Please describe.

Hello, I would like to thank the creators and maintainers of this project. It took less than half an hour to connect to my inverter and get meaningful data.

Do you accept donations or GitHub sponsorships? If not, do you have a preference which org I should donate some money to (e.g. Doctors Without Borders)?

### Describe the solution you'd like

A statement by the maintainers whether they accept donations, and if not, where to send money to instead.

### Describe alternatives you've considered

Looked for Kofi, GH sponsorship, donation statements in the README and discussions.

### Additional context

I am happy with the project and would like to donate some money as a "Thank you" gesture.

DcFamas, 2023-06-15:
On the mainpage:

[image]

https://ko-fi.com/tbnobody

suhlig, 2023-06-15:
Not sure if I overlooked it, or if it is new, but anyway: Thanks!

closed 2023-06-15 by suhlig