tbnobody/OpenDTU #1042: after starting flashing this file with the erase-flasher following happened

DiskoFieber, 2023-06-16:
after starting flashing this file with the erase-flasher following happened

**Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)**

is there something wrong with the file or did I make a mistake?

Hardware
NMCU-ESP32
NRF24L01+

_Originally posted by @DiskoFieber in https://github.com/tbnobody/OpenDTU/discussions/1040_

DPO99, 2023-06-16:
Can yo uplease try to Flash with "Espressif Flash Download Tool (Windows)"?

https://www.espressif.com/en/support/download/other-tools

DiskoFieber, 2023-06-16:
Yes, i‘ve tried this Tool but no succsess. it end with this textABS_DONE_0 FalseI don‘t wat that means

https://www.espressif.com/en/support/download/other-tools

DPO99, 2023-06-16:
@tbnobody ggf. was für Dich?

tbnobody, 2023-06-16:
I've just doublechecked it and it worked. That means that it's most likely not related to the generated .bin file
[image]

Could be related to the ESP itself, to the cable or a defect of the board. (We already hat several defect boards in the previous time.)

You could try to erase the ESP (press the erase button)
This has to work. It could also be that the ESP does not just enter the flash mode automatically. In this case you can try to press the "BOOT" button before you press Erase or Flash.

DiskoFieber, 2023-06-17:
I have tested two boards and both have the same problem.
I can only flash the generic.bin with the ESP flasher, but with many errors and also not configurable.
But I can flash the Ahoy project with their flasher and also configure it. This runs stable on both boards. Ergo, both boards and the cable work

tbnobody, 2023-06-17:
Which URL did you use for the download and how large is the .bin file you are using?

DiskoFieber, 2023-06-17:
I use this Link 

https://github.com/tbnobody/OpenDTU/suites/13610629568/artifacts/750444834

The file size 1,5 MB or did you need byte counting?

I know that the generic.bin only goes ota. I just did it to see if the tools work

DiskoFieber, 2023-06-22:
I solved the problem. 
I erased the flash memory with the ESPtool via the command line and reloaded the factory.bin in the same way. 
After deleting the memory, I couldn't reload the factory.bin with the ESP-flasher or the flash-download-tool. 
This was only possible on the command line level.

tbnobody, 2023-06-23:
This is interesting because I would have expected both Flash tools to do the same thing.

closed 2023-06-23 by tbnobody