tbnobody/OpenDTU #1044: Bei Wechselrichter nur Phase 1-änderung?

Hubert1975, 2023-06-16:
### Is your feature request related to a problem? Please describe.

Bei Wechselrichter nur Phase 1-änderung in diesem Feld möglich? siehe Bild.

### Describe the solution you'd like

Bei Wechselrichter Änderungsmöglichkeit Phase 1, 2, 3 oder Farben schwarz, braun .....
[image]

DPO99, 2023-06-16:
Ich glaube du verwechselst den Sinn dieser Anzeige. Die ist an sich nur dafür da, die Phasen des HM anzuzeigen. Du möchtest aber quasi angeben, an welchem Leiter L1,L2,L3 das Ding an Deinem Netz hängt oder?

tbnobody, 2023-06-17:
Das  was @DPO99 sagst ist korrekt. bei HM/HMS invertern gibt es nur eine Phase. Diese hat auch nur diesen einen Namen. Bei HMT Invertern sieht das anders aus. Diese Zuordnung wird sich aber ggf. in zukunft noch ändern.

closed 2023-06-17 by tbnobody

Hubert1975, 2023-06-19:
Ja ich habe 3 Wechselrichter Hm 1500 und jeder auf einer Phase, damit im Haus gut aufgeteilt.... deswegen meine Frage!