tbnobody/OpenDTU #1046: Unbekanntes Model: TSUN TSOL-M800

HansWuer, 2023-06-16:
### What happened?

Unbekanntes Model: TSUN TSOL-M800

Hardware-Teilenummer | 65536000

### To Reproduce Bug

Unbekanntes Model: TSUN TSOL-M800

### Expected Behavior

Normale Anzeige für  TSUN TSOL-M800 - Daten können ausgelesen werden

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

23.6.2

homeautomation2022, 2023-06-18:
Bitte Screenshot der ganzen Inverter-Informationen (ohne SN) posten, vermutlich behebt ein Neustart der DTU bereits das Ausleseproblem. Irgendwo her kommt immer mal wieder ein fehlgepartses Paket, Plausiblitäts-Check wäre auch nur eine halbe Lösung, richtig wäre die Ursache im Log zu finden, aber ist schwer zu provozieren...

tbnobody, 2023-06-23:
Closing this issue due to inactivity.

closed 2023-06-23 by tbnobody