tbnobody/OpenDTU #1050: [Request] Is it possible to add "General topics" and "Inverter total topics" to MQTT Topics for Home Assistant?

januszwoj, 2023-06-18:
### Is your feature request related to a problem? Please describe.

No

### Describe the solution you'd like

I know that I can create entities/helpers and perform mathematical calculations to get, for example, the total sum of daily production.
But why if this data already exists in DTU and is sent via MQTT, only Home Assistant cannot see it.
Could you add these topics to: "Home Assistant MQTT Auto Discovery Parameters"?

### Describe alternatives you've considered

I have no idea

### Additional context

Thank you in advance and best regards.
Janusz

DejanBukovec, 2023-07-04:
I vote for this feature...
I think this will be simple task to do because MQTT data are already there...
Im myself manualy publish 4 configurations and simply get data to HA:
[image]

januszwoj, 2023-07-26:
> I vote for this feature... I think this will be simple task to do because MQTT data are already there... Im myself manualy publish 4 configurations and simply get data to HA: [image]

Hi DejanBukovec
Could you write how you did it? It would certainly come in handy before they will introduce it to the firmware.
Did you send these commands using MQTT-Explorer and what did they look like?
Thanks in advance.
Janusz

DejanBukovec, 2023-07-26:
Hi,
First Im send it by MQTT Explorer BUT after every HA restart they has been lost so I need resend it... Then Im make in HA automation which trigger after start.

Down in post will be my Automation in yaml. Please replace next names with your preffered ones:
YOUR_DTU_SERIAL - With your OpenDTU serial which you can see/set in "Settings" -> "DTU Settings" -> "Serial"
YOUR_DTU_NAME - With name of your DTU
YOUR_DTU_IP - With IP of your DTU
YOUR_DTU_VERSION - With your DTU version for example v23.7.22

```
alias: Solar MQTT Publish
description: ""
trigger:
  - platform: homeassistant
    event: start
condition: []
action:
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: homeassistant/sensor/dtu_YOUR_DTU_SERIAL/dc_power/config
      payload: |-
        {
            "name":"YOUR_DTU_NAME DC Power",
            "stat_t":"solar/dc/power",
            "uniq_id":"YOUR_DTU_SERIAL_PowerDC",
            "unit_of_meas":"W",
            "dev":{"name":"YOUR_DTU_NAME","ids":"YOUR_DTU_SERIAL","cu":"http://YOUR_DTU_IP","mf":"OpenDTU","mdl":"ESP32","sw":"YOUR_DTU_VERSION"},
            "exp_aft":81,
            "dev_cla":"power",
            "stat_cla":"measurement"
        }
    alias: Solar - DTU - DC Power
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: homeassistant/sensor/dtu_YOUR_DTU_SERIAL/ac_power/config
      payload: |-
        {
            "name":"YOUR_DTU_NAME AC Power",
            "stat_t":"solar/ac/power",
            "uniq_id":"YOUR_DTU_SERIAL_PowerAC",
            "unit_of_meas":"W",
            "dev":{"name":"YOUR_DTU_NAME","ids":"YOUR_DTU_SERIAL","cu":"http://YOUR_DTU_IP","mf":"OpenDTU","mdl":"ESP32","sw":"YOUR_DTU_VERSION"},
            "exp_aft":81,
            "dev_cla":"power",
            "stat_cla":"measurement"
        }
    alias: Solar - DTU - AC Power
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: homeassistant/sensor/dtu_YOUR_DTU_SERIAL/ac_yieldday/config
      payload: |-
        {
            "name":"YOUR_DTU_NAME Yield Day",
            "stat_t":"solar/ac/yieldday",
            "uniq_id":"YOUR_DTU_SERIAL_YieldDay",
            "unit_of_meas":"kWh",
            "dev":{"name":"YOUR_DTU_NAME","ids":"YOUR_DTU_SERIAL","cu":"http://YOUR_DTU_IP","mf":"OpenDTU","mdl":"ESP32","sw":"YOUR_DTU_VERSION"},
            "exp_aft":81,
            "dev_cla":"energy",
            "stat_cla":"measurement"        
        }
    alias: Solar - DTU - AC Yield Day
  - service: mqtt.publish
    data:
      qos: 0
      retain: false
      topic: homeassistant/sensor/dtu_YOUR_DTU_SERIAL/ac_yieldtotal/config
      payload: |-
        {
            "name":"YOUR_DTU_NAME Yield Total",
            "stat_t":"solar/ac/yieldtotal",
            "uniq_id":"YOUR_DTU_SERIAL_YieldTotal",
            "unit_of_meas":"kWh",
            "dev":{"name":"YOUR_DTU_NAME","ids":"YOUR_DTU_SERIAL","cu":"http://YOUR_DTU_IP","mf":"OpenDTU","mdl":"ESP32","sw":"YOUR_DTU_VERSION"},
            "exp_aft":81,
            "dev_cla":"energy",
            "stat_cla":"measurement"        
        }
    alias: Solar - DTU - AC Yield Total
mode: single
```

januszwoj, 2023-07-26:
> 
Hi DejanBukovec
Thank you very much. Everything works great.
I noticed only one problem, for sure in my case:
             "name":"YOUR_DTU_NAME Yield Day",
             "stat_t":"solar/ac/yieldday",
             "uniq_id":"YOUR_DTU_SERIAL_YieldDay",
             "unit_of_meas":"kWh", <----- in my case it should be "Wh"

Besides, it's great.
Thanks again.
Regards Janusz

AzraelsDisk, 2024-04-12:
vote 👍

stefan123t, 2024-07-10:
@januszwoj I am not sure whether I understand your Feature Request correctly:

You want us to add several total values to the future MQTT Topics on our list in #759 ?
* dtu/total/dc/power (solar/dc/power)
* dtu/total/ac/power (solar/ac/power)
* dtu/total/ac/yieldday (solar/ac/yieldday)
* dtu/total/ac/yieldtotal (solar/ac/yieldtotal)

Can you please review the list of MQTT Topics which are available, especially the [Inverter Total Topics](https://www.opendtu.solar/firmware/mqtt_topics/#inverter-total-topics) ?

@DejanBukovec what does your Home Assistant YAML actually do:

Does this setup a Dashboard Item with the four values or does it somehow calculate those and publish them back to the MQTT server ?

Why is the Home-Assistant Auto Discovery option not used / enough for your purpose of setting up a dashboard item ?

DejanBukovec, 2024-07-11:
It is old issue and don't remember what exactly is wrong but:
HA Auto Discovery has been enabled. It is create most MQTT topics but not ones which Im write on comment. I don't know if this has been fixed in last year...

stefan123t, 2024-07-11:
i understood from other issues you are using AhoyDTU at the moment and can not test / verify this anymore.

Let us wait for the OP @januszwoj to reply to his issue.

AzraelsDisk, 2024-07-11:
My problem is, that for each inverter the values are reported… but the current overall power is missing.. at the web fronted it is shown but not on mqtt

stefan123t, 2024-07-12:
> My problem is, that for each inverter the values are reported… but the current overall power is missing.. at the web fronted it is shown but not on mqtt

removed mail reply gibberish ;)

@AzraelsDisk yes this is correct the total values are sent via /api/livedata/status but not via mqtt.

Hence I have linked this issue with #759 add additional MQTT topics.

@tbnobody has to decide if we need to keep this open as a reminder too.

stefan123t, 2024-07-14:
IMHO they are either not implemented in the Home Assistant Auto-Discovery one-off routine, 
or they are not sent in general. I would expect the former.

tbnobody, 2024-10-03:
Implemented in 2fb026074ad8a89fcc2fe5ba894dfcba5402ea77

closed 2024-10-03 by tbnobody