tbnobody/OpenDTU #1060: [Request] Favicon

nonflammable, 2023-06-20:
### Describe the solution you'd like

Please add favicon for smartphones and browsers
[image]

closed 2023-06-26 by tbnobody

NintendoTito, 2023-07-27:
Can somebody describe where I can find the configuration for the touch icon? I can't find it.

NintendoTito, 2023-09-24:
Kann mir bitte jemand sagen, wie ich ein entsprechendes Favicon einstellen kann? Ich bekomme es nicht hin.

NintendoTito, 2023-11-10:
In der Hoffnung, dass ich hier nochmal ne Antwort bekomme... Wo stelle ich das Favicon ein?

stefan123t, 2023-11-11:
Das Favicon wurde im Sommer diesen Jahres zum Web Interface hinzugefügt. Es sollte also entsprechend geladen werden wenn es mit dern richtigen webkit/safari headern annonciert wird und auch die entsprechend von iOS erwarteten Bild-Formate vorliegen.
Vielleicht musst Du also die WebApp neu anlegen.

In der Hoffnung dass Du noch beachreibst was Du bereits versucht hast ?

NintendoTito, 2023-11-12:
> In der Hoffnung dass Du noch beachreibst was Du bereits versucht hast ?

Ich öffne die Weboberfläche der DTU in Chrome Android. Da gehe ich dann auf die 3 Punkte und wähle "zum Startbildschirm hinzufügen". Da erscheint dann aber kein Favicon.

NintendoTito, 2023-11-16:
@nonflammable Du hast meinen Kommentar geliked... Hast du das Favicon auch noch nicht?

nonflammable, 2023-11-16:
> @nonflammable Du hast meinen Kommentar geliked... Hast du das Favicon auch noch nicht?

No 🤷‍♂️

NintendoTito, 2023-11-16:
Mmh, interessant. Favicon soll vorhanden sein, Thema als gelöst markiert, "keiner" weiß wie es funktioniert. Naja, ist ja kein Dealbreaker. Soll auch hier nicht falsch rüberkommen, OpenDTU ist genial und danke an die Entwickler, ich will mich da nicht beschweren!

tbnobody, 2023-11-23:
@NintendoTito wird mit der nächsten Version behoben. Hatte bis heute kein Android Phone zur Hand um das zu testen. 
Hintergrund: Das Icon benötigt u.a. eine Mindestauflösung von 144x144 damit es dargestellt wird.

NintendoTito, 2023-12-21:
@tbnobody Nur als Feedback: Habe das aktuelle Update eingespielt (die Version mit Einführung der Grafik). Ein Favicon ist unter Android weiterhin nicht vorhanden. Muss ich was bestimmtes anpassen oder kann ich mit irgend nem Log oder Infos unterstützen?