tbnobody/OpenDTU #1068: Event Log: ID 12, Unknown

Hermarcel, 2023-06-23:
### What happened?

OpenDTU v23.6.21 with four HMS-2000-4T v01.00.20. Every inverter shows below message in it's event log:
[image]
Also with older versions.

### To Reproduce Bug

Just running the system

### Expected Behavior

Please, add explanation for ID 12?

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.21

### Anything else?

Used this link to download the firmware: https://discord.com/channels/984173303147155506/1060347063281401876/1121509260291997807

tbnobody, 2023-06-23:
If there would be an explanation for the ID 12 it would be mentioned there. So it's just not possible as it is not explained in any documentation.

closed 2023-06-25 by tbnobody

stefan123t, 2023-10-08:
@Hermarcel Danke für die Meldung, ja es gibt noch ein paar Unbekannte / unknown Fehlermeldungen.
U.a. 
* 12 (https://github.com/tbnobody/OpenDTU/issues/1068),
* 14 (https://github.com/tbnobody/OpenDTU/issues/131),
* 12, 13, 14, 36,  (https://github.com/tbnobody/OpenDTU/issues/121),
* 12, 14, 36, 46, 47, 73 https://github.com/tbnobody/OpenDTU/discussions/590

Ich bin in den `assets/warn_code.json` der Hoymiles S-Miles Installer App fündig geworden. 🥳 

Hier mal die bisher aufgetretenen Fehler oder Warnings aus dem in #121 angehängten warn_code.json (GitHub versteht nur ZIP ?).

11: Grid voltage surge
12: Grid voltage sharp drop
13: Grid frequency mutation
14: Grid phase mutation
36: INV overvoltage or overcurrent
46: FB overvoltage
47: FB overcurrent
71: VW function enable
72: FW function enable
73: TW function enable

Es wird von @tbnobody im Rahmen von #121 in OpenDTU implementiert und daher kann dieses Issue als Duplicate geschlossen werden.