tbnobody/OpenDTU #1071: MQTT with TLS support

Cybolord, 2023-06-23:
### What happened?

when configuring OpenDTU with MQTT target containing private CA certificate for active TLS v1.2 support connection / handshake will not work:

sudo tail -f /var/log/mosquitto/mosquitto.log

1687562945: New connection from 192.168.1.23 on port 8883.
1687562945: OpenSSL Error: error:14094412:SSL routines:ssl3_read_bytes:sslv3 alert bad certificate

### To Reproduce Bug

as soon as i activate TLS in MQTT with private CA certificate (=PEM coded) this behaviour appears....

### Expected Behavior

sudo tail -f /var/log/mosquitto/mosquitto.log

1687562967: New connection from 192.168.1.23 on port 8883.
1687562967: New client connected from 192.168.1.64 as OpenDTU-13824536 (c1, k15, u'cybolord').

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.21

### Relevant log/trace output

```Shell
please see sudo tail -f /var/log/mosquitto/mosquitto.log output above
```

### Anything else?

thank you for your affort in advance.

tbnobody, 2023-06-24:
This usually happens if the hostname which you entered in the OpenDTU MQTT settings does not match the CN of the server certificate of your broker.

Cybolord, 2023-06-24:
thank you for your qucik reply!

**CN is IP:192.168.64 plus SAN IP:192.168.64 and DNS record as well, but private CA signed...**

...meanwhile got a few more MQTT clients running TLS v1.2 against my broker like a charme BUT:
got function like "rejectUnauthorized=false"; does OpenDTU need an official signed CA or also private CA possible?
is there any available flag regarding "signed cert from private CA" acceptence?

Thx again.

tbnobody, 2023-06-24:
I tested it yesterday evening with a own CA. Worked perfectly. I just added the CA cert into the opendtu settings.

Cybolord, 2023-06-24:
pardon me: what du you mean exactly with "_ip is cut off_"? it is CN and SAN at same time...

Cybolord, 2023-06-24:
<html><body>

TLS | aktiv
-- | --
cert. version     : 3 serial number     : 01:C6:7B:60:66:A3:99:AD:3E:AA:E5:AA:4F:C3:04:6B:BA:E9:8D:27 issuer name       : C=DE, L=Cologne, CN=192.168.1.2 subject name      : C=DE, L=Cologne, CN=192.168.1.2 issued  on        : 2023-06-23 22:54:16 expires on        : 2033-06-20 22:54:16 signed using      : RSA with SHA-256 RSA key size      : 2048 bits basic constraints : CA=true
<html><body>

Anmeldung mit TLS Zertifikat | nicht aktiv
-- | --

</body>
</html>

</body>
</html>

tbnobody, 2023-06-25:
Seems that the SSL implementation of ESP32 does not support IP addresses as SAN

https://www.esp32.com/viewtopic.php?t=19580