tbnobody/OpenDTU #1073: "Welchselrichter ist hinter Leistungsmesser" switch does not work

alexseuf, 2023-06-24:
### What happened?

if i what to switch off "Welchselrichter ist hinter Leistungsmesser" in menue mowerlimiter and press the "speichern" button I get the following "error message": "Bitte klicken Sei diese Kästchen um fortfahren zu können".
changes will be not saved. if i  reload menue mowerlimiter swithc is still enabled.
have you got a wiring diagramm how power meter is connected at default settings? 
for me zero consumtion regulation only works if i place shelly 3em after hoymiles connection point?

Thank you

### To Reproduce Bug

try to switch off "Welchselrichter ist hinter Leistungsmesser" in menue mowerlimiter and press the "speichern" button 

### Expected Behavior

switch work 

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

f018a01

### Relevant log/trace output

```Shell
RX Period End
13:51:09.100 > Success
13:51:09.100 > Fetch inverter: 116484424196
13:51:09.205 > TX RealTimeRunData 865.00 MHz --> 15 84 42 41 96 80 12 06 72 80 0B 00 64 96 D8 AD 00 00 00 00 00 00 00 00 DB 16 23 
13:51:09.317 > Interrupt received
13:51:09.420 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 01 00 01 00 0C 00 0C 00 01 00 00 00 00 00 00 00 00 63 | -33 dBm
13:51:09.523 > Interrupt received
13:51:09.615 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 03 02 08 0A AA 0A BC 00 00 02 F4 00 00 02 46 01 53 9D | -33 dBm
13:51:09.715 > Interrupt received
13:51:09.766 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 04 01 54 09 56 13 89 14 55 00 00 00 D9 03 E8 01 6A EE | -34 dBm
13:51:09.832 > Interrupt received
13:51:09.894 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 85 00 1F F6 D9 D7 | -34 dBm
13:51:10.025 > RX Period End
13:51:10.025 > Middle missing
13:51:10.025 > Request retransmit: 2
13:51:10.025 > TX RequestFrame 865.00 MHz --> 15 84 42 41 96 80 12 06 72 82 60 
13:51:10.144 > Interrupt received
13:51:10.230 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 02 00 00 00 00 00 00 00 00 00 00 02 10 02 11 02 05 66 | -34 dBm
13:51:10.301 > RX Period End
13:51:10.301 > Success
13:51:14.108 > Fetch inverter: 116484424196
13:51:14.220 > TX RealTimeRunData 865.00 MHz --> 15 84 42 41 96 80 12 06 72 80 0B 00 64 96 D8 B2 00 00 00 00 00 00 00 00 EB 9A 80 
13:51:14.292 > Interrupt received
13:51:14.426 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 01 00 01 00 0C 00 0C 00 01 00 00 00 00 00 00 00 00 63 | -34 dBm
13:51:14.497 > Interrupt received
13:51:14.660 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 02 00 00 00 00 00 00 00 00 00 00 02 10 02 11 02 05 66 | -33 dBm
13:51:14.748 > Interrupt received
13:51:14.830 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 03 02 08 0A AA 0A BC 00 00 02 F4 00 00 02 47 01 53 9C | -34 dBm
13:51:14.895 > Interrupt received
13:51:15.041 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 04 01 55 09 55 13 89 14 55 00 00 00 D9 03 E8 01 6A EC | -33 dBm
13:51:15.155 > Interrupt received
13:51:15.246 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 85 00 1F B6 DB 95 | -34 dBm
13:51:15.365 > RX Period End
13:51:15.365 > Success
13:51:17.338 > PowerMeterClass: TotalPower: 523.71
13:51:17.338 > [PowerLimiterClass::loop] ******************* ENTER **********************
13:51:17.338 > [PowerLimiterClass::loop] dcVoltage: 52.80 Voltage Start Threshold: 25.00 Voltage Stop Threshold: 24.00 inverter->isProducing(): 1
13:51:17.338 > [PowerLimiterClass::loop] newPowerLimit: 514
13:51:17.338 > [PowerLimiterClass::loop] Limit Non-Persistent: 514 W
13:51:17.338 > [PowerLimiterClass::loop] Status: SolarPT enabled 0, Drain Strategy: 0, canUseDirectSolarPower: 0, Batt discharge: 1
13:51:17.338 > [PowerLimiterClass::loop] Status: StInterrupt received
13:51:17.499 > RX 865.00 MHz --> D1 84 42 41 96 80 12 06 72 81 00 00 0B 00 14 07 BF | -34 dBm
13:51:19.343 > RX Period End
13:51:19.343 > Success
13:51:19.343 > Fetch inverter: 116484424196
13:51:19.453 > TX RealTimeRunData 865.00 MHz --> 15 84 42 41 96 80 12 06 72 80 0B 00 64 96 D8 B7 00 00 00 00 00 00 00 00 BB A5 EA 
13:51:19.561 > Interrupt received
13:51:19.649 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 01 00 01 00 0C 00 0C 00 01 00 00 00 00 00 00 00 00 63 | -34 dBm
13:51:19.720 > Interrupt received
13:51:19.777 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 02 00 00 00 00 00 00 00 00 00 00 02 10 02 11 02 09 6A | -33 dBm
13:51:19.868 > Interrupt received
13:51:19.926 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 03 02 0C 0A BF 0A D2 00 00 02 F4 00 00 02 47 01 53 E3 | -33 dBm
13:51:20.073 > Interrupt received
13:51:20.161 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 04 01 55 09 55 13 89 14 7D 00 04 00 DB 03 E7 01 6B CC | -34 dBm
13:51:20.277 > Interrupt received
13:51:20.366 > RX 865.00 MHz --> 95 84 42 41 96 80 12 06 72 85 00 1F 9E 72 14 | -34 dBm
13:51:20.436 > RX Period End
13:51:20.436 > Success
```

tbnobody, 2023-06-24:
To be honest, I have no clue what you are talking about. This project does not contain such parameters or functions which you are refering to. 
Maybe you are using https://github.com/helgeerbe/OpenDTU-OnBattery or some other code modification, but then you should open an issue there.

alexseuf, 2023-06-24:
sorry my mistake, you are right

alexseuf, 2023-06-24:
wrong github branch

closed 2023-06-24 by alexseuf