tbnobody/OpenDTU #1076: Unbekanntes Modell HM-1500

yetility, 2023-06-25:
### What happened?

HM-1500
Hardware-Teilenummer | 132449480

### To Reproduce Bug

Wird in den Wechselrichterinformationen als unbekanntes Modell angezeigt.

### Expected Behavior

Sollte als Modell HM-1500 angezeigt werden.

[image]

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.21

tbnobody, 2023-06-26:
Could you please doublecheck if you still get this Hardware-Teilenummer after a reboot of the inverter (e.g. at night?) Because this Teilenummer looks strange compared to the other ones we currently found.

homeautomation2022, 2023-06-26:
Die Hardware-Version ist ebenfalls unrealistisch...

Hast du schon eine Idee warum immer mal wieder ein falsches Paket als DevInfo geparst wird? Der schonmal angemerkte Plausiblitätscheck würde in dem obigen Fall auch nur äußerst schwer funktionieren. Ich habe jetzt leider nicht im Kopf, was bereits alles implementiert ist, aber die Länge des Paketes sowie die CRC sollte ja geprüft werden???

tbnobody, 2023-06-26:
> aber die Länge des Paketes sowie die CRC sollte ja geprüft werden

Die CRC wird geprüft. Und alle Checks sind in dem Moment i.O. Nachdem die CRC ok ist muss auch die länge passen. Weil wenn auf dem Weg Daten verloren gehen würden, würde die CRC nicht stimmen. 
Mir ist hier noch nicht ganz klar was der Inverter in dem Moment treibt.

tbnobody, 2023-07-08:
Closing this issue due to inactivity... feel free to open it again

closed 2023-07-08 by tbnobody