tbnobody/OpenDTU #1084: Wrong values after sunset - HMS-450

nonflammable, 2023-06-26:
### What happened?

[image]

### To Reproduce Bug

it happens periodically after sunset

### Expected Behavior

0V, 0A

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

 582867c

tbnobody, 2023-06-26:
To determine where this value is coming from it would be required to get a log trace (Info --> Console) from the period of after sunset when the value switches from the daily value to this one.

nonflammable, 2023-06-26:
18:36:26.350 > Fetch inverter: 1124
18:36:26.421 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE 8A 00 00 00 00 00 00 00 00 9F 7A 3D 
18:36:26.492 > Interrupt received
18:36:26.558 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 00 F8 00 02 00 06 00 00 06 40 00 3F 08 FA 6D | -70 dBm
18:36:26.943 > RX Period End
18:36:26.943 > Last missing
18:36:26.943 > Request retransmit: 2
18:36:26.943 > TX RequestFrame 865.00 MHz --> 15 83 74 85 28 80 13 46 00 82 18 
18:36:27.010 > Interrupt received
18:36:27.046 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 85 00 00 00 00 00 00 00 00 01 36 00 07 26 42 5A | -70 dBm
18:36:27.097 > RX Period End
18:36:27.097 > Success
18:36:31.481 > Fetch inverter: 1124
18:36:31.553 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE 8F 00 00 00 00 00 00 00 00 CF 45 57 
18:36:31.609 > Interrupt received
18:36:31.710 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 00 F7 00 02 00 06 00 00 06 40 00 3F 08 F7 6F | -70 dBm
18:36:31.768 > Interrupt received
18:36:31.887 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 84 00 00 00 00 00 00 00 00 01 36 00 07 B9 8D 0B | -70 dBm
18:36:32.035 > RX Period End
18:36:32.035 > Success
18:36:36.589 > Fetch inverter: 1124
18:36:36.638 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE 94 00 00 00 00 00 00 00 00 3F FB 02 
18:36:36.809 > Interrupt received
18:36:36.913 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 00 F6 00 02 00 06 00 00 06 40 00 3F 08 FA 63 | -70 dBm
18:36:36.985 > Interrupt received
18:36:37.124 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 85 00 00 00 00 00 00 00 00 01 36 00 07 BF BC 3D | -70 dBm
18:36:37.176 > RX Period End
18:36:37.176 > Success
18:36:41.682 > Fetch inverter: 1124
18:36:41.735 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE 99 00 00 00 00 00 00 00 00 AF A3 C7 
18:36:42.298 > RX Period End
18:36:42.298 > All missing
18:36:42.298 > Nothing received, resend whole request
18:36:42.298 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE 99 00 00 00 00 00 00 00 00 AF A3 C7 
18:36:42.352 > Interrupt received
18:36:42.415 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 00 F5 00 02 00 06 00 00 06 40 00 3F 08 FC 66 | -70 dBm
18:36:42.446 > Interrupt received
18:36:42.491 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 85 00 00 00 00 00 00 00 00 01 36 00 07 7A 3B 7F | -70 dBm
18:36:42.808 > RX Period End
18:36:42.808 > Success
18:36:46.792 > Fetch inverter: 1124
18:36:46.849 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE 9E 00 00 00 00 00 00 00 00 9F 85 D6 
18:36:46.926 > Interrupt received
18:36:46.954 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 00 F4 00 02 00 06 00 00 06 40 00 3F 08 FE 65 | -70 dBm
18:36:47.008 > Interrupt received
18:36:47.164 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 86 00 00 00 00 00 00 00 00 01 36 00 07 3A 42 45 | -70 dBm
18:36:47.397 > RX Period End
18:36:47.397 > Success
18:36:51.918 > Fetch inverter: 1124
18:36:51.979 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE A4 00 00 00 00 00 00 00 00 3E AF 67 
18:36:52.523 > RX Period End
18:36:52.523 > All missing
18:36:52.523 > Nothing received, resend whole request
18:36:52.523 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE A4 00 00 00 00 00 00 00 00 3E AF 67 
18:36:52.577 > Interrupt received
18:36:52.789 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 00 F2 00 02 00 06 00 00 06 40 00 3F 08 FF 62 | -71 dBm
18:36:52.844 > Interrupt received
18:36:52.914 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 85 00 00 00 00 00 00 00 00 01 36 00 07 37 87 8E | -71 dBm
18:36:53.032 > RX Period End
18:36:53.032 > Success
18:36:57.005 > Fetch inverter: 1124
18:36:57.058 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE A9 00 00 00 00 00 00 00 00 AE F7 A2 
18:36:57.122 > Interrupt received
18:36:57.162 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 00 F1 00 02 00 06 00 00 06 40 00 3F 08 FE 60 | -71 dBm
18:36:57.285 > Interrupt received
18:36:57.348 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 86 00 00 00 00 00 00 00 00 01 35 00 07 33 B2 BF | -70 dBm
18:36:57.564 > RX Period End
18:36:57.564 > Success
18:37:02.133 > Fetch inverter: 1124
18:37:02.198 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE AE 00 00 00 00 00 00 00 00 9E D1 B3 
[... 191 lines cut ...]
18:38:13.730 > Interrupt received
18:38:13.882 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 81 0B 00 09 00 50 04 CD | -71 dBm
18:38:14.168 > RX Period End
18:38:14.168 > Success
18:38:18.542 > Fetch inverter: 1124
18:38:18.636 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE FA 00 00 00 00 00 00 00 00 5D 1F EA 
18:38:19.146 > RX Period End
18:38:19.146 > All missing
18:38:19.146 > Nothing received, resend whole request
18:38:19.146 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE FA 00 00 00 00 00 00 00 00 5D 1F EA 
18:38:19.699 > RX Period End
18:38:19.699 > All missing
18:38:19.699 > Nothing received, resend whole request
18:38:19.699 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE FA 00 00 00 00 00 00 00 00 5D 1F EA 
18:38:20.249 > RX Period End
18:38:20.249 > All missing
[... cut ...] send whole request
18:38:20.249 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE FA 00 00 00 00 00 00 00 00 5D 1F EA 
18:38:20.804 > RX Period End
18:38:20.804 > All missing
18:38:20.804 > Nothing received, resend whole request
18:38:20.804 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 99 BE FA 00 00 00 00 00 00 00 00 5D 1F EA 
18:38:21.445 > RX Period End
18:38:21.445 > All missing
18:38:21.445 > Nothing received, resend count exeeded

nonflammable, 2023-06-26:
Everyday after sunset values are the same: 230.4V / 204.84A

jstammi, 2023-06-27:
Just had a quick look only

And from what I see there are received for/associated to RealTimeRunData TX, that do not match from fragments count nor their size. As shown in lines e.g. 113-115, 135-138 of your "Logfile" sheet.

Similar to what I observe for issue #1022

Since yesterday I have a patched version running, where I check for number of fragments and their sizes prior parsing them.

Please have a look to the changes in https://github.com/jstammi/OpenDTU/tree/bugfix/up-1022_validate-packets-before-parsing that I applied, being the basis for such checking (currently only checking such for RealTimeRunData aka Statistics of HM-600. But the way I added this finally (different from what I announced first as I got packet handling impl wrong that time) is extensible to all inverters.

And please feel free to provide me any feedback on the way I added it. As at some time in the future I intend to send you a PR on this ;-).

tbnobody, 2023-06-27:
Please checkout todays version v23.6.27

nonflammable, 2023-06-27:
19:33:48.726 > Fetch inverter: 1124
19:33:48.778 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 7F 00 00 00 00 00 00 00 00 FE BC CE 
19:33:49.332 > RX Period End
19:33:49.332 > All missing
19:33:49.332 > Nothing received, resend whole request
19:33:49.332 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 7F 00 00 00 00 00 00 00 00 FE BC CE 
19:33:49.884 > RX Period End
19:33:49.884 > All missing
19:33:49.884 > Nothing received, resend whole request
19:33:49.884 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 7F 00 00 00 00 00 00 00 00 FE BC CE 
19:33:50.433 > RX Period End
19:33:50.433 > All missing
19:33:50.433 > Nothing received, resend whole request
19:33:50.433 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 7F 00 00 00 00 00 00 00 00 FE BC CE 
19:33:50.991 > RX Period End
19:33:50.991 > All missing
19:33:50.991 > Nothing received, resend whole request
19:33:50.991 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 7F 00 00 00 00 00 00 00 00 FE BC CE 
19:33:51.502 > RX Period End
19:33:51.502 > All missing
19:33:51.502 > Nothing received, resend count exeeded
19:33:53.732 > Fetch inverter: 1124
19:33:53.791 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 84 00 00 00 00 00 00 00 00 CA CB 76 
19:33:54.334 > RX Period End
19:33:54.334 > All missing
19:33:54.334 > Nothing received, resend whole request
19:33:54.334 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 84 00 00 00 00 00 00 00 00 CA CB 76 
19:33:54.394 > Interrupt received
19:33:54.457 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 01 19 00 2C 00 7C 00 00 06 93 00 40 08 F4 7B | -75 dBm
19:33:54.508 > Interrupt received
19:33:54.554 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 8A 00 76 00 00 00 05 03 E8 00 D8 00 09 E0 77 DF | -75 dBm
19:33:54.840 > RX Period End
19:33:54.840 > Success
19:33:58.734 > Fetch inverter: 1124
19:33:58.807 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 89 00 00 00 00 00 00 00 00 5A 93 B3 
19:33:58.890 > Interrupt received
19:33:58.943 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 01 19 00 2C 00 7C 00 00 06 93 00 40 08 F6 79 | -74 dBm
19:33:58.999 > Interrupt received
19:33:59.064 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 8B 00 76 00 00 00 05 03 E8 00 D8 00 09 60 F7 DE | -74 dBm
19:33:59.291 > RX Period End
19:33:59.291 > Success
19:34:03.737 > Fetch inverter: 1124
19:34:03.787 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 8E 00 00 00 00 00 00 00 00 6A B5 A2 
19:34:04.387 > RX Period End
19:34:04.387 > All missing
19:34:04.387 > Nothing received, resend whole request
19:34:04.387 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 8E 00 00 00 00 00 00 00 00 6A B5 A2 
19:34:04.448 > Interrupt received
19:34:04.506 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 01 21 00 2B 00 7B 00 00 06 93 00 40 08 F3 44 | -73 dBm
19:34:04.559 > Interrupt received
19:34:04.614 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 8A 00 75 00 00 00 05 03 E8 00 D8 00 09 39 0D 7F | -74 dBm
19:34:04.841 > RX Period End
19:34:04.841 > Success
19:34:08.737 > Fetch inverter: 1124
19:34:08.820 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 93 00 00 00 00 00 00 00 00 3A 20 7A 
19:34:08.869 > Interrupt received
19:34:08.921 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 01 1B 00 2C 00 7C 00 00 06 93 00 40 08 F2 7F | -75 dBm
19:34:08.978 > Interrupt received
19:34:09.132 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 8A 00 76 00 00 00 05 03 E8 00 D7 00 09 A4 3E DD | -75 dBm
19:34:09.295 > RX Period End
19:34:09.295 > Success
19:34:13.790 > Fetch inverter: 1124
19:34:13.941 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 98 00 00 00 00 00 00 00 00 0A 53 32 
19:34:14.387 > RX Period End
19:34:14.387 > All missing
19:34:14.387 > Nothing received, resend whole request
19:34:14.387 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D 98 00 00 00 00 00 00 00 00 0A 53 32 
19:34:14.939 > RX Period End
19:34:14.939 > All missing
19:34:14.939 > Nothing received, resend whole request
[... 42 lines cut ...]
19:34:24.852 > Success
19:34:28.750 > Fetch inverter: 1124
19:34:28.801 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D A7 00 00 00 00 00 00 00 00 FB 46 E9 
19:34:29.353 > RX Period End
19:34:29.353 > All missing
19:34:29.353 > Nothing received, resend whole request
19:34:29.353 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D A7 00 00 00 00 00 00 00 00 FB 46 E9 
19:34:29.912 > RX Period End
19:34:29.912 > All missing
19:34:29.912 > Nothing received, resend whole request
19:34:29.912 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D A7 00 00 00 00 00 00 00 00 FB 46 E9 
19:34:30.458 > RX Period End
19:34:30.458 > All missing
19:34:30.458 > Nothing received, resend whole request
19:34:30.458 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D A7 00 00 00 00 00 00 00 00 FB 46 E9 
19:34:31.006 > RX Period End
19:34:31.006 > All missing
19:34:31.006 > Nothing received, resend whole request
19:34:31.006 > TX RealTimeRunData 865.00 MHz --> 15 83 74 85 28 80 13 46 00 80 0B 00 64 9B 1D A7 00 00 00 00 00 00 00 00 FB 46 E9 
19:34:31.063 > Interrupt received
19:34:31.114 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 01 00 01 01 1A 00 2B 00 79 00 00 06 93 00 40 08 F0 7E | -74 dBm
19:34:31.177 > Interrupt received
19:34:31.223 > RX 865.00 MHz --> 95 83 74 85 28 80 13 46 00 82 13 89 00 73 00 00 00 05 03 E8 00 D7 00 09 29 03 6B | -74 dBm
19:34:31.514 > RX Period End
19:34:31.514 > Success

nonflammable, 2023-06-27:
waiting to sunset

jstammi, 2023-06-27:
> Please checkout todays version v23.6.27

Ah, you implemented it the generic way, much easier that what I did.

Maybe I am wrong, but imho all `StatisticParser` instances will now use the same `maxByteCount` ... don't they?
Would be a problem only for someone using multiple inverters with statistics packets of different sizes.

tbnobody, 2023-06-28:
>  Would be a problem only for someone using multiple inverters with statistics packets of different sizes.

You are right. I just removed the release from yesterday again... Will fix it this evening. 
For some reasons I assumed that a static variable in a member function is unique for each class instance which is obviously not the case.

nonflammable, 2023-06-28:
> Please checkout todays version v23.6.27

it was better, but after sunset the values did not drop to zero, but stayed all night

[image]

tbnobody, 2023-06-28:
> after sunset the values did not drop to zero, but stayed all night

Yes, thats the normal behavior...

nonflammable, 2023-06-28:
after upgrade in my case it can be closed

closed 2023-06-28 by tbnobody