tbnobody/OpenDTU #1097: [Request] Check when i upload a config.json or pin_mapping.json

tuvok81, 2023-06-28:
### Is your feature request related to a problem? Please describe.

I have download a pin_mapping json file .. from github with saving with right click .. but i have no idea it was correct.. and this have save a wrong input in the file .. but openDTU have upload them .. and i have problems that the openDTU is running correct..

The Problem is iam not a programmer and i have no idea for the syntax from json .. we need very long time on the discord with angry peoples because of my stupidity to fix this problem.

### Describe the solution you'd like

a check system that check some .json what is need to run correct. with syntax or other things

mroenne2022, 2023-06-29:
Hello @tuvok81 , i think you must open the file on github with one "click" , and then use the button to download the file. See screenshot, i hope it is correct.
[image]

stefan123t, 2024-07-10:
@tuvok81 I am sorry for your troubles, but the validation of the JSON file prior to uploading can not easily be done on the OpenDTU side (ESP32). 

@tbnobody however we may consider to at least validate the general JSON structure on the client / browser side using some library before uploading it to the OpenDTU ?

[How to make a JSON Linter / Validator with Vue](https://dev.to/codeply/how-to-make-a-json-linter-validator-with-vue-3m82)

schlimmchen, 2024-10-27:
> the validation of the JSON file prior to uploading can not easily be done on the OpenDTU side (ESP32).

I disagree. When uploading the file, we can have ArduinoJson read the contents, and tell us if it is indeed JSON or not.

stefan123t, 2024-10-27:
@schlimmchen yes you are right, that is true as well. Now it is up to @tbnobody to decide whether he prefers to use the Desktop / Mobile Browser to validate the JSON or he wants to use the limited CPU cycles on the OpenDTU ESP32 backend side to check its validity.

tbnobody, 2024-11-07:
Implemented in 8019eaf18218a4efb52995cb1bf861ba7e44450e

closed 2024-11-07 by tbnobody