tbnobody/OpenDTU #1102: Wrong Data in Webinterface

kthemall, 2023-07-01:
### What happened?

power is measured not correctly on hm-800 with 2 different power supplies. see pics below. just measured with amp-meter and output of the hm-800 is 780w not 459w as shown in the opendtu gui.
[image]

### To Reproduce Bug

use 24v on string one and 50v on string 2 of the hm-800 and check the power.

### Expected Behavior

should show the correct values.

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

v23.6.28

### Anything else?

[image]
[image]
[image]

Solarteur, 2023-07-01:
Same problem as discussed here: https://github.com/tbnobody/OpenDTU/discussions/1078

tbnobody, 2023-07-01:
Please provide some output of the log (either serial log or from the web console). I like to doublecheck if the values are coming directly out of the inverter like this (then most likely there cannot be done anything as the problem does not relay on OpenDTU) or if there is some kind of display/calculation isssue.

Dekalus, 2023-07-01:
> Please provide some output of the log (either serial log or from the web console). I like to doublecheck if the values are coming directly out of the inverter like this (then most likely there [...]

Was my post taken into account at all? Somehow there is no response to messages/questions.

[Here](https://github.com/tbnobody/OpenDTU/discussions/1078#discussioncomment-6276495) I have already well documented all the essential data. So I have already inserted a console recording here:

https://github.com/tbnobody/OpenDTU/files/11861321/HM-800.-.Konsolenmitschnitt.txt

tbnobody, 2023-07-01:
Please see attached detailed analysis of the raw data.
The 100W as well as the 50W are coming directly out of the inverter. (please keep in mind, that it's a known issue that the firmware of the 2 channel inverters send most likely wrong power factors)

Not sure why this is happening or if its somehow related to the reactive power vs active  power. Especially the ower factor is very low if the voltage is low (that's what I see from my hm-1500)

[GH_Issue_1102.xlsx](https://github.com/tbnobody/OpenDTU/files/11926087/GH_Issue_1102.xlsx)

kthemall, 2023-07-01:
so if you need any more data here it is:

Websocket: [/livedata][42] disconnect
20:14:42.192 > Fetch inverter: 114184607335
20:14:42.244 > TX RealTimeRunData Channel: 40 --> 15 84 60 73 35 80 11 20 52 80 0B 00 64 A0 6D 11 00 00 00 00 00 00 00 00 93 80 74 
20:14:42.293 > Interrupt received
20:14:42.480 > RX Channel: 3 --> 95 84 60 73 35 84 60 73 35 01 00 01 02 14 01 A8 08 C6 01 04 04 05 0A 66 00 00 8C | -80 dBm
20:14:42.685 > Interrupt received
20:14:42.792 > RX Channel: 23 --> 95 84 60 73 35 84 60 73 35 83 00 04 00 C3 03 E8 01 0E 00 0A C2 8C 71 | -80 dBm
20:14:42.894 > RX Period End
20:14:42.894 > Middle missing
20:14:42.894 > Request retransmit: 2
20:14:42.894 > TX RequestFrame Channel: 61 --> 15 84 60 73 35 80 11 20 52 82 D6 
20:14:42.943 > Interrupt received
20:14:43.094 > RX Channel: 3 --> 95 84 60 73 35 84 60 73 35 02 BA BF 00 00 D5 69 00 08 00 10 09 60 13 85 12 41 9A | -80 dBm
20:14:43.297 > RX Period End
20:14:43.297 > Success

fritz wattmeter measures 786W and OpenDTU shows 463W. hope that helps

tom

kthemall, 2023-07-05:
hi, found out that since i have both panel connectors on 48v (before one was 48v and the other was 24v) everything works correctly. just for info. thx tom

closed 2024-01-14 by tbnobody

Cartman586, 2024-01-18:
Hallo,

bei 100% zeigt er bei Opendtu 494,6 Watt, in Wahrheit sind es aber 810,9 Watt bei 100%.
Das Problem ist nicht die opendtu, auch die Hoymiles DTU zeigt zeigt 494,6 Watt.
Durch die „niedrige Lifepo4 Spannung von rund 25,6V kommt der Wechselrichter nicht ganz klar.
Daher die Bitte um einen Offset.
Geprüft durch ein externes Strommessgerät, eingespeist werden wirklich 810,9 Watt
Bitte um Hilfe vielen Dank

Details beim Messen
25,6V 100Ah LifePO4

Measuring device" OpenDTU/Hoymiles Abweichung / Deviation
100% 804,3 W 61% 490,7 39% Too much
90% 804,3 W 55% 490,7 35% Too much
80% 804,3 W 49% 490,8 31% Too much
70% 804,3 W 43% 491,3 27% Too much
60% 708,4 W 41% 479,5 19% Too much
50% 418,7 W 47% 392,6 3% OK
40% 335 W 39% 325,1 1% OK
30% 244,6 W 29% 232,4 1% OK
20% 145,1 W 19% 136,1 1% OK
10% 60,1 W 10% 60,9 0% OK

Denke ich auch. 
Darum möchte ich gerne einen Offset für 30%,40%,50%,60%,70%,80%,90,100% festlegen können.
Das Problem ist generell bekannt und so könnte man entgegen wirken.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]