tbnobody/OpenDTU #1105: NRF24L01+ connection unstable due to NRF24L01+ supply issue

ZufallsAnwendernameGeneratorGeraet, 2023-07-01:
### What happened?

I have had issues with failed commands and failed updates in live view. Basically very unstable connection to the HM1500 .Then i thought i might add an small 10uF capacitor to the NRF24L01+ module. After that i had no issues any more. A quick search found that this is commonly suggested when using the NRF24L01 module. 

may be this should be mentioned in the README.md.

[image]

### To Reproduce Bug

Connect an NRF24L01+ module by jumper wire to the ESP32 module

### Expected Behavior

Basically stable connection to the HM1500 (only a few meter distance).

### Install Method

Pre-Compiled binary from GitHub

### What git-hash/version of OpenDTU?

Any

tbnobody, 2023-07-01:
The capacitor is mentioned in the troubleshooting area:

The NRF24L01+ needs relatively much current. With bad power supply (and especially bad cables!) a 10 µF capacitor soldered directly to the NRF24L01+ board connector brings more stability (pin 1+2 are the power supply). Note the polarity of the capacitor…

his1220, 2023-07-09:
added the capactior and my problems with unknown commands and interruptions is gone. Thanks for the very great suggestion!!

112Feuerball112, 2023-07-11:
Could you upload a guide on how to expand the capacitor with charging?

ZufallsAnwendernameGeneratorGeraet, 2023-07-16:
@112Feuerball112 
what do you exactly mean with "expand" the capacitor?

tbnobody, 2024-01-14:
The capacitors are now also shown in the [documentation](https://tbnobody.github.io/OpenDTU-docs/hardware/esp32nodemcu_38pin/#schematic)

closed 2024-01-14 by tbnobody