tbnobody/OpenDTU #1108: [Request] Automatic limit to 800W

Bankaifan, 2023-07-03:
### Describe the solution you'd like

Can we have a function, that limits the total output of all (selected) inverters to 800W as required by the EU directive?

In the morning, when the eastern invers deliver the most energy (lets say 600W in total), the western invers would deliver something like 400W in total. So those should be limited to 200W. When the total energy goes below 800W, the limit should be opened, again, till power reaches 800W again. 

Maybe it would also work to limit all invers in total in 1% steps till they are below 800W and step up again, when total power falls below.

maenzm, 2023-07-03:
That would be a great feature. But the limit should be freely selectable.

DPO99, 2023-07-04:
Ich glaube nicht, dass es sinnvoll ist so etwas einzubauen...überlegt doch mal. Du nutzt z.B. 4 Hoymiles. Jetzt fällt über einen HM ein schatten. Woher weiß die DTU denn, wie viel Leistung die übrigen HM liefern könnten? Man müsste also die HM nach und nach hochschrauben, bis die Leistung wieder erreicht ist. In die andere Richtung müsste er immer prüfen. welchen HM er nun runterschrauben kann/muss, damit die Leistung nicht überschritten wird. Unterm Strich wird das ewig lange dauern und ggf. ist die DTU bei sehr wechselreichem Wetter nur noch am Regeln + du würdest sicher nicht unerheblich oft über die 800w drüber rutschen und somit sowieso gegen die Regelung verstoßen. Zudem ist es m.E. gem. dieser Regelung auch nicht zulässig eine höhere Leistung als 800w zu installieren, auch wenn man diese "theoretisch" nur x mal überschreitet. Was passiert denn z.B. auch wenn die DTU mal ausfällt? Im besten fall über Nacht, wenn die Inverter alle neu starten und die Limits auf 100% zurück gesetzt werden, dann fackelst Du dir noch die Bude ab...

Ich kann persönlich schon nachvollziehen, dass man das maximalste aus seiner Anlage herausholen will ABER das ganze über Mikrowechselrichter und am besten ohne Stromzähler oder andere weitere Hardware nur durch die DTU regeln zu lassen geht einfach unfassbar weit an der Grundidee der DTU vorbei. Es gibt Gründe, wieso div. Hersteller für Ihre Solarsteuerung Geld verlangen, da das eben nicht einfach so mal über einen kleinen ESP32 mit minimalster Hardware umzusetzen ist.

MrReSc, 2023-07-16:
The topic has been discussed many times here. OpenDTU is an interface to read the values of the inverters. It does not perform any control tasks.

I also have two inverters. I set the persistent limit so that the inverters do not exceed 600 W after waking up in the morning. This ensures that when my controller is not running, the limits are not exceeded. After that, I use the "non persistent limit" to dynamically control the limits depending on the consumption in the house so that never more than 600 W is fed into the grid. I use Home Assistant for this. As the previous speaker already mentioned, there are a few things to consider and the whole thing is also quite slow.

DejanBukovec, 2023-07-17:
Im do this with automation in HA. My zwave meter report mi positive or negative power on phases on small changes. Then I in automation every 20 seconds calculate depend if I need more or less power from inverters how much I need power and divide power by inverter count and add/substract this value to current power output of inverter and send this new limit by MQTT... 

My house power meter installed by distributor have limit for power distribution to network set to 824W if I in 1 minute(average of last minute) exceed this power then it shutdown house power and I need manually press button on meter to enable it... So I set that my max return to grid is 100W so it regulate inverter power every 20 seconds and try to be between -100W and 0W :)

Main problem is that there is MQTT publish interval which add some delay to get current values to HA. There is pooling delay which add delay and there is another issue(reported on other issue) if commands are send to often then pooling is delayed and can take a lot of time to get correct current data from inverters... Maybe this is not problem with 1 inverter but for example with 3 as I have GUI of opendtu do not work most of time(Also reported on other issue)... 

So is not optimal solution but for me currently work ok:
[image]

I don't know how accurate and fast is original DTU together with DTSU666 meter, maybe someone who try them can tell more...

stefan123t, 2024-07-10:
@Bankaifan please double check the **Dynamic Power Limiter (DPL)** function in the daughter project[ OpenDTU-OnBattery](https://github.com/helgeerbe/OpenDTU-OnBattery/).

I have described a [possible PM+DPL setup](https://github.com/tbnobody/OpenDTU/issues/272#issuecomment-2219644352) with a Shelly Pro3EM as **Power Meter (PM)** here in issue #272. 
Just change -300W to -800W which allows your inverters to donate up to 800W to the power continuum.